StephanJoubert/home_assistant_solarman #324: deye_4mptt.yaml negative value from the day before in the statistic

OLLIBERTUS, 2023-06-11:
Hello, I have implemented the Deye600 (two times) with 4mqtt.yaml and it works very good but after I import the "Dayli_Production" into the energy dashboard I got the negativ value from the day before in the graph (yellow/light yellow) at the beginning of the day (4:00am). Why it is like this? After the WR works everything is okay and it collect all there data very good.

There are still I two additional shelly plugs (brown/light brown) that's the reason that you see ins summary four production values.

[image]

pascal-mar, 2023-06-12:
Hello, I had the same problem. If you take the total production it is displayed correctly. Greetings Pascal

OLLIBERTUS, 2023-06-12:
Hello together, with support from the community from https://community.simon42.com I have solved the problem. I have add in the "Daily Production" at state_class: "total_increasing" and not only "total", and now it works......I can close the topic
[image]
[image]

closed 2023-06-12 by OLLIBERTUS