StephanJoubert/home_assistant_solarman #326: Sensor values dropping to 0 intermittently

EspeeFunsail, 2023-06-13:
Loving the integration! Just see that something is causing the values to drop to 0 as can be seen in the attached image. It does not do that on the solarman app. So seems like some communication drop between logger and HA. Another question - it seems as though the delay for data to reach HA is about 30 seconds  (Example switching on an appliance, the load only reflects on HA 30 seconds later. Is this normal?
[image]

EspeeFunsail, 2023-06-13:
Similar issue to mine which seems to have been fixed, however I am still getting this problem. Anything I should update or change to have the fix implemented?
https://github.com/StephanJoubert/home_assistant_solarman/issues/268

PV-Joe, 2023-06-13:
My Production has the issue dropping to zero with the latest version.

 The update Intervall is set to 30 seconds by default but still uses 1 minute 🤔
[image]

DHLF, 2023-08-11:
Depending on the inverter (and also the configuration of those) there is a minimum time between real data reading and transferring.
For example "vanilla" installed deye microinverter has a polling time of 5 minutes...
HA may poll them every 30 seconds but to no use.. as the data may be still up to 4 minutes "old".
You have to config the microinverter to shorten the polling time(minimum 1 minute?).

And on the other hand HA seems to poll every 30 seconds even if data changes every second -.-
(feel free to correct me ;) )

EspeeFunsail, 2023-08-11:
> The problem was solved doing a manual update 👍
> Changed the update Interval to 20 seconds now and it is working again 👍

How did you do a manual update?

SP van Zyl

Civil Engineer

GLS Consulting

PV-Joe, 2023-08-12:
Replace the Solarman.py from here and restart HA::
https://github.com/charles-sd/home_assistant_solarman/tree/fix-gaps-missing-data/custom_components/solarman

EspeeFunsail, 2023-08-12:
Ok will try. Did you change any value in solarman.py before replacing it to
get the 20 second refresh time?

PV-Joe, 2023-08-12:
The refresh is changed in constant.py:
MIN_TIME_BETWEEN_UPDATES = timedelta(seconds=20)

EspeeFunsail, 2023-08-12:
Ok cool thanks.

Do you think it will be possible to go lower to 10 or even 5 seconds?

PV-Joe, 2023-08-12:
Try it 😄