StephanJoubert/home_assistant_solarman #330: Bluesun.yaml

sven0833, 2023-06-16:
can some one do a bluesun.yaml for bluesun inverter, all the .yaml file ive tested and all say disconnected

giedriuskilcauskas, 2024-03-12:
[Inverter Modbus Communication  Protocol 2.2(1).pdf](https://github.com/StephanJoubert/home_assistant_solarman/files/14569794/Inverter.Modbus.Communication.Protocol.2.2.1.pdf)
This is registry map. I was able to get values using pysolarmanv5 python directly, but so far integration doesn't work.

this is my current try of definition:
```yaml
requests:
  - start: 2964
    end: 3139
    mb_functioncode: 0x04
  - start: 3200
    end: 3230
    mb_functioncode: 0x03
  - start: 3300
    end: 3340
    mb_functioncode: 0x10
  - start: 4000
    end: 4046
    mb_functioncode: 0x06

parameters:
  - group: Inverter
    items:
      # - name: "Working Mode"
      #   class: ""
      #   state_class: ""
      #   uom: ""
      #   scale: 1
      #   rule: 1
      #   registers: [3040]
      #   icon: 'mdi:home-lightning-bolt'
      #   lookup:
      #       - key: 0
      #         value: "No response mode"
      #       - key: 1
      #         value: "Volt–watt default"
      #       - key: 2
      #         value: "Volt–var"
      #       - key: 3
      #         value: "Fixed power factor"
      #       - key: 4
      #         value: "Fix reactive power"
      #       - key: 5
      #         value: "Power-PF"
      #       - key: 6
      #         value: "Rule21Volt–watt"

      # - name: "Grid status"
      #   class: ""
      #   state_class: ""
      #   uom: ""
      #   scale: 1
      #   rule: 1
      #   registers: [4999]
      #   icon: 'mdi:home-lightning-bolt'

      - name: "Inverter Temperature"
        class: "temperature"
        state_class: "measurement"
        uom: "°C"
        scale: 0.1
        rule: 2
        registers: [3025]
        icon: "mdi:thermometer"

      - name: "Product Model"
        class: ""
        state_class: ""
        uom: ""
        scale: 1
        rule: 5
[... 213 lines cut ...]
        state_class: "total_increasing"
        uom: "kWh"
        scale: 1
        rule: 3
        registers: [3103, 3102]
        icon: "mdi:solar-power"

      - name: "Yearly Generation"
        class: "energy"
        state_class: "total_increasing"
        uom: "kWh"
        scale: 1
        rule: 3
        registers: [3041, 3040]
        icon: "mdi:solar-power"

      - name: "Total Generation"
        class: "energy"
        state_class: "total_increasing"
        uom: "kWh"
        scale: 0.1
        rule: 3
        registers: [3039, 3038]
        icon: "mdi:solar-power"
```

demand187, 2024-08-05:
Have you managed to get it to work in the meantime?

giedriuskilcauskas, 2024-09-06:
> Have you managed to get it to work in the meantime?

@demand187 , I did, here is my custom_parameters.yaml file:
`# BlueSun BSM10k
#
requests:
  - start: 3000
    end: 3020
    mb_functioncode: 0x04
  - start: 3021
    end: 3042
    mb_functioncode: 0x04

parameters:
  - group: Inverter
    items:
      # - name: "Working Mode"
      #   class: ""
      #   state_class: ""
      #   uom: ""
      #   scale: 1
      #   rule: 1
      #   registers: [3040]
      #   icon: 'mdi:home-lightning-bolt'
      #   lookup:
      #       - key: 0
      #         value: "No response mode"
      #       - key: 1
      #         value: "Volt–watt default"
      #       - key: 2
      #         value: "Volt–var"
      #       - key: 3
      #         value: "Fixed power factor"
      #       - key: 4
      #         value: "Fix reactive power"
      #       - key: 5
      #         value: "Power-PF"
      #       - key: 6
      #         value: "Rule21Volt–watt"

      # - name: "Grid status"
      #   class: ""
      #   state_class: ""
      #   uom: ""
      #   scale: 1
      #   rule: 1
      #   registers: [4999]
      #   icon: 'mdi:home-lightning-bolt'

      - name: "Inverter Temperature"
        class: "temperature"
        state_class: "measurement"
        uom: "°C"
        scale: 0.1
        rule: 2
        registers: [3025]
        icon: "mdi:thermometer"

      # - name: "Product Model"
      #   class: ""
      #   state_class: ""
      #   uom: ""
      #   scale: 1
      #   rule: 5
      #   registers: [3228, 3227, 3226, 3225, 3224, 3223, 3222, 3221]
      #   isstr: true

      # - name: "DSP Software Version"
      #   class: ""
      #   state_class: ""
[... 207 lines cut ...]
      #   state_class: "total_increasing"
      #   uom: "kWh"
      #   scale: 1
      #   rule: 3
      #   registers: [3103, 3102]
      #   icon: "mdi:solar-power"

      - name: "Yearly Generation"
        class: "energy"
        state_class: "total_increasing"
        uom: "kWh"
        scale: 1
        rule: 3
        registers: [3041, 3040]
        icon: "mdi:solar-power"

      - name: "Total Generation"
        class: "energy"
        state_class: "total_increasing"
        uom: "kWh"
        scale: 0.1
        rule: 3
        registers: [3039, 3038]
        icon: "mdi:solar-power"
`

Funky-0815, 2025-04-29:
This YAML file didn't work for me. In this post, I've added my working configuration and a link to the Modbus addresses: #284