StephanJoubert/home_assistant_solarman #336: no connected homeassistant

Antoniocampo, 2023-07-01:
I have a DEYE 5k inverter

Until recently, I was monitoring it with HomeAssistant.

But recently I have changed the inverter wifi dongle for an adapter to RJ45 cable and connect it directly to the router.

Since then, HomeAssistant has not received information from the inverter.

However, on the solarman website the inverter data is updated (when I have modified the data of the logger).

I have the same IP address in inverter.

and I have modified the device serial number in home assistant solarman integration, but all entities are shown as not available.

I have used the same template that worked before: deye_hybrid.yaml

Some idea how to receive data from inverter to homeassistant?

Thanks.

NemesisXB, 2023-07-02:
As far as I understand this integration is specifically for the dongle. And even though it is an RJ45 connector, it is not a network connection, it is an RS485 connection so you can not and should not connect it directly to your router.

sanderbaks, 2023-07-02:
> As far as I understand this integration is specifically for the dongle. And even though it is an RJ45 connector, it is not a network connection, it is an RS485 connection so you can not and should not connect it directly to your router.

If this is the case, how come the Solarman website does receive the data using this cable connected to a router? Seems strange ?

NemesisXB, 2023-07-03:
> If this is the case, how come the Solarman website does receive the data using this cable connected to a router? Seems strange ?

I misread. I see OP is using an adapter. @Antoniocampo can you share link to the adapter you are using?

Antoniocampo, 2023-07-03:
So, if the same thing happens to @sanderbaks, I no longer doubt that it is a problem with my installation, because I have already tried everything and homeassistant still does not show data from the inverter.

sanderbaks, 2023-07-03:
@Antoniocampo My installation will not connect. I can see that in the log. But installation states it is succesful.
I think we are having exactly the same issue.

NemesisXB, 2023-07-04:
@sanderbaks, @Antoniocampo are you able to log into the web page of the dongle? As shown here: https://github.com/StephanJoubert/home_assistant_solarman#preparation

Have you tried making the IP 0.0.0.0 and serial number 0 for auto discovery in the integration?

sanderbaks, 2023-07-04:
> @sanderbaks, @Antoniocampo are you able to log into the web page of the dongle? As shown here: https://github.com/StephanJoubert/home_assistant_solarman#preparation
> 
> Have you tried making the IP 0.0.0.0 and serial number 0 for auto discovery in the integration?

Yes I can access the Dongle.
Will try auto discovery now and let you know

sanderbaks, 2023-07-04:
@NemesisXB Just did the installation. Notification was Succes. Found the Dye-Hybrid. (I have Sofar inverter) But I continued installation. No Error messages. However, No connection: See screenshot.
[image]

sanderbaks, 2023-07-04:
BTW: I am using the Solarman_LSW3 .. I have a 3phase inverter, and the LSW dongle. Am i correct?

NemesisXB, 2023-07-04:
> BTW: I am using the Solarman_LSW3 .. I have a 3phase inverter, and the LSW dongle. Am i correct?

The 3-phase registers are quite different, rather try one of the 3 phase definition files like [deye_sg04lp3.yaml](https://github.com/StephanJoubert/home_assistant_solarman/blob/main/custom_components/solarman/inverter_definitions/deye_sg04lp3.yaml)

NemesisXB, 2023-07-04:
@sanderbaks, @Antoniocampo. Looks like the port is different for the LAN logger. It is 30003.

sanderbaks, 2023-07-04:
And what is the Config file ? Can I use the Sofar_LSW3 ?

> On 4 Jul 2023, at 11:36, Christiaan Alberts ***@***.***> wrote:
> 
> 30003

NemesisXB, 2023-07-04:
> And what is the Config file ? Can I use the Sofar_LSW3 ?
> […](#)
> On 4 Jul 2023, at 11:36, Christiaan Alberts ***@***.***> wrote: 30003

No harm in trying them all. But start with the one that is the closest match for your inverter, so I would start with that the Sofar_LSW3

sanderbaks, 2023-07-04:
> … <x-msg://25/#>

This did not work yet.. I will try all of them..

sanderbaks, 2023-07-04:
Is there a way to check activity on the port ?

NemesisXB, 2023-07-04:
> Is there a way to check activity on the port ?
> […](#)

Seems you can configure the dongle Server B to use port 8899 but there are 4 options for mode as per: https://github.com/jmccrohan/pysolarmanv5/issues/22#issuecomment-1301938184

Seems the default mode for port 30003 is standard Modbus-TCP and not Solarman V5. Maybe try setting up server B and maybe select the solarman-ethernet mode.

sanderbaks, 2023-07-04:
Pff .. This getting soooo complicated :)

Dou you guys have it working? Or is it also giving issues?

I have got to go now.. Will try more later.

PV-Joe, 2023-07-04:
> And what is the Config file ? Can I use the Sofar_LSW3 ?
> […](#)

Try the Sofar files:
[image]

Antoniocampo, 2023-07-04:
Hi @NemesisXB ,

I have change port to 30003 in homeassistant (see image), restart and wait about 5 minutes, but still not working.

the same thing that happens to @sanderbaks  

[image]

[image]

NemesisXB, 2023-07-05:
> > > @sanderbaks, @Antoniocampo are you able to log into the web page of the dongle? As shown here: https://github.com/StephanJoubert/home_assistant_solarman#preparation
> > > Have you tried making the IP 0.0.0.0 and serial number 0 for auto discovery in the integration?
> > 
> > 
> > Yes I can access the Dongle. Will try auto discovery now and let you know
> 
> YES, I can log into the web page of the dongle.

It seems the ethertnet dongle does not use the solarmanv5 protocol. It uses pure modbus-tcp. This integration is written specifically for the solarmanv5 protocol.

sanderbaks, 2023-07-05:
OMG .. This sucks ..

@Antoniocampo Lets help each other shall we? I'll try to find an other plugin

sanderbaks, 2023-07-05:
@Antoniocampo 
I am looking at this now:
https://www.youtube.com/watch?v=pnP-DLMsIPE

Antoniocampo, 2023-07-05:
Hello.

It's quite difficult for me to do all that. 

My knowledge of HomeAssistant is very limited.

However, I offer my collaboration to solve this problem :-)

NemesisXB, 2023-07-05:
> @Antoniocampo I am looking at this now: https://www.youtube.com/watch?v=pnP-DLMsIPE

This one seems to be for sofar inverters: https://github.com/wills106/homeassistant-solax-modbus
Here is list of compatible inverters: https://github.com/wills106/homeassistant-solax-modbus/wiki/Compatible-Inverters

sanderbaks, 2023-07-05:
@NemesisXB This wiki also states that you can only use the COM port not the Ethernet port.
Man, this is driving me crazy..

I will soldier on :)

Antoniocampo, 2023-07-05:
It don´t works for me @sanderbaks .

I attach some images files with that I do

[image]
[image]
[image]

sanderbaks, 2023-07-05:
pff super strange :-/.    i am not sure what to do..
I mam now investigating the Solax tool ..

Antoniocampo, 2023-07-09:
I have a raspberry pi 3 connected to the Deye 5kw inverter with a software called ICM solar.

I have installed docker + portainer + hacs + homeassistant on the rapsberry pi 3

Home assistant has immediately detected 3 shelly devices that I have at home and shows data reading, but when I have installed the solarman integration and entered the IP of the inverter, it does not show data.

I think the solarman integration only works with the wifi dongle.

Finally, I think I'm going to switch back to the wifi dongle.

sanderbaks, 2023-07-09:
You are right, it is super strange.  I wish there was someone in here that could help tackle this issue from the bottom up.
Is there not some kind of piece of code to read the raw modbus communication?
Or a very simple "trail and error" yaml profile in order to try to find one single register that can be easily found.

Who can help ??

sanderbaks, 2023-08-05:
So, Update from my side:  I removed the Ethernet logger and replaced it with a WiFi logger (they are not very expensive) ..
It immediately worked.
I also mailed with solarman in China. The confirmed that the Ethernet logger does not transit Modbus data.

So conclusion:

The Ethernet logger (LSE-3) does NOT work with any Home Assistant or Modbus depending app.

fluccats, 2023-08-06:
Hey Guys ! 

Do you know here I can find the version 4 of protocol? I believe that my Logger is a old version and ius not working with the last version....

** step 2 of preparation **
https://github.com/StephanJoubert/home_assistant_solarman#preparation 

Check the version of the solarman logger. If the serial number starts with 17xxxxxxx, 21xxxxxxx or 40xxxxxxx (protocol V5), the component should work. If not, you may need to try the component for V4 of the protocol mentioned above.

Antoniocampo, 2023-08-07:
> So, Update from my side: I removed the Ethernet logger and replaced it with a WiFi logger (they are not very expensive) .. It immediately worked. I also mailed with solarman in China. The confirmed [...]

I have done the same as you.

I have removed the ethernet dongle and bought the wifi dongle.

With the wifi dongle, he was connected to homeassistant in no time.

axxel70, 2023-08-30:
Has anyone found a solution for the lse-3 ethernet logger?
I've been looking for a solution for days but nothing!!!

michaelcorradi, 2023-09-02:
I have the LSE-3 ethernet logger connected with an inverter ZCS HYD 6000 HP.
And I'm looking for a solution to comunicate with solarman.

The LSW-3 is not expensive, but I don't want to use wireless configuration, because cabled configuration is more stable.

Do you think that we can find a solution?