StephanJoubert/home_assistant_solarman #337: Solarman LSW-5 (V3 protocol) - how to connect with HA?

lpawlak, 2023-07-04:
I have WiFI Stock Logger LSW-5 (I think it's V3 protocol). Device serial number3503193325. Can I use solarman integration to connect with HA?
https://d2l38nissjun1p.cloudfront.net/wp-content/uploads/2022/10/Stick-Logger-WiFi-LSW-5.pdf

thhomesystem, 2023-09-10:
Die you Find a sollution ?

siesiu78, 2024-05-19:
> I have WiFI Stock Logger LSW-5 (I think it's V3 protocol). Device serial number3503193325. Can I use solarman integration to connect with HA? https://d2l38nissjun1p.cloudfront.net/wp-content/uploads/2022/10/Stick-Logger-WiFi-LSW-5.pdf

I have the same problem with Afore  - did you resolve it?

psunny28, 2024-06-02:
I tried communicating, I was able to get Socket check
```
import socket
def main():
    sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM, socket.IPPROTO_UDP)
    sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    sock.setsockopt(socket.SOL_SOCKET, socket.SO_BROADCAST, 1)
    sock.settimeout(1.0)
    request = "WIFIKIT-214028-READ"
    address = ("<Your Data logger Stick IP>", 48899)
    sock.sendto(request.encode(), address)
    while True:
        try:
            data = sock.recv(1024)
        except socket.timeout:
            break
        keys = dict.fromkeys(["ipaddress", "mac", "serial"])
        values = data.decode().split(",")
        print(data.decode())
        result = dict(zip(keys, values))
        print(result)
if __name__ == "__main__":
    main()
```

Appart from I was not able to find Any thing