StephanJoubert/home_assistant_solarman #341: inverter definition deye_2mppt.yaml not available/existant (HACS installation)

new-issue, 2023-07-05:
Hi, thanks for providing this integration!
Unfortunately, I have a problem with the setup with HACS. 
In the folder `inverter_definitions` the file `deye_2mppt.yaml` is missing. Therefore of course also in the selection menu when configuring the integration.
Even if I add the file manually to the folder, it is not displayed in the selection menu.
Does anyone have an idea what this could be?

Solarman integration `v1.4.3`
HACS `1.32.1`
Home Assistant `2023.6.3` 
Supervisor `2023.06.4` 
Operating System `10.3` 
Frontend 20230608.0 - `latest `

UncleRiot, 2023-07-06:
Hi,

you uploaded the file manualy. 1st step OK.

2nd step: Edit /config/custom_components/solarman/const.py
Add: 'deye_2mppt.yaml',

Restart HomeAssistant and there it is in your drop down menu

new-issue, 2023-07-06:
That fixed the problem :smiley:  . Thx!

new-issue, 2023-07-06:
Closed by accident. I think that should be fixed in the code.

idl0r, 2023-07-18:
Yup. That works for me as well.