StephanJoubert/home_assistant_solarman #342: not connecting, please help

prankousky, 2023-07-05:
Hi everybody,

I have a `Deye` brand inverter that I am trying to connect to Home Assistant.

# Information

## index_cn.html

Web Ver: 1.0.24

### Inverter

|                          |         |
| --                       | --      |
| Inverter serial number   | 230*    |
| Firmware version (main)  | *blank* |
| Firmware version (slave) | *blank* |
| Inverter model           | ---     |
| Rated power              | --- W   |
| Current power            | 103 W   |
| Yield today              | 0.0 kWh |
| Total yield              | 0.0 kWh |
| Alerts                   | ---     |
| Last updated             | 0       |

Yeah, weirdly enough, it does not display any firmware...

### Device information

|                      |                   |
| --                   | --                |
| Device serial number | 3914*             |
| Firmware version     | MW3_16U_5406_1.53 |

### Remote server information

|                 |               |
| --              | --            |
| Remote server A | connected     |
| Remote server B | Not connected |

## config_hide.html

* Working mode: Data collection (other option is "Transparency")
* Server A Setting
    * IP address: 47.102.152.71
    * Domain name: access1.solarmanpv.com
    * Port 10000
    * Connection: TCP

* Optional Server Setting:
    * IP address: 47.254.74.158
    * Domain name: access2.solarmanpv.com
    * Port: 10000
    * Connection: TCP

* Serial port parameters setting
    * Baud rate: 9600
    * Data bit: 8
    * Parity bit: None
    * Stop bit: 1
    * CTSRTS: Disable

* Internal server parameters setting
    * Protocol: TCP-Server
    * Port: 8899
    * TCP time out setting (no more than 600s): 300

* Mode Select: AP+STA (other option is "AP/STA")
* Invertor Brand Select: Deye (no other options)

## Mobile App (SOLARMAN)

This is all in German, trying to translate as best as possible:

* Kommunicationsprotocolversion: v0.2.0.1
* Hardwareversion: V0.1.0.2
* Firmware-Version DC-Master: V0.1.1.7
* Firmware-Version DC-Slaves: V0.0.0.0
* AC-Version: V1.2.1.8

* Bit of inverter systrem flags: 1
* Number of MPPT-Paths: 2
* Number of phases: 1
* Running state: 4
* MECD state: 0
* GFDI: Disable
* Debug Info 1/2: 0%

Logger:

* Embedded Device SN: 3914* (same as serial above)
* Version:
    * Module Version No.: MW3_16U_5406_1.53
    * Extended System Version: V1.1.00.0F

* Real-time Information:
    * Data Uploading Period: 5Min
    * Data Acquisition Period: 60s
    * Max. No. of Connected Devices: 1
    * Signal Strength: 76
    * Heart Tate: 120s
    * Extended Function: 15
    * IV Curve Supported: 1
    * Batch Command supported: 1
    * Support Reporting Progress: 1
    * AT+UPGRADE COmmand Supported: 254
    * Support Data Block Transparent Transmission: Enable
    * Extended Function2: 1,2,3,4,5,6,7
    * Support AntiBackflow Function Instructions: Enable
    * Method of Protocol Upgrade: 255
    
The app does show production values etc.; for example, currently I have 63.00W production.

All entities in Home Assistant regarding this integration are unavailable.

I was not able to find any other settings, nor was I able to check for possible firmware updates.

I have tried

* deye_4mppt.yaml
* deye_hybrid.yaml (initial attempt)
* deye_sg04lp3.yaml
* deye_string.yaml

This repos README.me states

> Check the version of the solarman logger. If the serial number starts with 17xxxxxxx or 21xxxxxxx (protocol V5), the component should work. If not, you may need to try the component for V4 of the protocol mentioned above.

What does this mean? I don't see anything mentioned above. 

The app seams to read correct values, the production W keeps changing. 

Can you please help me fix this? Thank you in advance for your support :)

prankousky, 2023-07-06:
Update: I suddenly started receiving values in Home Assistant, without changing anything.

I posted the original issue yesterday, I believe it was around 3 in the aftenoon. Around 4.30 in the afternoon, values started popping up in Home Assistant.

However, I am still not sure, which `yaml` is adequate to use. Please take a look at the sensors that have been created, screenshot below. I am using deye_4mppt.yaml at the moment. Should I use a different one? I am hesitant to just try  switching between different ones, because it seems like the integration needs quite some time to start working (so if I change now, I might not receive any values for hours, thus not knowing whether the yaml I switched to is false or just taking its time).

The model is `SUN-M80G3-EU-Q0`, by the way.

[image]

DHLF, 2023-07-06:
Zuerst einmal zum Verständnis:
Du hast einen Deye Microinverter mit 2mppts..
was spricht dagegen **deye_2mppt.yaml** zu verwenden??

elodur, 2023-07-12:
Weil 2mppt.yaml möglicherweise fehlt. Ich habe das Plugin auch gerade via HACS installiert, die Datei fehlt.

prankousky, 2023-07-12:
Mir war zu dem Zeitpunkt einfach nicht klar, dass die deye_2mppt.yaml einfach hätte funktionieren müssen. Das erste Update nach dem Einrichten lässt (anscheinend?) etwas auf sich warten, deshalb wurden mir anfangs gar keine Werte angezeigt.

Nachdem ich die Integration einmal vollständig gelöscht und Home Assistant neu gestartet hatte, lies sie sich problemlos mit 2mppt einrichten.

Die Datei war bei mir von Anfang an dabei, war einfach ein Benutzerfehler.

closed 2023-07-12 by prankousky