StephanJoubert/home_assistant_solarman #349: Wrong status for Deye SUN-M80G3-EU-230

kugel-, 2023-07-14:
Hello,

I bought a mini solar plant [1] and it says it has a SUN-M**80**G3-EU-230 inverter. I believe it's pretty much the same as SUN-**800**G3-EU-230 that is already supported. Maybe just a typo by the seller.

Anyway, after lots of fiddling I managed to integrate that into HA using this awesome add-on. Only issue I found is that the status is displayed as "Fault" while the inverter says everything alright. On this project (https://github.com/DaKarlii/Deymoticz_DeyeInverter) I found that value 4 for the status should be "Normal-Connected" instead of "Fault", and value 3 would be "Fault". See https://github.com/DaKarlii/Deymoticz_DeyeInverter/blob/main/DYRealTime.json#L146

https://www.amazon.de/Balkonkraftwerk-Solaranlage-Steckerfertig-Photovoltaik-genehmigungsfrei-Schwarz/dp/B0C9VYWR7W

idl0r, 2023-07-18:
That's a new version or revision. It's not a typo.

idl0r, 2023-07-19:
How did you manage to get data at all? Getting 
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 546, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/__init__.py", line 28, in async_unload_entry
    hass.data[DOMAIN].pop(entry.entry_id)
    ~~~~~~~~~^^^^^^^^
KeyError: 'solarman'
```
for my SUN-M80G3

kugel-, 2023-07-19:
I installed using the "manual method", thus a very recent git clone.

```
commit 485f0f01208dc18e05e94e41fa62d6d6cb610a39 (HEAD -> main, origin/main, origin/HEAD)
Merge: afd6dd2 c2647de
Author: StephanJoubert <stephan.joubert@gmail.com>
Date:   Wed Jul 12 15:31:53 2023 +0200

    Merge pull request #332 from virtualdj/fix-hyd3k-6k
    
    Fix typos and unit of measurements/entity case for hyd3k-6k
```

For the record, for a test I disconnected AC from the inverter, such that the LED flashes red. The status was still reported as 4. So I think Fault could actually be valid and the inverter simply does not report it correctly.

kugel-, 2023-07-19:
yes

idl0r, 2023-07-22:
I can't get it working for some reason. No data at all :(

idl0r, 2023-08-07:
Ok, turns out, I used the wrong serial number. Using the logger one works and I also have that wrong status, like you do.

DHLF, 2023-08-11:
While one deye2000 status reports "normal" my deye600 anytime throws "fault".
As other people reports the same behaviour it seems to be more like a "faulty" reporting instead of a real fault and may be ignored?
(because using direct web access to the deye600 shows NO error..)