StephanJoubert/home_assistant_solarman #354: Reset daily production at midnight

elodur, 2023-07-17:
Running and monitoring a Deye Sun M80 works fine overall, thanks for your work!
Since it disconnects at night, the daily production counter does not reset at midnight and the old value stays in Home Assistant. When the inverter starts the next day, resets the daily production counter, Home Assistant counts a negative solar production by the value of yesterdays production. 
Is it possible to reset this counter at midnight? I'm afraid I don't know how to do it myself or to contribute to your code.

kugel-, 2023-07-19:
I also experience this.

JaySNL, 2023-07-19:
Can confirm:
[image]

DHLF, 2023-08-11:
You *may* avoid this if you dont use the *daily_production sensor in the energy dashboard setting.
Instead change the energy dashboard setting to *total_production.

May sound strange but HA is clever enough to calculate the difference in values using the TOTAL production as "fake" daily_production.

Also helps if you dont(have/want) the inverter connected to the web..

[image]
[image]

elodur, 2023-08-11:
Actually solved it that way, thank you.

DHLF, 2023-08-11:
Nevermind :)
I missed the date of your question.
Answering a 3 week old question feels like necromancing nowadays ^^
Thank you for not complaining about my necromancy

elodur, 2023-08-11:
Really thank you, I either didn't find the time to post my solution.

kugel-, 2023-08-11:
Wonderful workaround, thank you for necromancing (in my view solving, or at least working around, an open issue is never necromancing, even if years have passed).

hennerdiy, 2023-08-15:
Without a workaround (imo the right way) you have to change the state_class of Daily Production from "total" to "total_increasing" in the inverter_definitions file you use. 
this change ist needed for example in deye_2mppt.yaml or deye_4mppt.yaml.
@StephanJoubert mabye you can change this in every inverters_definitions yaml.

The sofar_hyd3k-6k-es.yaml file is correct.

explaination: 
the state_class "total_increasing" effect a reset to zero by negative values.

tested on deye_2mppt.yaml and deye_4mppt.yaml Solar Inverters.

[image]

neuhaus, 2025-05-18:
I read that the time (clock) of the Deye Microinverter can be set via modbus.
Will that fix the missing reset of the daily production counter?