StephanJoubert/home_assistant_solarman #359: Solarman Stick Logger

helgard999, 2023-07-27:
Hi
Under **Preparation** point 2. "Check the version of the solarman logger. If the serial number starts with 17xxxxxxx, 21xxxxxxx or 40xxxxxxx (protocol V5), the component should work. If not, you may need to try the component for V4 of the protocol mentioned above."

My problem/question: We recently installed two brand new Deye 5kw hybrid inverters, with serial numbers: 
2774842373
2774684439

The loggers are on WiFi, verified IP addresses, can ping them. 
On the Solarman integration followed the instructions, but I do not get any values back.

> **Sensors**
> AC Temperature Unknown 
> Alert                    Unknown
> Battery Current Unknown

Using the `deye_hybrid.yam`l setup

The entitie IDs looks like this:
```
sensor.deye1_communication_board_version_no
sensor.deye2_communication_board_version_no
```
 
The setup was working on different site, with other IP address and other Stick Loggers. Deleted the Integration entries, and recreated two new ones with the new IP addresses for the new site.

Any Ideas what could be wrong? How would I debug this? Is the protocol compatible, I figured newer should be on new protocol.

Regards
Helgard

ondfrukt, 2023-09-16:
I havning this problem to.
My brand new DEYE SUN-15K-G05 inverter has a serial number that starts with 27xxxxxxx. Folowed the instructions and hav tried all setup-files but I cant get any values out from the sensors. All are "Unknown"

wandi34, 2024-03-14:
Same problem here with SN `22xxxxxxx`

goxiplattler, 2024-04-29:
Same problem here with SN 27xxxxxxxx- What can I do?