StephanJoubert/home_assistant_solarman #361: Suddenly no more connection

mcasteels, 2023-08-01:
Currently no more information from the Solarman integration.  I can connect to the inverter over IP, nothing was changed.  Rebooted the HA, rebooted the network, unfortunately no more updates (after the reboot, entities stay on unknown.  Current version of HA: 2023.7.1, latest version of Solarman.

What I can find in the log:

This error originated from a custom integration.

Logger: custom_components.solarman.solarman
Source: custom_components/solarman/solarman.py:200
Integration: Solarman (documentation, issues) 
First occurred: 14:16:04 (12 occurrences) 
Last logged: 14:27:04

Querying registers [512 - 581] failed, aborting.

Turbotrent, 2023-08-01:
I have exactly the same problem. Everything was working fine until yesterday. I have removed the integration and set it up again, but it didn’t help, the sensors remain empty and the status is disconnected, even though the inverter is up and running, and I can connect to it by using it’s ip address.

PV-Joe, 2023-08-03:
No problem with Solarman for Home Assistant 2023.8.0. Everything is running smooth.

mcasteels, 2023-08-04:
Was able to upgrade HA but still the same issue.  This seems not to be related to HA but to or this integration or to a possible a remote update of the Sofar?

PV-Joe, 2023-08-04:
Did you try reinstalling Solarman for Home Assistant and check if the sensors are there or even doubled? I remember having the issue of new sensors being created after updating a while ago.

CDNJsFiles, 2023-08-05:
Nothing work now :(

nejib1, 2023-08-05:
Unfortunately, nothing works since yesterday :(

jefvlamings, 2023-08-06:
I'm experiencing the same issue. I assume Sofar has pushed an update which changed the register addresses. If this is the case an update of the configuration files should fix it. 

I'm checking if the original python module allows me to scan the registers. No luck so far.

This library can be used to run a scan: https://github.com/austinmroczek/modbus_mapper.
Currently trying to make it work but having issues with loading dependencies.

jefvlamings, 2023-08-07:
I noticed my datalogger went offline and isn't reporting to solarman server anymore. 

- Mobile app has no data anymore and mentions datalogger (LSE-3) is offline
- Leds on datalogger indicate miscommunication between the inverter and the datalogger 
  - NET = on
  - COM = off
  - SER = blinking
  
 This would explain why the register scan showed no result whatsoever.

 Anyone having this issue too?

mcasteels, 2023-08-07:
> I noticed my datalogger went offline and isn't reporting to solarman server anymore.
>
> * Mobile app has no data anymore and mentions datalogger (LSE-3) is offline
> * Leds on datalogger indicate [...]

This is not the issue for my problem.  Also my data logger is still communicating with the SOLARMAN servers.  Reboot your inverter?

jefvlamings, 2023-08-08:
Yes rebooting solved that specific issue. I might try finding the right registers again later this week.

danielmessem, 2023-08-09:
same this side
This error originated from a custom integration.
Logger: custom_components.solarman.solarman
Source: custom_components/solarman/solarman.py:173
Integration: Solarman ([documentation](https://github.com/StephanJoubert/home_assistant_solarman/blob/main/README.md), [issues](https://github.com/StephanJoubert/home_assistant_solarman/issues))
First occurred: 12:09:43 PM (10 occurrences)
Last logged: 12:14:13 PM

mcasteels, 2023-08-09:
> same this side This error originated from a custom integration. Logger: custom_components.solarman.solarman Source: custom_components/solarman/solarman.py:173 Integration: Solarman ([documentation](https://github.com/StephanJoubert/home_assistant_solarman/blob/main/README.md), [issues](https://github.com/StephanJoubert/home_assistant_solarman/issues)) First occurred: 12:09:43 PM (10 occurrences) Last logged: 12:14:13 PM
> 
> Did a factory reset on the logger, changed config in solarman to the new logger IP, restarted home assistant and it works now. I redownloaded the repo before that, not sure that made a difference Phew, almost had a full battery without HA knowing to use the excess to heat the geysers more

Sounds weird that a factory reset would resolve the issue, same concerning the IP.  Even if I wools do the reset, the device will always get the same IP because of the MAC reservation on my router.

Turbotrent, 2023-08-16:
After some investigation, I have the impression that the people that lost connection are all using the LAN logger, not the wifi logger. Is that correct? In that case, it seems to be a problem with the LAN module and the solarman integration. The replacement of the LAN module with a Wifi module could solve the problem, but I would like to avoid that of course.

sanderbaks, 2023-08-16:
I used to have the lan logger. Never got it to work. The wifi is working instantly. Placed a wifi accespoint next to it.(PoE) So still connecting through lan effectively. Sander

mcasteels, 2023-08-16:
> After some investigation, I have the impression that the people that lost connection are all using the LAN logger, not the wifi logger. Is that correct? In that case, it seems to be a problem with [...]

I can confirm I have a LAN logger.  It’s connected with a network cable.  And I used the WiFi Sofar setup and that worked fine until 2 weeks ago.

mcasteels, 2023-08-16:
> After some investigation, I have the impression that the people that lost connection are all using the LAN logger, not the wifi logger. Is that correct? In that case, it seems to be a problem with [...]

Would be nice to know the difference.  And if I can remember the LAN could also be used to connect trough WiFi.

danielmessem, 2023-08-20:
> After some investigation, I have the impression that the people that lost connection are all using the LAN logger, not the wifi logger. Is that correct? In that case, it seems to be a problem with [...]

I am using the WiFi logger, no issues since I did the factory reset though

mcasteels, 2023-08-20:
> >
> 
> I am using the WiFi logger, no issues since I did the factory reset though 

I have the LAN module but it also puts out a wifi signal if I'm correct.
What does the factory reset do and what do you need to do to get it up and running again?  I hate factory resets ;-)

danielmessem, 2023-08-20:
> > > 
> > 
> > 
> > I am using the WiFi logger, no issues since I did the factory reset though
> 
> I have the LAN module but it also puts out a wifi signal if I'm correct. What does the factory reset do and what do you need to do to get it up and running again? I hate factory resets ;-)

Not sure if it's the same for both modules, but there is a button you hold in, I was trying to do the basic reset but ended up holding it in and did the fictory reset. To set it up after that I had to connect to it's WiFi hotspot and add in my home WiFi details using the solarman smart app, not too painful.

Turbotrent, 2023-08-30:
Did somebody already find a solution in the meantime?

mcasteels, 2023-08-30:
I’m also waiting.  Did a reset, power of and on, nothing.  Solarman Smart app works perfectly.  And from what I can read only people with a wired network have issues 😓

Turbotrent, 2023-08-30:
It seems that the people in this topic are experiencing the same issue:

https://github.com/StephanJoubert/home_assistant_solarman/issues/336

Unfortunately, the final solution seems to be to replace the LAN logger by the wifi logger...

mcasteels, 2023-09-02:
> It seems that the people in this topic are experiencing the same issue:
> 
> #336
> 
> Unfortunately, the final solution seems to be to replace the LAN logger by the wifi logger...

I don’t mind buying that wifi logger, however my supplier verified the S/N prior to shipping and they all start with 23xxxxxxx and from what I read here it should start with 17xxxxxxx, 21xxxxxxx of 40xxxxxxx

niclas911, 2023-09-07:
Hier auch das Problem mit dem deye sun 600 Firmware VersionMW3_16U_5406_2.27.

Verbindungsaufbau mit Version 1.50 trotz geänderter .yaml nicht möglich. Mit Version 1.43 keine Probleme.

mcasteels, 2023-09-08:
Changed the LAN logger to the WIFI logger and all works!  Easy to do, just need to find a supplier and make certain it starts with the correct S/N.

Turbotrent, 2023-09-08:
Can you tell me which one you ordered, and where you ordered it?

mcasteels, 2023-09-08:
Solarman LSW-3.  Found a seller in the Netherlands.  Doubt he will ship to other countries than Netherlands or Belgium.

https://www.spqpr.com/artikel/478/wifi-data-logger-lsw-3-db-9.html

setup was easy.  Removed the power from the inverter.  Removed the LAN module, installed
The WIFI module.  Powers all on.  Used the Solarman app to add a logger (scan QR) code.  Follow instructions and before adding it asks if you want to set it as extra or swap with an existing one.  Took me roughly 15 minutes.

afterwards reconfigured the Solarman integration and it worked directly  I also updated to the new version afterwards, 1.5