StephanJoubert/home_assistant_solarman #367: Querying [3 - 128] failed with exception [IllegalDataValueError]

lenwar, 2023-08-13:
Hi,

I have a few microinverters which used to work with the 2mptt yaml file.
Since a few days they no longer work and provide the following error:

`2023-08-13 17:10:37.318 DEBUG (SyncWorker_2) [pysolarmanv5.pysolarmanv5] Socket setup completed... <socket.socket fd=27, family=2, type=1, proto=6, laddr=('192.168.xxx.yyy', 60194), raddr=('192.168.yyy.zzz', 8899)>
2023-08-13 17:10:37.318 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 128] failed with exception [IllegalDataValueError]
2023-08-13 17:10:37.321 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 128] failed, [5] retry attempts left
2023-08-13 17:10:37.323 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 128] failed with exception [IllegalDataValueError]
2023-08-13 17:10:37.325 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 128] failed, [4] retry attempts left
2023-08-13 17:10:37.327 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 128] failed with exception [IllegalDataValueError]
2023-08-13 17:10:37.329 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 128] failed, [3] retry attempts left
2023-08-13 17:10:37.331 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 128] failed with exception [IllegalDataValueError]
2023-08-13 17:10:37.332 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 128] failed, [2] retry attempts left
2023-08-13 17:10:37.333 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 128] failed with exception [IllegalDataValueError]
2023-08-13 17:10:37.336 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 128] failed, [1] retry attempts left
2023-08-13 17:10:37.338 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 128] failed with exception [IllegalDataValueError]
2023-08-13 17:10:37.339 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 128] failed, [0] retry attempts left
2023-08-13 17:10:37.340 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying registers [3 - 128] failed, aborting `

I really haven’t a clue where to start looking.
I updated to the latest ‘Main’. (I was back an x-number of versions of the ‘Main’. I don’t know which moment did work)

b-reich, 2023-08-15:
same error here

UndIch, 2023-08-20:
Did anyone try to use the main and just manually replace the deye_mppt2 with the old working one? Little bit annoying that one can only try to do sth with daylight …

lenwar, 2023-08-21:
I don’t have the ‘old’ mppt2 yaml file, but the current/live one appears to work for me. It still throws the errors, but ‘only’ a few times per day

stvogel, 2023-08-30:
> 

I can't find deye_2mppt.yaml in v1.4.3
which yaml did you use?

lenwar, 2023-08-30:
The one from the main branch.

custom_components/solarman/inverter_definitions/deye_2mppt.yaml

stvogel, 2023-08-31:
Ok, I used that one, but got IllegalDataValueError with Deye SUN-M80G3-EU-Q0
There are so many values and configurations in deye_2mppt.yaml ... 
 
Did someone fix that error for this microinverter?

UndIch, 2023-09-01:
I just used the 1.4.3 and the deye_4mptt with my 800G3 so this might work for the M80 as well?

oxynatOr, 2023-09-03:
the problem is, both files 2mppt and 4mppt have a ending value of 0x0080.

In the PDF of the inverter the say:
0x0001~ 0x007D

my guess is:

requests:
  - start: 0x0003
    end:  0x007D
    mb_functioncode: 0x03

may fix it. you can give it a try

oxynatOr, 2023-09-04:
need more debug later.

`2023-09-04 09:05:13.870 DEBUG (SyncWorker_11) [custom_components.solarman.solarman] frame_len does not match payload_len.`
this error will lead to

`2023-09-04 09:05:13.870 WARNING (SyncWorker_11) [custom_components.solarman.solarman] Querying 3 - 112 - failed with exception :V5FrameError: V5 frame contains invalid start or end values`

at least, its my guess.

btw, i have a DeyeSun800-G3-EU since a week, so i'm pretty new to this. 

`
2023-09-04 09:05:13.561 WARNING (SyncWorker_11) [custom_components.solarman.solarman] Querying 3 - 112 - failed with exception :V5FrameError: V5 frame contains invalid start or end values
2023-09-04 09:05:13.562 INFO (SyncWorker_11) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192.168.167.99:8899
2023-09-04 09:05:13.859 WARNING (SyncWorker_11) [custom_components.solarman.solarman] Querying [3 - 112] failed, [1] retry attempts left
2023-09-04 09:05:13.861 INFO (SyncWorker_11) [custom_components.solarman.solarman] Connecting to solarman data logger 192.168.167.99:8899
2023-09-04 09:05:13.866 DEBUG (SyncWorker_11) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=25, family=2, type=1, proto=6, laddr=('10.0.100.10', 58264), raddr=('192.168.167.99', 8899)>
2023-09-04 09:05:13.866 DEBUG (SyncWorker_11) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 9d 00 d0 99 4f e9 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 6e 34 26 7b 15
2023-09-04 09:05:13.870 DEBUG (SyncWorker_11) [custom_components.solarman.solarman] RECD: 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 36 5f 32 2e 30 36 0d 0a 0d 0a
2023-09-04 09:05:13.870 DEBUG (SyncWorker_11) [custom_components.solarman.solarman] frame_len does not match payload_len.
2023-09-04 09:05:13.870 WARNING (SyncWorker_11) [custom_components.solarman.solarman] Querying 3 - 112 - failed with exception :V5FrameError: V5 frame contains invalid start or end values
2023-09-04 09:05:13.872 INFO (SyncWorker_11) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192.168.167.99:8899
2023-09-04 09:05:14.168 WARNING (SyncWorker_11) [custom_components.solarman.solarman] Querying [3 - 112] failed, [0] retry attempts left
2023-09-04 09:05:14.170 WARNING (SyncWorker_11) [custom_components.solarman.solarman] Querying registers [3 - 112] failed, aborting.
2023-09-04 09:05:14.173 DEBUG (SyncWorker_11) [custom_components.solarman.sensor] No value recorded for PV1 Voltage
2023-09-04 09:05:14.175 DEBUG (SyncWorker_5) [custom_components.solarman.sensor] No value recorded for PV2 Voltage
2023-09-04 09:05:14.178 DEBUG (SyncWorker_9) [custom_components.solarman.sensor] No value recorded for PV1 Current
2023-09-04 09:05:14.183 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for PV2 Current
`

oxynatOr, 2023-09-06:
``
`requests:`
`  - start: 0x0001`
`    end: 0x007D`
`    mb_functioncode: 0x03`
``
``
`2023-09-06 08:04:20.833 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Starting to query for [1] ranges...`
`2023-09-06 08:04:20.833 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [1 - 125]...`
`2023-09-06 08:04:20.833 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 07 00 d0 99 4f e9 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 01 00 7d d4 2b 97 15`
`2023-09-06 08:04:27.212 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] RECD: a5 0d 01 10 15 07 34 d0 99 4f e9 02 01 85 96 07 00 0c 11 00 00 e5 7f f0 64 01 03 fa 01 00 02 01 32 33 30 34 32 33 33 32 35 39 00 01 00 00 12 0c 07 00 01 02 01 17 12 18 00 00 1f 40 00 00 02 01 00 4b 00 00 00 3c 17 09 06 08 04 13 00 00 00 00 0a be 07 08 14 50 12 8e 00 00 00 00 00 00 13 9c 00 2c 00 00 00 00 00 00 00 00 00 64 00 00 00 00 00 01 00 00 00 00 00 01 00 00 00 00 00 01 ff d5 00 00 00 0a 00 02 00 00 00 00 00 00 00 00 00 00 00 04 00 00 00 00 00 00 01 27 00 00 00 00 00 00 00 00 00 00 00 90 00 00 00 90 00 00 09 4c 00 00 00 00 00 01 00 00 00 00 13 88 00 00 00 00 00 00 00 00 00 00 00 00 01 86 00 00 00 00 00 00 0c 1c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 be 00 02 01 4c 00 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 4b 30 20 15`
`2023-09-06 08:04:27.213 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [1 - 125] succeeded`
`2023-09-06 08:04:27.213 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] All queries succeeded, exposing updated values.`

With Deye Sun800 G3 EU and [FW 2.06](http://47.254.36.66/0_D0002_18/MW3_16U_5406_2.06.bin)

My Serial starts with 39xxx 

hf!

lenwar, 2023-09-06:
Hi,

Just an update.
1.5.0 has just been released.
When I use 1.5.0 (with the 2mppt yaml file) I get the errors mentioned above.
I downgraded to 1.4.3 and use the same 2mppt yaml file (copied in from 1.5.0) and that works fine for me. (it does give an error every now and then, but it does work)

monoboy-hh, 2023-09-06:
TL;DR
Using DEYE 800 G3 fw 1.57 (MW3_16U_5406_1.57)
After upgrade to solarman 1.5.0 errors were thrown (IllegalDataValueError)
Fixed by downgrade to 1.4.3 incl. yaml configs

long
With both HA 2023-07-xy and HA 2023-08-xy the solarman integration was working fine in version 1.4.3
After upgrading to 1.5.0 incl. yaml configs the integration refused to connect to the inverter.
After replacing the bundled yaml configs with my previous one from a backup the integration was still not working.
It was not until I downgraded the integration to 1.4.3 (and using my backup config from 1.4.3 + HA restarts) it finally connected to the inverter as befroe.
No LAN / IP changes.
Connected via WiFi

oxynatOr, 2023-09-06:
they have merged 2 projects into one, if i read the merge-request correctly.

for a while i get the results:
```
2023-09-06 09:53:41.342 DEBUG (SyncWorker_11) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 a8 00 d0 99 4f e9 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 01 00 7d d4 2b 38 15

2023-09-06 09:53:41.349 DEBUG (SyncWorker_11) [custom_components.solarman.solarman] RECD: 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 36 5f 32 2e 30 36 0d 0a 0d 0a
2023-09-06 09:53:41.350 DEBUG (SyncWorker_11) [custom_components.solarman.solarman] frame_len does not match payload_len.
```

the payload  need to start with `a5` only then pysolarmanv5 will handle the result. 

here is a full example:

```
2023-09-06 10:04:41.007 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] Starting to query for [1] ranges...
2023-09-06 10:04:41.007 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] Querying [1 - 125]...
2023-09-06 10:04:41.007 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] Send Request: 1 with length 125
2023-09-06 10:04:41.007 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 57 00 d0 99 4f e9 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 01 00 7d d4 2b e7 15
2023-09-06 10:04:41.334 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] RECD: a5 0d 01 10 15 57 33 d0 99 4f e9 02 01 8f b2 07 00 e4 02 00 00 09 80 f0 64 01 03 fa 01 00 02 01 32 33 30 34 32 33 33 32 35 39 00 01 00 00 12 0c 07 00 01 02 01 17 12 18 00 00 1f 40 00 00 02 01 00 4b 00 00 00 3c 17 09 06 0a 03 0e 00 00 00 00 0a be 07 08 14 50 12 8e 00 00 00 00 00 00 13 9c 00 2c 00 00 00 00 00 00 00 00 00 64 00 00 00 00 00 01 00 00 00 00 00 01 00 00 00 00 00 01 ff d5 00 00 00 0a 00 02 00 00 00 00 00 00 00 00 00 00 00 04 00 08 00 00 00 00 01 2f 00 00 00 04 00 04 00 00 00 00 00 94 00 00 00 94 00 00 09 4c 00 00 00 00 00 18 00 00 00 00 13 88 00 00 00 00 00 00 00 00 00 00 00 00 16 a8 00 00 00 00 00 00 15 5e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 dd 00 3d 01 e0 00 3b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f8 11 e6 15
2023-09-06 10:04:41.337 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] Response: [256, 513, 12851, 12340, 12851, 13106, 13625, 1, 0, 4620, 1792, 258, 279, 4632, 0, 8000, 0, 513, 75, 0, 60, 5897, 1546, 782, 0, 0, 2750, 1800, 5200, 4750, 0, 0, 0, 5020, 44, 0, 0, 0, 0, 100, 0, 0, 1, 0, 0, 1, 0, 0, 1, 65493, 0, 10, 2, 0, 0, 0, 0, 0, 4, 8, 0, 0, 303, 0, 4, 4, 0, 0, 148, 0, 148, 0, 2380, 0, 0, 24, 0, 0, 5000, 0, 0, 0, 0, 0, 0, 5800, 0, 0, 0, 5470, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 477, 61, 480, 59, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2023-09-06 10:04:41.338 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] Querying [1 - 125] succeeded
2023-09-06 10:04:41.338 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] All queries succeeded, exposing updated values.
2023-09-06 10:05:09.715 DEBUG (Thread-15 (_data_receiver)) [custom_components.solarman.solarman] [POLL-DISCARDED] RECD: 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 36 5f 32 2e 30 36 0d 0a 0d 0a
2023-09-06 10:05:10.132 DEBUG (Thread-15 (_data_receiver)) [custom_components.solarman.solarman] [POLL-DISCARDED] RECD: a5 0f 01 10 15 00 35 d0 99 4f e9 02 01 ac b2 07 00 00 03 00 00 09 80 f0 64 01 03 fc 32 33 30 34 32 33 33 32 35 39 00 01 00 00 12 0c 07 00 01 02 01 17 12 18 00 00 1f 40 00 00 02 01 00 4b 00 00 00 3c 17 09 06 0a 03 0e 00 00 00 00 0a be 07 08 14 50 12 8e 00 00 00 00 00 00 13 9c 00 2c 00 00 00 00 00 00 00 00 00 64 00 00 00 00 00 01 00 00 00 00 00 01 00 00 00 00 00 01 ff d5 00 00 00 0a 00 02 00 00 00 00 00 00 00 00 00 00 00 04 00 08 00 00 00 00 01 2f 00 00 00 04 00 04 00 00 00 00 00 94 00 00 00 94 00 00 09 4c 00 00 00 00 00 18 00 00 00 00 13 88 00 00 00 00 00 00 00 00 00 00 00 00 16 a8 00 00 00 00 00 00 15 5e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 dd 00 3d 01 e0 00 3b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 b6 7c 15
2023-09-06 10:05:10.564 DEBUG (Thread-15 (_data_receiver)) [custom_components.solarman.solarman] [POLL-DISCARDED] RECD: a5 0f 01 10 15 00 36 d0 99 4f e9 02 01 ac b2 07 00 01 03 00 00 09 80 f0 64 01 03 fc 32 33 30 34 32 33 33 32 35 39 00 01 00 00 12 0c 07 00 01 02 01 17 12 18 00 00 1f 40 00 00 02 01 00 4b 00 00 00 3c 17 09 06 0a 03 0e 00 00 00 00 0a be 07 08 14 50 12 8e 00 00 00 00 00 00 13 9c 00 2c 00 00 00 00 00 00 00 00 00 64 00 00 00 00 00 01 00 00 00 00 00 01 00 00 00 00 00 01 ff d5 00 00 00 0a 00 02 00 00 00 00 00 00 00 00 00 00 00 04 00 08 00 00 00 00 01 2f 00 00 00 04 00 04 00 00 00 00 00 94 00 00 00 94 00 00 09 4c 00 00 00 00 00 18 00 00 00 00 13 88 00 00 00 00 00 00 00 00 00 00 00 00 16 a8 00 00 00 00 00 00 15 5e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 dd 00 3d 01 e0 00 3b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 b6 7e 15
2023-09-06 10:05:41.008 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] Starting to query for [1] ranges...
2023-09-06 10:05:41.008 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125]...
[... cut ...]
Response: [256, 513, 12851, 12340, 12851, 13106, 13625, 1, 0, 4620, 1792, 258, 279, 4632, 0, 8000, 0, 513, 75, 0, 60, 5897, 1546, 782, 0, 0, 2750, 1800, 5200, 4750, 0, 0, 0, 5020, 44, 0, 0, 0, 0, 100, 0, 0, 1, 0, 0, 1, 0, 0, 1, 65493, 0, 10, 2, 0, 0, 0, 0, 0, 4, 8, 0, 0, 303, 0, 4, 4, 0, 0, 148, 0, 148, 0, 2380, 0, 0, 24, 0, 0, 5000, 0, 0, 0, 0, 0, 0, 5800, 0, 0, 0, 5470, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 477, 61, 480, 59, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
```

now i'm restarting HA to get some results when it's false.

stvogel, 2023-09-06:
Great. 
I used the values you provided for my Deye SUN-M80G3-EU-Q0
`requests:
  - start: 0x0001
    end: 0x007D
    mb_functioncode: 0x03`
I rebooted the raspberry ... and voila ... all values are there.
Thanks for diving into this issue.

rosch100, 2023-09-06:
With these changes, it also works for the Deye SUN-M60G3-EU-Q0

oxynatOr, 2023-09-06:
@stvogel @rosch100 

u getting all the values you need, is something missing/wrong displayd?

rosch100, 2023-09-06:
> u getting all the values you need, is something missing/wrong displayd?
All the values seem to make sense.

idl0r, 2023-09-06:
SUN-M80G3-EU-230 here. Works just fine now. Up to now there's only one error still but it look way better than before, even with 1.4.3.
```
Querying [1 - 125] failed with exception [Empty: ]
Querying [1 - 125] failed, [1] retry attempts left
```
Otherwise there's only #349 left but that's no big deal. Thanks so far!

oxynatOr, 2023-09-06:
here is my yaml file:
```
# First version : 27.08.2023
# Microinverter SUNX (DEYE/VESDAS ??? - X because i have no dang clue what shares those registers  :) )
# v0.1  Added all the cool registers from Deye_Modbus.-.-.V118.pdf
#
## Tested with:
### SUN-800-G3-EU     FW: 2.06
########## May work with:
### SUN-M80G3-EU-230  FW: ?.??
### SUN-M60G3-EU-Q0   FW: ?.??
### SUN-M80G3-EU-Q0   FW: ?.??
###
###
requests:
  - start: 0x0001
    end: 0x007D
    mb_functioncode: 0x03

parameters:
  - group: solar
    items:
      - name: "PV1 Voltage"
        class: "voltage"
        state_class: "measurement"
        uom: "V"
        scale: 0.1
        rule: 1
        registers: [0x006D]
        icon: "mdi:solar-power"

      - name: "PV2 Voltage"
        class: "voltage"
        state_class: "measurement"
        uom: "V"
        scale: 0.1
        rule: 1
        registers: [0x006F]
        icon: "mdi:solar-power"

      - name: "PV1 Current"
        class: "current"
        state_class: "measurement"
        uom: "A"
        scale: 0.1
        rule: 1
        registers: [0x006E]
        icon: "mdi:solar-power"

      - name: "PV2 Current"
        class: "current"
        state_class: "measurement"
        uom: "A"
        scale: 0.1
        rule: 1
        registers: [0x0070]
        icon: "mdi:solar-power"

      - name: "Daily Production"
        class: "energy"
        state_class: "total"
        uom: "kWh"
        scale: 0.1
        rule: 1
        registers: [0x003C]
        icon: "mdi:solar-power"

      - name: "Daily Production 1"
        class: "energy"
        state_class: "total"
[... 401 lines cut ...]
        icon: "mdi:information-slab-circle-outline"
        isstr: true
        lookup:
          - key: 0
            value: "None"
          - key: F07
            value: "F07 - DC/DC_Softsart_Fault"

      - name: "Alert"
        class: ""
        state_class: ""
        uom: ""
        scale: 1
        rule: 6
        registers: [0x0075, 0x0076]
        icon: "mdi:message-alert-outline"
        isstr: true
        lookup:
          - key: 0
            value: "None"

```
you will find some hex values, i need to figure out what the values mean. 

would be nice if you can give it a try and give me feedback

oxynatOr, 2023-09-06:
> SUN-M80G3-EU-230 here. Works just fine now. Up to now there's only one error still but it look way better than before, even with 1.4.3.
>
> ```
> Querying [1 - 125] failed with exception [Empty: ] [...]

yes, this is a pain... 

i can post what the PDF says:

实时运行数据区  (Run the data area in real time)
059 (0x003B) 运行状态  (run state) R (read-only) [0,5] (data range) -

0x0000 待机 (Standby)
0x0001 自检 (Self-Test)
0x0002 正常 (Normal)
0x0003 告警 (Alarm)
0x0004 故障 (Fault)

for me, the value is static. its always 4.

idl0r, 2023-09-07:
> you will find some hex values, i need to figure out what the values mean.
> 
> would be nice if you can give it a try and give me feedback

So far it seems to look good. Only one entity is missing now, which is "solarman_update_time"

oxynatOr, 2023-09-07:
> > you will find some hex values, i need to figure out what the values mean.
> > would be nice if you can give it a try and give me feedback
> 
> So far it seems to look good. Only one entity is missing now, which is "solarman_update_time". Perhaps a separate inverter definition is needed for our ones. It'll probably break other 2mppt's I guess.

okay, i will look it up in the PDF when i'm back home.

niclas911, 2023-09-07:
Hier auch das Problem mit dem deye sun 600 Firmware VersionMW3_16U_5406_2.27.

Verbindungsaufbau mit Version 1.50 trotz geänderter .yaml nicht möglich. Mit Version 1.43 keine Probleme.

idl0r, 2023-09-07:
> Hier auch das Problem mit dem deye sun 600 Firmware VersionMW3_16U_5406_2.27.
> 
> Verbindungsaufbau mit Version 1.50 trotz geänderter .yaml nicht möglich. Mit Version 1.43 keine Probleme.

Did you try https://github.com/StephanJoubert/home_assistant_solarman/issues/367#issuecomment-1708839274?

wunderbaum, 2023-09-07:
@niclas911 
Change the first lines of /config/custom_components/solarman/inverter_definitions/deye_2mppt.yaml:

```
# First version : 22.2.2023
# Microinverter SUN600G3 (DEYE/VESDAS)
# 2x MPPT, 2x inverter
# 1x Logger, 2x Module, 

requests:
  - start: 0x0001
    end:  0x007D
    mb_functioncode: 0x03

```
Restart home-assistant. Works for me.

niclas911, 2023-09-07:
> > Hier auch das Problem mit dem deye sun 600 Firmware VersionMW3_16U_5406_2.27.
> > Verbindungsaufbau mit Version 1.50 trotz geänderter .yaml nicht möglich. Mit Version 1.43 keine Probleme.
> 
> [Hast du #367 (Kommentar)](https://github.com/StephanJoubert/home_assistant_solarman/issues/367#issuecomment-1708839274) ausprobiert ?

Ja, leider keinen Erfolg (Solarmann status_connection disconnected).

niclas911, 2023-09-07:
> @niclas911 Ändern Sie die ersten Zeilen von /config/custom_components/solarman/inverter_definitions/deye_2mppt.yaml:
> 
> ```
> # First version : 22.2.2023
> # Microinverter SUN600G3 (DEYE/VESDAS)
> # 2x MPPT, 2x inverter
> # 1x Logger, 2x Module, 
> 
> requests:
>   - start: 0x0001
>     end:  0x007D
>     mb_functioncode: 0x03
> ```
> 
> Starten Sie Home-Assistant neu. Funktioniert bei mir.

> @niclas911 Ändern Sie die ersten Zeilen von /config/custom_components/solarman/inverter_definitions/deye_2mppt.yaml:
> 
> ```
> # First version : 22.2.2023
> # Microinverter SUN600G3 (DEYE/VESDAS)
> # 2x MPPT, 2x inverter
> # 1x Logger, 2x Module, 
> 
> requests:
>   - start: 0x0001
>     end:  0x007D
>     mb_functioncode: 0x03
> ```
> 
> Starten Sie Home-Assistant neu. Funktioniert bei mir.

Leider ohne Erfolg. Status ist weiterhin disconnected.

UndIch, 2023-09-07:
So I tried it with my Deye Sun 800. It does work every now and then after the modification of the 2mptt like mentioned from @niclas911. But then it stoppea working again and pushes a “entity is non-numeric” despite the SOLARMAN Smart app showing devices still online … will report back how the behavior is tomorrow. Right now it’s quite unstable.

niclas911, 2023-09-07:
Ich bin mir auch nicht ganz sicher, ob die Problematik mit der Firmware VersionMW3_16U_5406_2.27 zusammen hängt, die in Verbindung mit dem Relais SUN-MI-RELAY-01 heruntergeladen wurde. Dieses externe Relais musste hier in Deutschland nachgerüstet werden (fehlender NA-Schutz). Seid dem neuen Update sind auch ein paar Werte in der Solarman App falsch wie z.b. die Wechselrichter Temperatur, diese soll -10 Grad betragen. Sehr merkwürdig.

oxynatOr, 2023-09-07:
> Ich bin mir auch nicht ganz sicher, ob die Problematik mit der Firmware VersionMW3_16U_5406_2.27 zusammen hängt, die in Verbindung mit dem Relais SUN-MI-RELAY-01 heruntergeladen wurde. Dieses externe Relais musste hier in Deutschland nachgerüstet werden (fehlender NA-Schutz). Seid dem neuen Update sind auch ein paar Werte in der Solarman App falsch wie z.b. die Wechselrichter Temperatur, diese soll -10 Grad betragen. Sehr merkwürdig.
> 
> @wunderbaum wurden bei dir sofort nach dem Neustart Werte angezeigt bei den Entitäten? Habe es testhalber 2 Stunden laufen gelassen ohne Erfolg.

Könntest du einen Auszug aus der Log schicken, am besten mit Debug mode on. 

Die FW `MW3_16U_5406_2.27` liegt leider nicht auf dem Deye server, so das ich es testen könnte.

wunderbaum, 2023-09-07:
Yes, ja. 😃
Bei mir lief es nach der Umstellung gleich wieder. Kann aber nicht sagen welche Firmware gerade drauf ist, im dunklen ist das ja offline.
Man muss natürlich auch in der Konfiguration dann nochmal deye2mpt einstellen.
Ich hatte allerdings gefühlt jetzt mehr Entitäten, sobald die Sonne aufgeht wissen wir mehr.

idl0r, 2023-09-07:
I'm glad I don't have that thing connected yet. My 80G3 logger unit is running MW3_16U_5406_1.57 and with the modified yank it works fine for about 24 hours now.

UndIch, 2023-09-07:
No I don’t think it’s related to the relays issue. Haven’t installed mine yet (just received it today) and it’s not working like the 1.4.3 did-

oxynatOr, 2023-09-08:
you guys use one or more Home-Assistant to read the data? 

because we use 2 Home-Assistants and when i add the 2nd one, it will stop working, and the first Home-Assistant gets all the result. ( even requests from 2nd Home-Assistant... what makes no sens at all)

idl0r, 2023-09-08:
Only one here.

Smeily84, 2023-09-08:
WR: SUN1600G3-EU-230 (4xMPPT)
Firmware VersionMW3_16U_5406_1.53

Es läuft endlich, danke der Yaml-Datei vom Post 367.
Mir fehlen leider noch die Eingänge von 3 und 4.
Hat da jemand noch eine Übersicht welche register dafür gibt?

Nur einen HA ist am start.

oxynatOr, 2023-09-08:
> WR: SUN1600G3-EU-230 (4xMPPT) Firmware VersionMW3_16U_5406_1.53
> 
> Es läuft endlich, danke der Yaml-Datei vom Post 367. Mir fehlen leider noch die Eingänge von 3 und 4. Hat da jemand noch eine Übersicht welche register dafür gibt?
> 
> Nur einen HA ist am start.

meinst du mit 3 und 4 die von PV-3 und PV-4 ? 

hier ist die PDF die ich nutze [Deye_Modbus.-.-.V118.pdf](https://github.com/StephanJoubert/home_assistant_solarman/files/12556241/Deye_Modbus.-.-.V118.pdf)

niclas911, 2023-09-08:
@oxynatOr ich hoffe das reicht ihnen.

2023-09-08 12:13:49.406 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] Starting to query for [1] ranges...
2023-09-08 12:13:49.406 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125]...
2023-09-08 12:13:49.407 INFO (SyncWorker_9) [custom_components.solarman.solarman] Connecting to solarman data logger 192............
2023-09-08 12:13:49.419 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=63, family=2, type=1, proto=6, laddr=('192............', 33168), raddr=('192............', 8899)>
2023-09-08 12:13:49.419 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 17 00 b1 56 8c f7 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 01 00 7d d4 2b 90 15
2023-09-08 12:13:49.527 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] RECD: 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 36 5f 32 2e 32 37 0d 0a 0d 0a
2023-09-08 12:13:49.527 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] frame_len does not match payload_len.
2023-09-08 12:13:49.527 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid start or end values]
2023-09-08 12:13:49.530 INFO (SyncWorker_9) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192............:8899
2023-09-08 12:13:49.626 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125] failed, [1] retry attempts left
2023-09-08 12:13:49.631 INFO (SyncWorker_9) [custom_components.solarman.solarman] Connecting to solarman data logger 192............:8899
2023-09-08 12:13:49.639 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=63, family=2, type=1, proto=6, laddr=('192.168.178.51', 33170), raddr=('192............', 8899)>
2023-09-08 12:13:49.639 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 82 00 b1 56 8c f7 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 01 00 7d d4 2b fb 15
2023-09-08 12:13:49.743 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] RECD: 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 36 5f 32 2e 32 37 0d 0a 0d 0a
2023-09-08 12:13:49.743 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] frame_len does not match payload_len.
2023-09-08 12:13:49.743 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid start or end values]
2023-09-08 12:13:49.744 INFO (SyncWorker_9) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192............:8899
2023-09-08 12:13:50.208 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125] failed, [0] retry attempts left
2023-09-08 12:13:50.212 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying registers [1 - 125] failed, aborting.
2023-09-08 12:13:50.216 DEBUG (SyncWorker_9) [custom_components.solarman.sensor] No value recorded for PV1 Voltage
2023-09-08 12:13:50.218 DEBUG (SyncWorker_8) [custom_components.solarman.sensor] No value recorded for PV2 Voltage
2023-09-08 12:13:50.221 DEBUG (SyncWorker_2) [custom_components.solarman.sensor] No value recorded for PV1 Current
2023-09-08 12:13:50.222 DEBUG (SyncWorker_4) [custom_components.solarman.sensor] No value recorded for PV2 Current
2023-09-08 12:13:50.224 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for Daily Production
2023-09-08 12:13:50.226 DEBUG (SyncWorker_6) [custom_components.solarman.sensor] No value recorded for Total Production
2023-09-08 12:13:50.228 DEBUG (SyncWorker_8) [custom_components.solarman.sensor] No value recorded for Grid Voltage L-L(A)
2023-09-08 12:13:50.230 DEBUG (SyncWorker_1) [custom_components.solarman.sensor] No value recorded for Grid Voltage L-L(B))
2023-09-08 12:13:50.231 DEBUG (SyncWorker_4) [custom_components.solarman.sensor] No value recorded for Grid Voltage L-L(C)
2023-09-08 12:13:50.234 DEBUG (SyncWorker_5) [custom_components.solarman.sensor] No value recorded for Grid Current A
2023-09-08 12:13:50.236 DEBUG (SyncWorker_9) [custom_components.solarman.sensor] No value recorded for Grid Current B
2023-09-08 12:13:50.238 DEBUG (SyncWorker_7) [custom_components.solarman.sensor] No value recorded for Grid Current C
2023-09-08 12:13:50.239 DEBUG (SyncWorker_2) [custom_components.solarman.sensor] No value recorded for Grid Frequency
2023-09-08 12:13:50.241 DEBUG (SyncWorker_4) [custom_components.solarman.sensor] No value recorded for Running Status
2023-09-08 12:13:50.242 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for Total Output AC Power
2023-09-08 12:13:50.244 DEBUG (SyncWorker_6) [custom_components.solarman.sensor] No value recorded for Input Active Power
2023-09-08 12:13:50.245 DEBUG (SyncWorker_8) [custom_components.solarman.sensor] No value recorded for Output Apparent Power
2023-09-08 12:13:50.247 DEBUG (SyncWorker_1) [custom_components.solarman.sensor] No value recorded for Output Active Power
2023-09-08 12:13:50.248 DEBUG (SyncWorker_3) [custom_components.solarman.sensor] No value recorded for Output Reactive Power
2023-09-08 12:13:50.249 DEBUG (SyncWorker_5) [custom_components.solarman.sensor] No value recorded for Inverter ID

oxynatOr, 2023-09-08:
> @oxynatOr ich hoffe das reicht ihnen.
> 
> 2023-09-08 12:13:49.406 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] Starting to query for [1] ranges... 2023-09-08 12:13:49.406 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125]... 2023-09-08 12:13:49.407 INFO (SyncWorker_9) [custom_components.solarman.solarman] Connecting to solarman data logger 192............ 2023-09-08 12:13:49.419 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=63, family=2, type=1, proto=6, laddr=('192............', 33168), raddr=('192............', 8899)> 2023-09-08 12:13:49.419 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 17 00 b1 56 8c f7 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 01 00 7d d4 2b 90 15 2023-09-08 12:13:49.527 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] RECD: 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 36 5f 32 2e 32 37 0d 0a 0d 0a 2023-09-08 12:13:49.527 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] frame_len does not match payload_len. 2023-09-08 12:13:49.527 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid start or end values] 2023-09-08 12:13:49.530 INFO (SyncWorker_9) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192............:8899 2023-09-08 12:13:49.626 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125] failed, [1] retry attempts left 2023-09-08 12:13:49.631 INFO (SyncWorker_9) [custom_components.solarman.solarman] Connecting to solarman data logger 192............:8899 2023-09-08 12:13:49.639 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=63, family=2, type=1, proto=6, laddr=('192.168.178.51', 33170), raddr=('192............', 8899)> 2023-09-08 12:13:49.639 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 82 00 b1 56 8c f7 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 01 00 7d d4 2b fb 15 2023-09-08 12:13:49.743 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] RECD: 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 36 5f 32 2e 32 37 0d 0a 0d 0a 2023-09-08 12:13:49.743 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] frame_len does not match payload_len. 2023-09-08 12:13:49.743 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid start or end values] 2023-09-08 12:13:49.744 INFO (SyncWorker_9) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192............:8899 2023-09-08 12:13:50.208 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125] failed, [0] retry attempts left 2023-09-08 12:13:50.212 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying registers [1 - 125] failed, aborting. 2023-09-08 12:13:50.216 DEBUG (SyncWorker_9) [custom_components.solarman.sensor] No value recorded for PV1 Voltage 2023-09-08 12:13:50.218 DEBUG (SyncWorker_8) [custom_components.solarman.sensor] No value recorded for PV2 Voltage 2023-09-08 12:13:50.221 DEBUG (SyncWorker_2) [custom_components.solarman.sensor] No value recorded for PV1 Current 2023-09-08 12:13:50.222 DEBUG (SyncWorker_4) [custom_components.solarman.sensor] No value recorded for PV2 Current 2023-09-08 12:13:50.224 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for Daily Production 2023-09-08 12:13:50.226 DEBUG (SyncWorker_6) [custom_components.solarman.sensor] No value recorded for Total Production 2023-09-08 12:13:50.228 DEBUG (SyncWorker_8) [custom_components.solarman.sensor] No value recorded for Grid Voltage L-L(A) 2023-09-08 12:13:50.230 DEBUG (SyncWorker_1) [custom_components.solarman.sensor] No value recorded for Grid Voltage L-L(B)) 2023-09-08 12:13:50.231 DEBUG (SyncWorker_4) [custom_components.solarman.sensor] No value recorded for Grid Voltage L-L(C) 2023-09-08 12:13:50.234 DEBUG (SyncWorker_5) [custom_components.solarman.sensor] No value recorded for Grid Current A 2023-09-08 12:13:50.236 DEBUG (SyncWorker_9) [custom_components.solarman.sensor] No value recorded for Grid Current B 2023-09-08 12:13:50.238 DEBUG (SyncWorker_7) [custom_components.solarman.sensor] No value recorded for Grid Current C 2023-09-08 12:13:50.239 DEBUG (SyncWorker_2) [custom_components.solarman.sensor] No value recorded for Grid Frequency 2023-09-08 12:13:50.241 DEBUG (SyncWorker_4) [custom_components.solarman.sensor] No value recorded for Running Status 2023-09-08 12:13:50.242 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for Total Output AC Power 2023-09-08 12:13:50.244 DEBUG (SyncWorker_6) [custom_components.solarman.sensor] No value recorded for Input Active Power 2023-09-08 12:13:50.245 DEBUG (SyncWorker_8) [custom_components.solarman.sensor] No value recorded for Output Apparent Power 2023-09-08 12:13:50.247 DEBUG (SyncWorker_1) [custom_components.solarman.sensor] No value recorded for Output Active Power 2023-09-08 12:13:50.248 DEBUG (SyncWorker_3) [custom_components.solarman.sensor] No value recorded for Output Reactive Power 2023-09-08 12:13:50.249 DEBUG (SyncWorker_5) [custom_components.solarman.sensor] No value recorded for Inverter ID
> 
> Getestet mit requests:
> 
[... cut ...]
Ich habe genau die gleichen fehler, wenn ich einen Zweiten-HomeAssistant  mit dem DataLogger abgreifen möchte. Meine Vermutung ist, das du nur einen HomeAssistant laufen hast?

Der Einwand mit der FW, kann ich ggf später / morgen prüfen.  
Da man die 2.27 nicht downloaden kann, vom FW-Server. Muss ich auf den Kollegen warten, der einen deye M80 mit diesem Relais bekommt. 

Benutzt du die _string Datei aus einem bestimmten Grund?  Dies ist mehr informativ für mich, da ich nicht wirklich weiß wieso das in 4 files geteilt wurde.. Hybrid, 2 und 4 kann ich mir noch erklären.

niclas911, 2023-09-08:
@oxynatOr ich benutze nur 1x Home Assistant System genau.

Die String hat damals von Anfang an funktioniert und tut es momentan in der 1.43 immer noch. Deshalb habe ich nicht gewechselt.

oxynatOr, 2023-09-08:
> 

tbh its not a fix, its still stop working and complain about wrong framlength. 

that the reason why i don't do i mergre request..  but u are welcome :)

oxynatOr, 2023-09-08:
> @oxynatOr ich benutze nur 1x Home Assistant System genau.
> 
> Die String hat damals von Anfang an funktioniert und tut es momentan in der 1.43 immer noch. Deshalb habe ich nicht gewechselt.

Danke für die Antwort. 

solltest du lust und Langeweile haben, magst du meine mal testen? 

https://github.com/StephanJoubert/home_assistant_solarman/issues/367#issuecomment-1708839274

es wird nicht neues dabei sein, leider werden dir entities als Unbekannt angezeigt, weil ich die Namen der mppt2 und mppt4 übernommen habe. 

Da es nicht deinem Problem helfen wird, kann ich verstehen wenn du keine lust auf Experimente hast.

oxynatOr, 2023-09-08:
> > 
> 
> Working for me again, why do you think it's not a fix?

because there is more then that.  you didn't read the debug message?! 
sure, the registers are fixed.. but not the whole issue ppl reporting here.

Kolbi, 2023-09-08:
> > > 
> > 
> > 
> > Working for me again, why do you think it's not a fix?
> 
> because there is more then that. you didn't read the debug message?! sure, the registers are fixed.. but not the whole issue ppl reporting here.

Ah no not yet. Will have a look into it. Just fixed my issue with no data :)

Kolbi, 2023-09-08:
I don't see any "relevant" loglines for solarman.

`2023-09-08 16:33:50.712 DEBUG (SyncWorker_5) [custom_components.solarman.solarman] Starting to query for [1] ranges...
2023-09-08 16:33:50.712 DEBUG (SyncWorker_5) [custom_components.solarman.solarman] Querying [1 - 125]...
2023-09-08 16:33:50.712 DEBUG (SyncWorker_5) [custom_components.solarman.solarman] SENT: 
2023-09-08 16:33:51.105 DEBUG (SyncWorker_5) [custom_components.solarman.solarman] RECD: 
2023-09-08 16:33:51.105 DEBUG (SyncWorker_5) [custom_components.solarman.solarman] Querying [1 - 125] succeeded
2023-09-08 16:33:51.105 DEBUG (SyncWorker_5) [custom_components.solarman.solarman] All queries succeeded, exposing updated values.
2023-09-08 16:34:20.713 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] Starting to query for [1] ranges...
2023-09-08 16:34:20.713 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] Querying [1 - 125]...
2023-09-08 16:34:20.713 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] SENT: 
2023-09-08 16:34:21.049 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] RECD: 
2023-09-08 16:34:21.049 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] Querying [1 - 125] succeeded
2023-09-08 16:34:21.049 DEBUG (SyncWorker_4) [custom_components.solarman.solarman] All queries succeeded, exposing updated values.`

oxynatOr, 2023-09-09:
for example...

```
2023-09-08 12:13:49.419 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 17 00 b1 56 8c f7 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 01 00 7d d4 2b 90 15
2023-09-08 12:13:49.527 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] RECD: 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 36 5f 32 2e 32 37 0d 0a 0d 0a
```

and 

```
2023-09-06 10:05:41.340 DEBUG (SyncWorker_9) [custom_components.solarman.solarman] All queries succeeded, exposing updated values.
2023-09-06 10:06:09.708 DEBUG (Thread-15 (_data_receiver)) [custom_components.solarman.solarman] [POLL-DISCARDED] RECD: 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 36 5f 32 2e 30 36 0d 0a 0d 0a
2023-09-06 10:06:10.132 DEBUG (Thread-15 (_data_receiver)) [custom_components.solarman.solarman] [POLL-DISCARDED] RECD: a5 0f 01 10 15 00 38 d0 99 4f e9 02 01 e8 b2 07 00 3c 03 00 00 09 80 f0 64 01 03 fc 32 33 30 34 32 33 33 32 35 39 00 01 00 00 12 0c 07 00 01 02 01 17 12 18 00 00 1f 40 00 00 02 01 00 4b 00 00 00 3c 17 09 06 0a 03 0e 00 00 00 00 0a be 07 08 14 50 12 8e 00 00 00 00 00 00 13 9c 00 2c 00 00 00 00 00 00 00 00 00 64 00 00 00 00 00 01 00 00 00 00 00 01 00 00 00 00 00 01 ff d5 00 00 00 0a 00 02 00 00 00 00 00 00 00 00 00 00 00 04 00 08 00 00 00 00 01 2f 00 00 00 04 00 04 00 00 00 00 00 94 00 00 00 94 00 00 09 4c 00 00 00 00 00 18 00 00 00 00 13 88 00 00 00 00 00 00 00 00 00 00 00 00 16 a8 00 00 00 00 00 00 15 5e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 dd 00 3d 01 e0 00 3b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 b6 f7 15
2023-09-06 10:06:10.562 DEBUG (Thread-15 (_data_receiver)) [custom_components.solarman.solarman] [POLL-DISCARDED] RECD: a5 0f 01 10 15 00 39 d0 99 4f e9 02 01 e8 b2 07 00 3d 03 00 00 09 80 f0 64 01 03 fc 32 33 30 34 32 33 33 32 35 39 00 01 00 00 12 0c 07 00 01 02 01 17 12 18 00 00 1f 40 00 00 02 01 00 4b 00 00 00 3c 17 09 06 0a 03 0e 00 00 00 00 0a be 07 08 14 50 12 8e 00 00 00 00 00 00 13 9c 00 2c 00 00 00 00 00 00 00 00 00 64 00 00 00 00 00 01 00 00 00 00 00 01 00 00 00 00 00 01 ff d5 00 00 00 0a 00 02 00 00 00 00 00 00 00 00 00 00 00 04 00 08 00 00 00 00 01 2f 00 00 00 04 00 04 00 00 00 00 00 94 00 00 00 94 00 00 09 4c 00 00 00 00 00 18 00 00 00 00 13 88 00 00 00 00 00 00 00 00 00 00 00 00 16 a8 00 00 00 00 00 00 15 5e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 dd 00 3d 01 e0 00 3b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 b6 f9 15
```

or 

```
2023-09-06 15:08:32.313 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] frame_len does not match payload_len.
2023-09-06 15:08:32.313 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying 1 - 125 - failed with exception :V5FrameError: V5 frame contains invalid start or end values
```

i'm still collecting debug Information, that why i ask all the ppl to post it. 

indeed, the message is gone, or better lead us to the next issues

stvogel, 2023-09-09:
So I'm using the prefix
```
start: 0x0001
end: 0x007D
mb_functioncode: 0x03
```
in deye_2mppt.yaml

And I got a clear log like this:
```
2023-09-09 07:56:29.847 DEBUG (SyncWorker_2) [pysolarmanv5.pysolarmanv5] Socket setup completed... <socket.socket fd=27, family=2, type=1, proto=6, laddr=('192.yyyy', 54960), raddr=('192.xxxxx', 8899)>
2023-09-09 07:56:29.848 DEBUG (SyncWorker_2) [pysolarmanv5.pysolarmanv5] SENT: a5 17 00 10 45 6e 00 03 71 f3 ea 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 03 00 7b f5 e9 8d 15
2023-09-09 07:56:30.174 DEBUG (SyncWorker_2) [pysolarmanv5.pysolarmanv5] RECD: a5 09 01 10 15 6e ba 03 71 f3 ea 02 01 b1 cf 37 00 9e 0f 00 00 5b 39 c4 64 01 03 f6 32 33 30 35 31 32 33 30 31 32 00 01 00 00 12 0c 07 00 01 02 01 17 12 18 00 00 1f 40 00 00 02 01 00 4b 00 00 00 3c 17 09 09 07 34 07 00 00 00 00 0a be 07 08 14 50 12 8e 00 00 00 00 00 00 13 9c 00 2c 00 00 00 00 00 00 00 00 00 49 00 00 00 00 00 01 00 00 00 00 00 01 00 00 00 00 00 01 ff d5 00 00 00 0a 00 02 00 00 00 00 00 00 00 00 00 00 00 04 00 00 00 00 00 00 08 0c 00 00 00 00 00 00 00 00 00 00 03 fd 00 00 03 dd 00 00 09 38 00 00 00 00 00 00 00 00 00 00 13 88 00 00 00 00 00 00 00 00 00 00 00 00 00 b4 00 00 00 00 00 00 09 f6 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 3c 00 02 01 48 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 69 d9 ac 15
```

Only once in a while:
```
2023-09-09 07:56:09.834 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solarman_pv1_voltage is taking over 10 seconds
```
I got one Home Assistant running and a Deye SUN-M80G3-EU-Q0 with two panels.
Everything's fine.
Well ok, "Running Status" is always "Fault"

matthiaskind, 2023-09-09:
> Der Einwand mit der FW, kann ich ggf später / morgen prüfen. Da man die 2.27 nicht downloaden kann, vom FW-Server. Muss ich auf den Kollegen warten, der einen deye M80 mit diesem Relais bekommt.
Es gibt inzwischen die 2.30 und 2.31, die werden vermutlich ähnlich zur 2.27 sein und man kann sie vom FW-Server downloaden.

oxynatOr, 2023-09-09:
> Es gibt inzwischen die 2.30 und 2.31, die werden vermutlich ähnlich zur 2.27 sein und man kann sie vom FW-Server downloaden.

danke für die info

lenwar, 2023-09-10:
Hi,

Final update. Version 1.5.1 has resolved the issue for me. This version contains the updated files that you guys had conjured up!

My sincere thanks to all that spend so much time and effort in resolving this!!

closed 2023-09-10 by lenwar