StephanJoubert/home_assistant_solarman #372: Missing inverter definitions in Home Assistant compared to repo/instructions

Creat, 2023-08-18:
I've just wanted to try out to integrate my Deye SUN600G3-EU-230 into homeassistant (2023.8.2). Installation in HACS (1.32.1) went fine, but when I went to add my inverter I couldn't find the lookup file for my inverter, as mentioned in the README.md. It's supposed to be `deye_2mppt.yaml`, but only `deye_hybrid.yaml` and `deye_4mppt.yaml` are present in the drop-down list when trying to add it with the "automatic (config flow)" method.

So I tried adding it via configuration.yaml, but that also left me with the following error, suggesting the corresponding definition is actually missing.

```
Error while setting up solarman platform for sensor

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 362, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solarman/sensor.py", line 74, in async_setup_platform
    _do_setup_platform(hass, config, async_add_entities)
  File "/config/custom_components/solarman/sensor.py", line 51, in _do_setup_platform
    inverter = Inverter(path, inverter_sn, inverter_host, inverter_port, inverter_mb_slaveid, lookup_file)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solarman/solarman.py", line 33, in __init__
    with open(self.path + self.lookup_file) as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/config/custom_components/solarman/inverter_definitions/deye_2mppt.yaml'
```

The file is present in this repo, but I guess it somehow wasn't packaged into the HACS release? This is a listing of files from the definitions directory in my home assistant. Only 11 of the 16 yaml inverter_definitions from this repo are present.
```
➜  ~ cd /config/custom_components/solarman/inverter_definitions
➜  inverter_definitions ls -la
total 196
drwxr-xr-x    2 root     root          4096 Aug 17 23:42 .
drwxr-xr-x    5 root     root          4096 Aug 17 23:55 ..
-rw-r--r--    1 root     root          5883 Aug 17 23:42 deye_4mppt.yaml
-rw-r--r--    1 root     root         11622 Aug 17 23:42 deye_hybrid.yaml
-rw-r--r--    1 root     root         11518 Aug 17 23:42 deye_sg04lp3.yaml
-rw-r--r--    1 root     root          4375 Aug 17 23:42 deye_string.yaml
-rw-r--r--    1 root     root         32399 Aug 17 23:42 sofar_g3hyd.yaml
-rw-r--r--    1 root     root         28083 Aug 17 23:42 sofar_hyd3k-6k-es.yaml
-rw-r--r--    1 root     root         14661 Aug 17 23:42 sofar_lsw3.yaml
-rw-r--r--    1 root     root         18676 Aug 17 23:42 sofar_wifikit.yaml
-rw-r--r--    1 root     root          4166 Aug 17 23:42 solis_1p8k-5g.yaml
-rw-r--r--    1 root     root         19623 Aug 17 23:42 solis_hybrid.yaml
-rw-r--r--    1 root     root         21222 Aug 17 23:42 zcs_azzurro-ktl-v3.yaml

```

Digging around a bit, I found out that the file was just (slightly) newer than the latest release. The file was apparently created on 2023-02-22, and the latest release is from the 16th. So I assume the HACS pulls not from main, but from the latest release or tag?

Creat, 2023-08-18:
Update: I've tried installing the "live" version by directly specifying "main" as the version, even though it doesn't seem wise to run the dev branch directly, but ok for testing.
I still can't get it to connect to my Deye, even though the suitable inverter definition (2 MPPT) now exists and can be selected.
I get this entry in the system logs:
```
This error originated from a custom integration.

Logger: custom_components.solarman.solarman
Source: custom_components/solarman/solarman.py:56
Integration: Solarman (documentation, issues)
First occurred: 14:57:28 (221 occurrences)
Last logged: 15:05:28

Querying [3 - 128] failed with exception [IllegalDataValueError]
Querying [3 - 128] failed, [2] retry attempts left
Querying [3 - 128] failed, [1] retry attempts left
Querying [3 - 128] failed, [0] retry attempts left
Querying registers [3 - 128] failed, aborting.
```

UndIch, 2023-08-20:
Unfortunately something seems to be quite not working with the „G3“ (I have the 800G3) - it worked fine a couple of weeks ago but after some update it stopped working. No clue why … tried everything from reinstalling etc without any breakthrough. 
Thought abou adding the mqtt Deye from
Github but I am unable to understand how to integrate a external mqtt code correctly into the mosquito mqtt.

UndIch, 2023-08-20:
Somehow after days of trying all of a sudden I got it back working …for my G800 I choose the 4mptt in the 1.4.3 version and this worked (instead of the 2mptt). Maybe give it a try with this one as well?

lenwar, 2023-08-21:
As I wrote in the issue I raised. I manually downloaded the ‘live’ version of the mppt2 file and used that in the version of 1.4.3.

this works for me in practice. It throws a few errors per day, but functionally it works

UndIch, 2023-08-21:
That’s really strange. For me the mptt „live“/„main“ version doesn’t work, therefore the old mptt of the 1.4.3…

lenwar, 2023-08-21:
I have 1.4.3 installed and I copied the contents of the ‘main’ version of mppt2 in custom_somethingsomething.yaml :) and referee to that.

Creat, 2023-09-05:
After some fiddling around, I'm even more confused now. For the first time ever, after reading some other issues and some googling, I got my inverter to show up, but using the "string" type, not 2MPPT type, despite my inverter being specifically listed as an example for 2MPPT. I'm still on the "main" build from HACS, not sure if that's up-to-date with github or a snapshot from some point in time.

But it's not exactly working well either, as the inverter keeps disappearing and it's entities show "unknown", despite it working just fine and making power. I know the inverters are powered only by the panels, so low light means it might come and go, but this seems to happen even at noon on a sunny day. I have no idea if the inverter is flaky, or picky about connections or what else it could be...

Creat, 2023-09-16:
Has anyone else with this issue tried the newly released 1.5.1 yet? I'm traveling and my setup is packed up at the moment (won't be for long, but is for now).