StephanJoubert/home_assistant_solarman #374: Show just export to grid (excluded import)

lordvalium, 2023-08-23:
How can i see the export to grid with the import excluded?

Hetzi74, 2023-08-26:
What exactly do you want to get? Du you want to get kWh of exported energy to grid? (This is stored at sensor.solarman_energy_selling_today at least at my hyd-zss-hp-3k-6k.yaml for my inverter)

Or do you want to get the power at every minute. That is something you have to make some math with a template. Something like this i do:

template:
  - sensor:
    - name: "Grid Power"
      unit_of_measurement: 'W'
      icon: 'mdi:solar-power'
      state: '{{ states("sensor.solarman_active_power_output_total") |float(0) - states("sensor.solarman_active_power_load_total" )|float(0) }}'

But this is still the combined case with export an import to grid combined. To get the export only you will have to add some if case around to just get the positive export values.

lordvalium, 2023-09-03:
Thanks that worked.