StephanJoubert/home_assistant_solarman #389: Question Power Regulation

GeneralRurra, 2023-09-07:
Hy I have a sun1600
Will it be possible later to control the active_power_regulations ?

smurfy, 2023-10-02:
Tried it with advanced write method, did not work for me.

active power regulations register should be `40 (0x28`).  Value should be in `percent * 10`
I tried value of `50` and since this did not work `500`, both result in an error:

[image]

```
2023-10-02 15:09:32.507 WARNING (MainThread) [custom_components.solarman.solarman] Service Call: write_holding_register : [40], value : [50] failed with exception [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
2023-10-02 15:12:13.337 WARNING (MainThread) [custom_components.solarman.solarman] Service Call: write_holding_register : [40], value : [500] failed with exception [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]
```

Any ideas?

smurfy, 2023-10-02:
Setting and reading value worked after installing: https://github.com/githubDante/deye-controller
I'm fairly new to HA, but it should work using scripts to trigger calls based on some conditions.

smurfy, 2023-10-18:
Using shell commands triggered by HA with `deye-regwrite` works great for me.
Using it to set power regulation and enable/disable inverter.

For enable/disable for some reason my SUN300 wants 1 as ON and 2 as OFF, while the deye2mpp definitions says 0 for OFF.

add this to you configuration.yaml:

```
input_number:
  solarman_power_regulation:
    name: Active power regulation
    initial: 100
    min: 25
    max: 100
    step: 1

input_boolean:
  solarman_enabled:
    name: Enable inverter
    icon: 'mdi:toggle-switch'

shell_command:
  set_solarman_power_regulation_slider: 'deye-regwrite IP SERIAL 40 {{ states("input_number.solarman_power_regulation") }}'
  set_solarman_enable_toggle: 'deye-regwrite IP SERIAL 43 {% if is_state("input_boolean.solarman_enabled", "on") %}1{% else %}2{% endif %}'
```

Add this to your automations.yaml

```
- id: '1697292557474'
  alias: run_set_solarman_power_regulation
  trigger:
    platform: state
    entity_id: input_number.solarman_power_regulation
  action:
    service: shell_command.set_solarman_power_regulation_slider
- id: '1697292557475'
  alias: run_set_solarman_enabled
  trigger:
    platform: state
    entity_id: input_boolean.solarman_enabled
  action:
    service: shell_command.set_solarman_enable_toggle
```

TomtomWa, 2023-12-29:
> Tried it with advanced write method, did not work for me.
>
> active power regulations register should be `40 (0x28`). Value should be in `percent * 10` I tried value of `50` and since this did not [...]

@smurfy: I see in the other comments, that you found a solution. Anyhow, if anyone else is wondering about the same problem, here is what worked for me:
[image]
Use the service "Write multiple holding registers" instead of "Write holding register".

But I have another question:
I have two Sun600 at the moment and want to control both independently. Currently the service only sets the Sun600, that I added as last. That means, I cannot control the first added Inverter. It seams to me, that the service was registered with the same generic name "solarman.write_...".
Anyone has an idea, how to get two services with different names for both inverters?

sdk69, 2025-04-01:
same issue for me, I have 3 inverters and the Write multiple holding registers only works for one, have you solved it ? thanks !