StephanJoubert/home_assistant_solarman #391: You made a mess

jolly12f, 2023-09-07:
I updated the integration and nothing works anymore I lost everything thank you very much

niclas911, 2023-09-07:
Kehre mit einem downgrade zur vorherigen Version zurück. Diese läuft bei mir auch noch, da die neuste auch keine Verbindung herstellen kann.

Quellweg22, 2023-09-08:
> Kehre mit einem downgrade zur vorherigen Version zurück. Diese läuft bei mir auch noch, da die neuste auch keine Verbindung herstellen kann.

leider läuft downgrade bei mir auch nicht "mehr"

jolly12f, 2023-09-08:
They changed the names of the sensors and the downgrade no longer works

DanielBakovic, 2023-09-08:
I can confirm. Nothing works after the update.

Steinbichler, 2023-09-08:
You should stop the delivery of this update.

The only way for me to get it working again is downgrate via backup restore.

In my case the update of the .yaml files does not fix the problem.

oxynatOr, 2023-09-08:
you guys have a Deye SUN /M ?   if yo, pls join: #367

Steinbichler, 2023-09-08:
I saw the post #36. But sorry beeing that clear.

I am not a programmer. I am a simple user.

I do not even understand most of the things in this post. 

I had it running and the update to 1.5 killed it.

AND:  I want to thank you all for the efford to do all this programming.

oxynatOr, 2023-09-08:
what would help, is posting the debug log. 
you know how to enable it?

Guenni75, 2023-09-09:
I downgraded (1.4.3) yesterday and it took a day to work again!

Steinbichler, 2023-09-09:
@oxynatOr 

Yes I know how to enable debug log, but this would mean going back to 1.5 and again crashing down my system.

hmmmmm

oxynatOr, 2023-09-10:
kein problem. ist kein muss.

Steinbichler, 2023-09-10:
Gut sag mir was Du brauchst. was ich machen soll.

Mit einem Backup ist der Weg zurück ja kein so riesen Aufwand.

jolly12f, 2023-09-10:
The downgrade doesn't work, I solved it by replacing the names of the sensors and now everything works again

LuckyTriple7, 2023-09-10:
V 1.5.1 is working for me, deye_2.mppt.yaml, Thx

niclas911, 2023-09-10:
Funktioniert mit der FW2.27 leider immer noch nicht.

Stefan051280, 2023-09-11:
Bei mir funktioniert es ebenfalls nicht FW2.27 Update auf 1.5.1

Steinbichler, 2023-09-11:
@oxynatOr

Auch mit 1.51 jetzt allerdings mit der 2.27 FW nach wie vor keine Funktion. Zurück per Backup Restore geht alles wieder.

Ich hatte jetzt mal den Debug Log mitlaufen lassen.
[home-assistant_solarman_2023-09-11T17-15-32.381Z.log](https://github.com/StephanJoubert/home_assistant_solarman/files/12578930/home-assistant_solarman_2023-09-11T17-15-32.381Z.log)

Vielleicht / hoffentlich hilfts.

oziee, 2023-09-12:
as per the readme
`create a lookup file "custom_parameters.yaml" and refer to it so that it will not be overwritten during updates.`

this update deleted my custom config file

jmccrohan, 2023-09-12:
> @oxynatOr
>
> Auch mit 1.51 jetzt allerdings mit der 2.27 FW nach wie vor keine Funktion. Zurück per Backup Restore geht alles wieder.
>
> Ich hatte jetzt mal den Debug Log mitlaufen lassen. [...]

The response 
`41542b595a434d505645523d4d57335f3136555f353430365f322e32370d0a0d0a`, when converted to ASCII is `AT+YZCMPVER=MW3_16U_5406_2.27` and is most certainly not a valid Modbus response.

Based on https://github.com/kbialek/deye-inverter-mqtt/issues/123, it would appear to be an issue with the new firmware rather than v1.5.0. If you can provide any additional information, for example if the previous version 1.4.3 is still working, I am willing to amend pysolarmanv5 to maintain backwards compatibility where possible.

Steinbichler, 2023-09-13:
Ok here some additional information.

I can not agree with your point about the firmware of the converter causes the problem.

The version 1.43 works fine with my deye converter in Firmware version 1.5x an the new version 2.27. 
In all cases when I tested the homeasistant software:

Converter 1.5x to your new version 1.50     »»     no more function
Converter 2.27 to your new version 1.51     »>     no more function

when I went back via backup to your software version 1.43 every thing works fine.

At the moment Converter runs with the new relay an firmware 2.27 and your software runs with 1.43 without any problems.

I again run the debug log with this running configuration an attached it to this post.

[home-assistant_solarman_2023-09-13T07-21-54.990Z.log](https://github.com/StephanJoubert/home_assistant_solarman/files/12594747/home-assistant_solarman_2023-09-13T07-21-54.990Z.log)

NikolajChristensen, 2023-09-13:
> I updated the integration and nothing works anymore I lost everything thank you very much

Let's remember this is free software that people spend their free time on. I think we can all appreciate that.

oxynatOr, 2023-09-14:
> > @oxynatOr
> > Auch mit 1.51 jetzt allerdings mit der 2.27 FW nach wie vor keine Funktion. Zurück per Backup Restore geht alles wieder.
> > Ich hatte jetzt mal den Debug Log mitlaufen lassen. [home-assistant_solarman_2023-09-11T17-15-32.381Z.log](https://github.com/StephanJoubert/home_assistant_solarman/files/12578930/home-assistant_solarman_2023-09-11T17-15-32.381Z.log)
> > Vielleicht / hoffentlich hilfts.
> 
> The response `41542b595a434d505645523d4d57335f3136555f353430365f322e32370d0a0d0a`, when converted to ASCII is `AT+YZCMPVER=MW3_16U_5406_2.27` and is most certainly not a valid Modbus response.
> 
> Based on [kbialek/deye-inverter-mqtt#123](https://github.com/kbialek/deye-inverter-mqtt/issues/123), it would appear to be an issue with the new firmware rather than v1.5.0. If you can provide any additional information, for example if the previous version 1.4.3 is still working, I am willing to amend pysolarmanv5 to maintain backwards compatibility where possible.

version 1.4.3 works pretty well. so i created a frok and will maintain it for my own, since i dont wanna steal anything. 
i'm on FW 2.62-D1, with a working relay. 

i do also block the china cloud per firewall. 

if you take a close look at https://github.com/StephanJoubert/home_assistant_solarman/issues/367  (Closed :( )
i posted a lot of debug messages. from all 3 inverter i have running.

Steinbichler, 2023-09-15:
Ok thanks again for your efford.

Ich will stop in this post, and will try to follow #367. But as I said: Lot of Details I even do not understand. 

If you need anything from me to test it on my Deye Sun 600 send me a message.

oxynatOr, 2023-09-15:
@jmccrohan 
i also tested the scripts you told me (py and golang, don't rember the names), all of them works out of the box.

oxynatOr, 2023-09-15:
> Ok thanks again for your efford.
> 
> Ich will stop in this post, and will try to follow #367. But as I said: Lot of Details I even do not understand.
> 
> If you need anything from me to test it on my Deye Sun 600 send me a message.

first off all, you need to update the fw asap. 2.27  is confirmed broken.. inverter will stop working with the relay box installed. 

http://47.254.36.66/0_D0002_18/MW3_16U_5406_2.32-D1.bin

this is the most up-2-date firmware.  

pls dont get me wrong, its offtopic and have nothing todo with the issue. 

Template<template=({{(states('sensor.solarman_pv1_current') | float)*(states('sensor.solarman_pv1_voltage') | float) | round(1)}}) renders=2>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1927, in forgiving_float_filter
    return float(value)
           ^^^^^^^^^^^^
ValueError: could not convert string to float: 'unknown'

you are creating a own sensor for current_watt?  
if so, you should do it lime this:

`
- name: Solar AC Output Power Aki
  unit_of_measurement: "W"
  #device_class: power
  unique_id: "Solar_AC_Output_Power_Aki"
  state_class: "measurement"
  icon: mdi:solar-power-variant-outline
  state: >-
    {% set volt = states('sensor.aki_m80_ac_voltage') %}
    {% set amp = states('sensor.aki_m80_grid_current') %}
    {% if is_number(volt) and is_number(amp) %}
    {{ float(volt,0) * float(amp,0) }}
    {% else %} 0.0
    {% endif %}
`

it will do a check for a numeric value (0-9), when this it something like 'unknown' or any other ascii-chars you will get a 0.0.
maybe this will help you in the future 

the, there is this message: `Querying [3 - 128] succeeded`
it means, the yaml file you use is starting at 0x03 till 0x80 . the range is not need like this. because 120-125 are just debug, with no communicated value. the Whitepaper about RTU says 0x001 - 0x007B 

could you give us a debug log with the new yaml files? ... or fell free to take mine 
https://github.com/StephanJoubert/home_assistant_solarman/issues/367#issuecomment-1708839274
(device time will be wrong', you need to remove register 0x0015 the first one.)

no clue why the dude with PR did not copy my file.. he only took the start-end register. because i checked the register with the RTU-Whitepaper.

Steinbichler, 2023-09-17:
Ok thank you for the hint about the firmware of the converter.

Ich checked ist an Deye did an update automaticaly to MW3_16U_5406_2.32-D1

So the firmware is ok.

To your points:

I did again update to 1.51 of your software.

I took the existing deye_2mppt.yaml after the update and deleted all content in the file.

Per copy / paste I took the content of your yaml file. (As your link)

I deleted the existing Integration and set up a new intergartion with the new yaml file with your content.

It does not work

Then I restartet hommeassistant with this configuration.

It does not work.

Here is a log file with the latest status:

[home-assistant_solarman_2023-09-17T15-45-24.467Z.log](https://github.com/StephanJoubert/home_assistant_solarman/files/12643184/home-assistant_solarman_2023-09-17T15-45-24.467Z.log)

After all:

I restored the total backup. Everythinks works fine.

jmccrohan, 2023-09-17:
@Steinbichler (+ @oxynatOr):

I've reviewed the previously attached v1.4.3 log file (home-assistant_solarman_2023-09-13T07-21-54.990Z.log).

```
2023-09-13 08:46:27.339 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] Starting to query for [1] ranges...
2023-09-13 08:46:27.346 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] Querying [3 - 128]...
2023-09-13 08:46:27.346 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] a51700104500004cbac3e902000000000000000000000000000001030003007e35eac415
2023-09-13 08:46:27.451 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] 41542b595a434d505645523d4d57335f3136555f353430365f322e32370d0a0d0a
2023-09-13 08:46:27.451 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] unexpected v5 packet start/stop
2023-09-13 08:46:27.451 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] Querying [3 - 128] failed, invalid response packet.
2023-09-13 08:46:27.451 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] Querying [3 - 128] failed, [5] retry attempts left
2023-09-13 08:46:27.451 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] a51700104500004cbac3e902000000000000000000000000000001030003007e35eac415
2023-09-13 08:46:27.753 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] aa030a00000000000000000000be9c
2023-09-13 08:46:27.754 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] unexpected v5 packet start/stop
2023-09-13 08:46:27.754 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] Querying [3 - 128] failed, invalid response packet.
2023-09-13 08:46:27.754 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] Querying [3 - 128] failed, [4] retry attempts left
2023-09-13 08:46:27.754 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] a51700104500004cbac3e902000000000000000000000000000001030003007e35eac415
2023-09-13 08:46:27.875 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] a50f01101500074cbac3e90201864e5f00581900003d0ca2640103fc3232313232383637323700010000120c07000101011612160000177000000201004b0000003c17090d082d23000000000abe0726141e128e000000000000139c002c00000000000000000064000000000001000000000001000000000001fff50000000a00020000000000000000000000040000000000000c94000000000000000000000634000006070000093800000000000000000000138800000000000000000000000000b40000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011c0002011100030000000000000000000000000000000000000000000000000000000000000000d346b915
2023-09-13 08:46:27.876 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] Querying [3 - 128] succeeded
2023-09-13 08:46:27.876 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] All queries succeeded, exposing updated values.
```

In both logs, the device spewes out non-sense replies - See the two retries above, before a correct `a50f01101500074..` response was received. I tightened up the timeouts/retries to ensure that HA recovered quicker after a read error, but in the case of this faulty device such as this, it means that HA disconnects before the device got a chance to send the correct reply.

Can you upgrade to v1.5.1 and replace the existing solarman.py file with [this](https://github.com/jmccrohan/home_assistant_solarman/blob/7972af87fdea68202dcf5d0741a1ad6d7388063b/custom_components/solarman/solarman.py) version please? You'll need to manually upload this file to your HA server under the path below

`/root/config/custom_components/solarman/solarman.py`

Thanks.

oxynatOr, 2023-09-18:
> @Steinbichler (+ @oxynatOr):
>
> I've reviewed the previously attached v1.4.3 log file (home-assistant_solarman_2023-09-13T07-21-54.990Z.log).
>
> ```
> 2023-09-13 08:46:27.339 DEBUG (SyncWorker_0) [...]

quick review... 
line 97 and 98 dont look right. 

there is nothing this script would do if result equal 0.

Steinbichler, 2023-09-18:
ok I have done the things as you asked me to do

Update to 1.51. Changed the solarman.py and restarted homeassistanf.

First of all: It does not work, back with backup restore everything ok.

But there are some differences wit the new solarman.py:

With the old solarman.py at 1.51 the message is:

Die Endität ist numerisch (The endity is not numeric)

With the new solarman.py at 1.51 the message is:

Die Entität ist nicht verfügbar (The edity is not available)

and there is an error message at home assist:

Home Assistant Core
Unable to prepare setup for platform solarman.sensor: Platform not found (Exception importing custom_components.solarman.sensor).
10:12:14 – (FEHLER) setup.py
Unexpected exception importing platform custom_components.solarman.sensor
10:12:13 – (FEHLER) Solarman (benutzerdefinierte Integration)

Again a logfile:
[home-assistant_solarman_2023-09-18T08-49-13.592Z.log](https://github.com/StephanJoubert/home_assistant_solarman/files/12646204/home-assistant_solarman_2023-09-18T08-49-13.592Z.log)

In addidtion I changed my own sensors as you suggested to do. They are now like this:

- sensor:
      - name: "Power_PV1"
        unique_id: "solar_pv1_power"
        state_class: "measurement"
        unit_of_measurement: W
        state: >-
            {% set volt1 = states('sensor.solarman_pv1_voltage') %}
            {% set amp1 = states('sensor.solarman_pv1_current') %}
            {% if is_number(volt1) and is_number(amp1) %}
            {{ float(volt1,1) * float(amp1,1) | round(1)}}
            {% else %} 0.0
            {% endif %}

here is a log file of the system running at 1.43 with my changes sensors

[home-assistant_solarman_2023-09-18T09-14-33.368Z.log](https://github.com/StephanJoubert/home_assistant_solarman/files/12646230/home-assistant_solarman_2023-09-18T09-14-33.368Z.log)

Thank you for your help!!!