StephanJoubert/home_assistant_solarman #394: sofar 8.8ktl-x does not work after updating to 1.5

tomkolp, 2023-09-09:
After updating to 1.5, all entities are unavailable.  Downgrade solves the problem.

lukazareba, 2023-09-11:
# SOFAR Solar 8.8KTLX-G3

requests:
  - start: 0x0400
    end: 0x042B
    mb_functioncode: 0x03
  - start: 0x0482
    end: 0x04A5
    mb_functioncode: 0x03
  - start: 0x0582
    end: 0x0589
    mb_functioncode: 0x03
  - start: 0x0682
    end: 0x068B
    mb_functioncode: 0x03

parameters:
  - group: Solar
    items:
      - name: 'PV Generation today
        class: 'energy'
        state_class: 'total'
        uom: 'kWh'
        scale: 0.01
        rule: 1
        registers: [0x0685, 0x0684]
        icon: 'mdi:solar-power'

      - name: 'PV Generation total'
        class: 'energy'
        state_class: 'total_increasing'
        uom: 'kWh'
        scale: 0.1
        rule: 3
        registers: [0x0687, 0x0686]
        icon: 'mdi:solar-power'

      - name: 'PV1 Power'
        class: 'power'
        state_class: 'measurement'
        uom: 'W'
        scale: 10
        rule: 1
        registers: [0x0586]
        icon: 'mdi:lightning-bolt'

      - name: 'PV2 Power'
        class: 'power'
        state_class: 'measurement'
        uom: 'W'
        scale: 10
        rule: 1
        registers: [0x0589]
        icon: 'mdi:lightning-bolt'

      - name: 'PV1 Voltage''
        class: 'voltage'
        state_class: 'measurement'
        uom: 'V'
        scale: 0.1
        rule: 1
        registers: [0x0584]
        icon: 'mdi:sine-wave'

      - name: 'PV2 Voltage'
        class: 'voltage'
        state_class: 'measurement'
        uom: 'V'
        scale: 0.1
        rule: 1
[... 724 lines cut ...]
            value: 'ID180 BMS low temperature alarm'
          - key: 16
            value: 'ID181 BMS charge/discharge overload alarm'
          - key: 32
            value: 'ID182 BMS short circuit alarm'
          - key: 64
            value: 'ID183 BMS version inconsistency'
          - key: 128
            value: 'ID184 BMS CAN version inconsistency'
          - key: 256
            value: 'ID185 BMS CAN version is too low'
          - key: 512
            value: 'ID186 '
          - key: 1024
            value: 'ID187 '
          - key: 2048
            value: 'ID188 '
          - key: 4096
            value: 'ID189 Arc device communication failure'
          - key: 8192
            value: 'ID190 DC arc alarm fault'
          - key: 16384
            value: 'ID191 PID repair failed'
          - key: 32768
            value: 'ID192 PLC module heartbeat loss'

lukazareba, 2023-09-11:
Also add an entry in the const.py file

```
from datetime import timedelta

DOMAIN = 'solarman'

DEFAULT_PORT_INVERTER = 8899
DEFAULT_INVERTER_MB_SLAVEID = 1
DEFAULT_LOOKUP_FILE = 'deye_hybrid.yaml'
LOOKUP_FILES = [
    'deye_2mppt.yaml',
    'deye_4mppt.yaml',
    'deye_hybrid.yaml',
    'deye_sg04lp3.yaml',
    'deye_string.yaml',
    'kstar_hybrid.yaml',
    'sofar_g3hyd.yaml',
    'sofar_hyd3k-6k-es.yaml',
    **### 'sofar_ktl_x_g3.yaml',**
    'sofar_lsw3.yaml',
    'sofar_wifikit.yaml',
    'solis_1p8k-5g.yaml',
    'solis_3p-4g.yaml',
    'solis_hybrid.yaml',
    'solis_s6-gr1p.yaml',
    'zcs_azzurro-ktl-v3.yaml',
    'custom_parameters.yaml'
]

MIN_TIME_BETWEEN_UPDATES = timedelta(seconds=15)

CONF_INVERTER_HOST = 'inverter_host'
CONF_INVERTER_PORT = 'inverter_port'
CONF_INVERTER_SERIAL = 'inverter_serial'
CONF_INVERTER_MB_SLAVEID = 'inverter_mb_slaveid'
CONF_LOOKUP_FILE = 'lookup_file'

SENSOR_PREFIX = 'Solarman'
```

rimelis, 2023-09-12:
Version 1.5.0 has a **breaking change** - the modbus protocol implementation using pysolarmanv5 module instead of internal one existing in previous versions.

lukazareba, 2023-09-13:
I don't understand, in general it means it won't work on some inverters now?

rimelis, 2023-09-13:
Yes, It may not work at all or work partially. In my case the stick (LSW3_15_270A_1.50) is buggy and pysolarmanv5 is working somehow but keep crashing threads of Home Assistant and generates a lot of mess into log. I decided to revert to older "local implementation" found in version < 1.5.0

NemesisXB, 2023-12-04:
@jmccrohan, Any idea?