StephanJoubert/home_assistant_solarman #395: PySolarmanV5 no load

CrazyUs3r, 2023-09-09:
Moin beim 1.5.0 sagt ha PySolarmanV5 kann nicht geladen werden.

CrazyUs3r, 2023-09-10:
2023-09-10 18:42:54.214 DEBUG (SyncWorker_12) [pysolarmanv5.pysolarmanv5] SENT: a5 17 00 10 45 64 00 62 c8 7c f8 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 01 00 7d d4 2b f1 15
2023-09-10 18:42:54.541 DEBUG (SyncWorker_12) [pysolarmanv5.pysolarmanv5] RECD: a5 0d 01 10 15 63 33 62 c8 7c f8 02 01 e9 87 58 00 d8 25 00 00 24 6a a5 64 01 03 fa 01 00 02 01 32 32 31 30 31 31 39 34 34 36 00 01 00 00 12 0c 07 00 00 00 01 14 12 12 00 00 1f 40 00 00 02 01 00 4b 00 00 00 3c 17 09 0a 12 26 0b 00 00 00 00 0a be 07 08 14 50 12 8e 00 00 00 00 00 00 13 9c 00 2c 00 00 00 00 00 00 00 00 00 64 00 00 00 00 00 01 00 00 00 00 00 01 00 00 00 00 00 01 00 00 00 00 00 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 04 00 0e 00 00 00 00 07 1f 00 00 00 06 00 07 00 00 00 00 03 69 00 00 03 65 00 00 09 1a 00 00 00 00 00 00 00 00 00 00 13 7d 00 00 00 00 00 00 00 00 00 00 00 00 00 d2 00 00 00 00 00 00 11 1c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 36 00 03 01 0e 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 78 38 15

CrazyUs3r, 2023-09-10:
Source: custom_components/solarman/solarman.py:56
Integration: Solarman (documentation, issues)
First occurred: 18:42:54 (373 occurrences)
Last logged: 19:38:54

Querying [1 - 125] failed with exception [V5FrameError]
Querying [1 - 125] failed, [5] retry attempts left
Querying [1 - 125] failed, [4] retry attempts left
Querying [1 - 125] failed, [3] retry attempts left
Querying inverter 4168927330 at 192.168.178.229:8899 failed on connection start with exception [NoSocketAvailableError]

knightswsift, 2023-09-17:
Relais-Box installiert? Habe das gleiche Problem seit FW 2.27 bzw 2.32-D1 in Verbindung mit dem Relais. Ich vermute fast, die Box blockiert den Kanal, über den HA normalerweise ausliest.

CrazyUs3r, 2023-09-17:
Nein das relais hab ich noch nicht dran der wr ist noch auf 2.06