StephanJoubert/home_assistant_solarman #396: 1.5.1 funktioniert nicht für FW2.31

niclas911, 2023-09-11:
Ich habe heute morgen auf die neuste Firmware 2.31 (MW3_16U_5406_2.31) für den deye 600 Microwechselrichter umgestellt. Leider funktioniert auch hier das neuste Update nicht. Folgend der Log. 
Da in Deutschland in den kommenden Wochen vorraussichtlich immer mehr auf das Relais umstellen werden. Wird es vermehrt zu Fehlern kommen.

2023-09-11 12:36:41.099 DEBUG (MainThread) [custom_components.solarman] __init__.py:async_setup_entry({'entry_id': '8222bd432e9089d09c88d6f91d206405', 'version': 1, 'domain': 'solarman', 'title': '192....', 'data': {'name': 'Solarman', 'inverter_host': '192....', 'inverter_serial': 12345678, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_string.yaml'}, 'options': {'name': 'Solarman', 'inverter_host': '192....', 'inverter_serial': 12345678, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_2mppt.yaml'}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'unique_id': None, 'disabled_by': None})
2023-09-11 12:36:42.100 ERROR (roombapy) [roombapy.remote_client] Can't connect to 192....., error: [Errno 111] Connection refused
2023-09-11 12:36:45.357 ERROR (roombapy) [roombapy.remote_client] Can't connect to 192....., error: [Errno 111] Connection refused
2023-09-11 12:36:47.360 DEBUG (MainThread) [custom_components.solarman.sensor] sensor.py:async_setup_entry: {'name': 'Solarman', 'inverter_host': '192....', 'inverter_serial': 12345678, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_2mppt.yaml'}
2023-09-11 12:36:47.360 DEBUG (MainThread) [custom_components.solarman.sensor] sensor.py:async_setup_platform: {'name': 'Solarman', 'inverter_host': '192....', 'inverter_serial': 12345678, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'deye_2mppt.yaml'}
2023-09-11 12:36:47.464 DEBUG (MainThread) [custom_components.solarman.sensor] sensor.py:_do_setup_platform: async_add_entities
2023-09-11 12:36:47.464 DEBUG

knightswsift, 2023-09-17:
Wurde zeitgleich die Relais-Box installiert?

saz, 2023-09-17:
I've installed the relay box today and my Inverter is now running 2.32.

I'm getting the same error.

knightswsift, 2023-09-17:
Solarman App reports MI-Relais has modbus address 1, is this relevant?

knightswsift, 2023-09-17:
Can't find any other address than that, sry

knightswsift, 2023-09-18:
Das ist interessant. Und die Relais-Box ist richtig eingebunden und wird in der App angezeigt? Angeblich laufen manche WR mit der 2.32 auch ohne Box, deshalb die Frage. Werde dann wenn ich Zeit habe mal downgraden und testen.

ChirpyTurnip, 2023-09-18:
Sounds similar to my logged issue #[405 ](https://github.com/StephanJoubert/home_assistant_solarman/issues/405).

Inverter data connections keep dropping. I went back to 1.4.3 and it has gone back to being really reliable.....

niclas911, 2023-09-18:
Ich habe genau die gleichen Vorraussetzungen wie @happy-paul7 und bei mir läuft die 1.4.3 sehr gut. Die 1.5.1 funktioniert leider nicht.

ChirpyTurnip, 2023-09-18:
Willkommen auf der Party, mein Freund. Diejenige, bei der wir auf der alten Integration sitzen bleiben und die Updates ignorieren, bis es etwas Neues zum Ausprobieren gibt ...

Und wenn das fehlschlägt, kehren wir zum alten zurück und das Spiel beginnt von vorne ...

Wenn ich programmieren könnte, könnte ich helfen, aber jetzt kann ich mich nur noch beschweren. Es ist also eine traurige Party...aber zumindest bin ich nicht allein :-)

happy-paul7, 2023-09-19:
Die Relaisbox wird in der Solarmann App und unter https://home.solarmanpv.com/plant/infos/device  RICHTIG angezeigt.
Werde weiter testen ;-(
Bei mir hat ein neues laden der Version von 1.5.2 nach 1.4.3 keinen Erfolg gehabt.
Habe ein altes Backup eingespielt !! Danach waren wieder Daten sichtbar.

thomasgregg, 2023-09-20:
Ich habe dasselbe Problem. Nach Installation der Relais bekomme ich in Solarman in HA nichts mehr angezeigt. In der Solarman App funktioniert alles einwandfrei.

[image]

knightswsift, 2023-09-22:
Das Relais blockiert den Port. Wenn man AC-seitig abschaltet und HA neustartet, kommen nach kurzer Zeit wie gewohnt die Werte .

happy-paul7, 2023-09-22:
Mit welcher Version von Solarman wurde der Test durchgeführt ?
Wird in HACS eine neue Version für die Integration angezeigt?
--------------------------------------------------------------------
ggf. komplettes Backup erstellen - 
**Testablauf heute :**
Version 1.43 
Verzeichnis /config/custom_components/solarman  -> auf PC gesichert
Neue Version über HACS - geladen
HA neu gestartet 
**Fehlermeldung siehe oben -**  z.B.  -- No value recorded for PV1 Voltage usw.
damit wieder Daten ausgelsen werden : 
gesichertes Verzeichnis (Solarman) von PC wieder in HA zurück kopiert
HA neu gestartet - 
Alle Werte werden wieder angezeigt - im Protokoll keine Fehler
Achtung: Es wird kein Update mehr über HACS angeboten -  
 -- zur Kontrolle  deye_2mppt.yaml  Start und Endadresse vergleichen !! - Änderungen sind ja in 
Update deye_2mppt.yaml  sichtbar - 
In Version 1.5.1 wurden natürlich noch andere Dateien verändert. 
const.py
parser.py
sensor.py
solarman.py
Um wieder auf eine alte Version komplett zurück zu kommen, sollte jetzt das Backup wieder eingepielt werden !!

knightswsift, 2023-09-24:
Theoretisch kann man die 1.4.3 drüberbügeln und dann die deye_2mppt.yaml in den entsprechenden Ordner schieben und in die const.py eintragen. Die Entitäten waren danach bei mir alle wieder verfügbar, auch ohne backup-jonglieren.

DennyKDE, 2023-10-01:
Ich habe gestern das Update auf 1.5.1 gemacht  und zuerst war alles wie gewohnt. vorher hatte ich glaube 1.3.3. Als mein sun800 (ohne Relaisbox) dann Abends nichts mehr produziert hat und auch offline war ist die Anzeige alle Werte auf „Unbekannt“ gesprungen. Bisher war es bei mir so das einfach der zu letzt gesendete Wert bis zum nächsten Morgen angezeigt wurde.

Ist das neue Verhalten denn so explizit gewünscht, das also wirklich nach Sonnenuntergang bis Sonnenaufgang dann die ganze Zeit nur Unbekannt steht?

ChirpyTurnip, 2023-10-01:
Nein, das ist ein Verhalten, das ich auch erlebe. Yoyu lädt es und führt es aus und alles sieht normal aus und dann funktioniert es ohne Vorwarnung nicht mehr. Wenn Sie HA neu starten, verbindet es sich wieder mit dem Solarwechselrichter und beginnt wieder zu arbeiten ... bis es das nächste Mal wieder stoppt. Dies ist erst seit 1.5.x ein Problem. Ich bin derzeit wieder bei 1.4.3 und alles ist in Ordnung ...

:-)

DennyKDE, 2023-10-02:
Bei mir funktionierte es mit 1.5.1 den ganzen Tag Problemlos, nur nach Sonnenuntergang stand halt überall „Unbekannt“ anstatt der zuletzt übermittelte Wert wie sonst.

Ich habe jetzt mal mein altes Solarman Verzeichnis aus einem Backup zurück gespielt und bin jetzt mal auf den heutigen Tag bzw. Sonnenuntergang gespannt.

Kleinheribert, 2023-10-02:
> Wo kann man denn die Version 1.4.3 denn noch als Paket herunterladen?

kann man in HACS ganz normal auswählen und downloaden. Bei Github sollte das ja auch möglich sein.

Bin wegen des selben Problems hier, zwei Wechselrichter, einer mit 1.57 und einer mit der neuen FW und Relais. Ratet mal, wer keine Daten mehr sendet? Bei meinem Nachbarn das 100% identische Setup, seit Wochen funktioniert das alles einwandfrei. Könnte kotzen.

ChirpyTurnip, 2023-10-02:
Es gibt immer subtile Unterschiede ... Code ist schwierig, weshalb die meisten von uns ihn nicht verwenden. Auch bei mir gehen die Werte auf „unbekannt“, wenn der Wechselrichter offline ist. Dies ist jedoch häufiger der Fall, wenn ich HA neu gestartet habe – dann verliert es den aktuellen Status und kann keinen neuen erstellen, da die Wechselrichter offline sind. Um dies zu umgehen, füge ich alle meine Daten in NodeRed ein und erhalte diese, um meine Ausgaben zu generieren (also wird „unbekannt“ einfach zu 0) – ich verwende dies auch, um die von Solarman zurückgezogenen Bitcodes zu interpretieren, damit Statuswerte und Fehlermeldungen angezeigt werden als lesbare Zeichenfolgen (z. B. 10000101 wird zu „Unterfrequenzfehler“).

....anyway, back to English for a minute.....is there actually any progress being made on this issue? Is there anything we can do to help?

Thanks!

KoblerS, 2023-11-06:
Any updates on this?

TobIYX, 2024-01-14:
Currently the only practical way if you want to continue extracting data is to remove the relay box.
This works without issues even in firmware 2.32-D1.

Too bad.

knightswsift, 2024-01-14:
Or, as said, use 1.4.3, works with 2.32 and relay-box

hucklecs88, 2024-01-28:
> Das Relais blockiert wohl den Port. Wenn man AC-seitig abschaltet und HA neustartet, kommen nach kurzer Zeit wie gewohnt die Werte.

wie kann ich den AC-seitig abschalten? Kannst du mir das näher erklären. Versuche verzweifelt den Wecchselrichter in HA zu integrieren.
Danke dir.