StephanJoubert/home_assistant_solarman #399: Historical battery discharge data missing from sofar_hyd3k-6k-es after entity rename in v1.5.0

leeky, 2023-09-12:
Since v1.5.0, I am missing battery discharge information. 

I believe this is a side effect of the typo fix in  https://github.com/StephanJoubert/home_assistant_solarman/pull/332/ which renamed "Battery daily Discharge" to "Battery Daily Discharge" (note the case change of the 'd' in Daily). 

Home Assistant created a new entity called  `sensor.solar_battery_daily_discharge_2` which now holds current battery discharge data, with historical data remaining in `sensor.solar_battery_daily_discharge`.

Gavin-A92, 2023-09-12:
Slightly tangential from your comment (well very tangential, sorry), but did you have to do anything special to make the sofar_hyd3k-6k-es parameters work with v1.5.x....when I've updated from 1.43 to 1.5.x I'm getting every sensor unavailable using the same name parameter file.....
(same happening with a custom_parameters file too)

leeky, 2023-09-13:
@Gavin-A92 Nope, apart from changing the battery daily discharge entity I was using in the energy dashboard, all the other entities are still working for me after the upgrade, so sadly I can't help you.

Gavin-A92, 2023-09-13:
thanks @leeky.....appreciate you taking the time to come back to me.......something on my end so 🙁