StephanJoubert/home_assistant_solarman #401: Integration Does Not Match Datalogger Upload Rate

alistairts, 2023-09-13:
My datalogger is set to upload every 1 min, and its data acquisition period is set to 60 sec.
After having Solarman reduce the interval to 1 min, I still only receive the data in 5 min intervals.
[image]

avanturadev, 2024-06-12:
Same here! Did you find a solution meanwhile?

alistairts, 2024-06-12:
No not from this integration, and have not received any feedback either. The solution is to use another (and better) integration by Kellerza. Here is the YouTube channel I followed to help with the setup. It requires an MQQT server, so it appears to be a round about method, but coupled with the synsynk-power-flow-card it was worth it.
https://www.youtube.com/watch?v=X_ypBi7iuJc