StephanJoubert/home_assistant_solarman #403: Solarman 1.5.x - no entities state

Wubarius, 2023-09-14:
Hi,

at the moment I use Solarman Version 1.4.3 - without Problems.
After upgrade to 1.5.0 or later 1.5.1, I don't get any state for my inverter entities.

Inverter Huayu 2000 Plus
I use custom yaml file "huayu_4mppt.yaml".
After upgrade to 1.5.x my yaml file missing under /config/custom_components/solarman/inverter_definitions/
I must be copy manuel yaml file again - but doesn't work.

Has anyone else tried custom yaml and got it working again?
Did I miss something after the update?

ChirpyTurnip, 2023-09-18:
From memory, or at least how it used to be, if you use a custom file you must call it custom_paramters.yaml. This file is placed in /config/custom_components/solarman/inverter_definitions/ location as you indicate. Unfortunately, and I have no idea why, that is where the file must be - it would be better if the files needs to be call solarman_custom_parameters.yaml and stored in /config. That way Solarman can check there for the file, use it if it exists, then carry on....

As it is now, as you have found and many others no doubt, when you update to a new version of the Solarman integration is *replaces* the entire contents of the /inverter_definitions/ location _including_ the customer_paramters.yaml file....thereby destroying your config. 

To fix you must find a copy of your old file and put it back in the /inverter_definitions/ folder...and then keep a copy of somewhere becuase you will need to do this _every_ time the integration is updated....sad....but true....

Wubarius, 2023-09-21:
Hi,
I did the following:
Contents of my "huayu_4mppt.yaml" file copied into new file "custom_paramters.yaml"
/config/custom_components/solarman/inverter_definitions/custom_parameters.yaml

Update Solarman 1.5.1 installed

Settings - Device/Service - Integration - Solarman
Both devices are displayed and point to "custom_paramters.yaml"

But no values are displayed.

Logs:
This error originated from a custom integration.

Logger: custom_components.solarman.solarman
Source: custom_components/solarman/solarman.py:61
Integration: Solarman (documentation, issues)
First occurred: 8:37:41 AM (230 occurrences)
Last logged: 8:48:42 AM

Querying [3 - 128] failed with exception [IllegalDataValueError: The value contained in the request data field is not an allowable value for the server. ]
Querying [3 - 128] failed, [1] retry attempts left
Querying [3 - 128] failed, [0] retry attempts left
Querying registers [3 - 128] failed, aborting.

ChirpyTurnip, 2023-09-21:
Just check you've spelled the customer_parameters.yaml file correctly - I had a typo. It shouldn't matter since you selected it from a drop down list, but still. 

Anyway....

Given the issues with v1.5x you'll really want to do any testing etc on v1.4.3 for now.....there's no guarantees that the problems you  now have are configuration/inverter versus the Solarman version.... So if you can use a known good version at least that removes one potential source of problems (for the most part). 

I'm sure a fix for Solarman will be coming - it is a thankless task being a integration developer....so we just patiently wait.