StephanJoubert/home_assistant_solarman #405: From v1.5 one of my two inverters goes offline after a while....

ChirpyTurnip, 2023-09-14:
Hi,

So I upgraded to v1.5 shortly after it was released, and also the v1.5.1, and in both cases one of my two inverters (both of which were rock solid on 1.4.3) goes offline after being online for a while. When it falls over is variable, but every day, at least once a day, the same inverter fall over. The *only* thing that will recover it is restarting HA - restarting the inverter, WiFi etc - nothing makes any difference - only a HA restart will fix the problem.....and then it is fine again until it stops....  :-(

Any ideas?

For now I have reverted back to an older version to see if the behaviour persists....but this is a bit of a pain.

ChirpyTurnip, 2023-09-18:
So I have now been running 1.4.3 again for several days and have not had a single outage. So this is definitely a problem with v1.5x....

ChirpyTurnip, 2023-09-18:
And this is not the same as the #367 problem.....mine work absolutely fine for a little while (a few hours maybe) and _then_ it falls over and fails to recover....

rimelis, 2023-10-27:
https://github.com/StephanJoubert/home_assistant_solarman/issues/394#issuecomment-1715877407