StephanJoubert/home_assistant_solarman #406: Scan Interval can be set in yaml, but not in GUI....

ChirpyTurnip, 2023-09-15:
A simple one for you....I've just moved from a yaml-based config to a GUI-based config as not seeing Solarman in the Devices & Services section was messing with my head. However, in having done that polling is now very, very slow! It used to poll every 15s and most of the time I got a response. Now it polls (be default) just once a minute and the updates are very very slow.....and sometimes it stops for two minutes as the reply doesn't come in time and then everything just freezes....so it can 2+ minutes for the sensor to update. 

Is there any reason why "Update Interval" cannot be included in the GUI-based configuration?

Thanks!

CP.

oxynatOr, 2023-09-16:
FYI: you have to reconfig (flash) a register. otherwise the pulling rate is ~2 or 5min

ChirpyTurnip, 2023-09-16:
This is news to me.....tell me more? It sounds like I'd need to change something in HA to increase the polling rate but also on the inverter? 

On the yaml-based config I had the polling rate set to 15 seconds and got a poll roughly every 15-20s....so my inverter can definitely go faster than once a minute....and MODBUS itself, cockroach protocol that it is, can definitely poll faster....

oxynatOr, 2023-09-17:
> This is news to me.....tell me more? It sounds like I'd need to change something in HA to increase the polling rate but also on the inverter?
>
> On the yaml-based config I had the polling rate set to [...]

if i get the other ppl right. you have to contact deye.  :'(

Yes, the software inside the inverter is set to 5min ( or 2min) this means, all the registers get filled in a interval of X-Minutes. 
at least, that is how i understood it. 

i did check the PDFs i have, but i was jot able to find the register and value. 
someone try it over the AT Codes. but failed. 

i'm also searching for way to crack this :)

ChirpyTurnip, 2023-09-17:
I have a pair of Solis Inverters and I'm running with a custom_parameters file (I can supply that if anyone wants to add it to the repository) and then I use NodeRed to bust up the multi-bit registers into entities that can report directly what mode(s) and alarm(s) are in effect. 

But with the yaml config I could poll faster....so for my inverter at least I don't think I need to do anything except for telling HA to poll faster....but that part I _cannot_ do in the Solarman GUI 'configure' screen....so I still think this should be added to the integration to allow this to be set (possible with a warning that mileage may vary depending on inverter).