StephanJoubert/home_assistant_solarman #407: SunM80g3 errors Firmware MW3_16U_5406_2.32-D1

glasoline, 2023-09-15:
Hi,

thanks for your work. Sadly the integration does not longer work for me. Here are the logfiles:

`2023-09-15 09:35:09.928 DEBUG (SyncWorker_3) [custom_components.solarman.solarman] Starting to query for [1] ranges...
2023-09-15 09:35:09.928 DEBUG (SyncWorker_3) [custom_components.solarman.solarman] Querying [1 - 125]...
2023-09-15 09:35:09.929 INFO (SyncWorker_3) [custom_components.solarman.solarman] Connecting to solarman data logger 192.168.69.220:8899
2023-09-15 09:35:09.941 DEBUG (SyncWorker_3) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=49, family=2, type=1, proto=6, laddr=('192.168.69.86', 45184), raddr=('192.168.69.220', 8899)>
2023-09-15 09:35:09.942 DEBUG (SyncWorker_3) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 fe 00 70 0d 27 eb 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 01 00 7d d4 2b 7c 15
2023-09-15 09:35:10.044 DEBUG (SyncWorker_3) [custom_components.solarman.solarman] RECD: 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 36 5f 32 2e 33 32 2d 44 31 0d 0a 0d 0a
2023-09-15 09:35:10.044 DEBUG (SyncWorker_3) [custom_components.solarman.solarman] frame_len does not match payload_len.
2023-09-15 09:35:10.044 WARNING (SyncWorker_3) [custom_components.solarman.solarman] Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid start or end values]
2023-09-15 09:35:10.060 INFO (SyncWorker_3) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192.168.69.220:8899
2023-09-15 09:35:10.467 WARNING (SyncWorker_3) [custom_components.solarman.solarman] Querying [1 - 125] failed, [1] retry attempts left
2023-09-15 09:35:10.467 INFO (SyncWorker_3) [custom_components.solarman.solarman] Connecting to solarman data logger 192.168.69.220:8899
2023-09-15 09:35:10.475 DEBUG (SyncWorker_3) [custom_components.solarman.solarman] Socket setup completed... <socket.socket fd=49, family=2, type=1, proto=6, laddr=('192.168.69.86', 45196), raddr=('192.168.69.220', 8899)>
2023-09-15 09:35:10.475 DEBUG (SyncWorker_3) [custom_components.solarman.solarman] SENT: a5 17 00 10 45 d1 00 70 0d 27 eb 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 00 01 00 7d d4 2b 4f 15
2023-09-15 09:35:10.581 DEBUG (SyncWorker_3) [custom_components.solarman.solarman] RECD: 41 54 2b 59 5a 43 4d 50 56 45 52 3d 4d 57 33 5f 31 36 55 5f 35 34 30 36 5f 32 2e 33 32 2d 44 31 0d 0a 0d 0a
2023-09-15 09:35:10.581 DEBUG (SyncWorker_3) [custom_components.solarman.solarman] frame_len does not match payload_len.
2023-09-15 09:35:10.581 WARNING (SyncWorker_3) [custom_components.solarman.solarman] Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid start or end values]
2023-09-15 09:35:10.584 INFO (SyncWorker_3) [custom_components.solarman.solarman] Disconnecting from solarman data logger 192.168.69.220:8899
2023-09-15 09:35:11.081 WARNING (SyncWorker_3) [custom_components.solarman.solarman] Querying [1 - 125] failed, [0] retry attempts left
2023-09-15 09:35:11.083 WARNING (SyncWorker_3) [custom_components.solarman.solarman] Querying registers [1 - 125] failed, aborting.
2023-09-15 09:35:11.084 DEBUG (SyncWorker_3) [custom_components.solarman.sensor] No value recorded for PV1 Voltage
2023-09-15 09:35:11.088 DEBUG (SyncWorker_4) [custom_components.solarman.sensor] No value recorded for PV2 Voltage
2023-09-15 09:35:11.089 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for PV1 Current
2023-09-15 09:35:11.091 DEBUG (SyncWorker_5) [custom_components.solarman.sensor] No value recorded for PV2 Current
2023-09-15 09:35:11.098 DEBUG (SyncWorker_3) [custom_components.solarman.sensor] No value recorded for Daily Production
2023-09-15 09:35:11.099 DEBUG (SyncWorker_2) [custom_components.solarman.sensor] No value recorded for Daily Production 1
2023-09-15 09:35:11.102 DEBUG (SyncWorker_4) [custom_components.solarman.sensor] No value recorded for Daily Production 2
2023-09-15 09:35:11.105 DEBUG (SyncWorker_7) [custom_components.solarman.sensor] No value recorded for Total Production
2023-09-15 09:35:11.106 DEBUG (SyncWorker_1) [custom_components.solarman.sensor] No value recorded for Total Production 1
2023-09-15 09:35:11.112 DEBUG (SyncWorker_2) [custom_components.solarman.sensor] No value recorded for Total Production 2
2023-09-15 09:35:11.114 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for Active Power Regulations
2023-09-15 09:35:11.115 DEBUG (SyncWorker_7) [custom_components.solarman.sensor] No value recorded for AC Voltage
2023-09-15 09:35:11.120 DEBUG (SyncWorker_1) [custom_components.solarman.sensor] No value recorded for Grid Current
2023-09-15 09:35:11.121 DEBUG (SyncWorker_2) [custom_components.solarman.sensor] No value recorded for AC Output Frequency
2023-09-15 09:35:11.123 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for Grid Voltage Upp Limit
2023-09-15 09:35:11.124 DEBUG (SyncWorker_5) [custom_components.solarman.sensor] No value recorded for Grid Voltage Lower Limit
2023-09-15 09:35:11.124 DEBUG (SyncWorker_3) [custom_components.solarman.sensor] No value recorded for Grid Frequency Upper Limit
2023-09-15 09:35:11.133 DEBUG (SyncWorker_2) [custom_components.solarman.sensor] No value recorded for Grid Frequency Lower Limit
2023-09-15 09:35:11.134 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for Overfrequency And Load Reduction Starting Point
2023-09-15 09:35:11.134 DEBUG (SyncWorker_5) [custom_components.solarman.sensor] No value recorded for Overfrequency And Load Reduction Percentage
2023-09-15 09:35:11.136 DEBUG (SyncWorker_1) [custom_components.solarman.sensor] No value recorded for ON-OFF Enable
2023-09-15 09:35:11.136 DEBUG (SyncWorker_2) [custom_components.solarman.sensor] No value recorded for Island Protection Enable
2023-09-15 09:35:11.137 DEBUG (SyncWorker_7) [custom_components.solarman.sensor] No value recorded for Overfrequency&Load-shedding Enable
2023-09-15 09:35:11.137 DEBUG (SyncWorker_5) [custom_components.solarman.sensor] No value recorded for Running Status
2023-09-15 09:35:11.138 DEBUG (SyncWorker_1) [custom_components.solarman.sensor] No value recorded for Total AC Output Power (Active)
2023-09-15 09:35:11.142 DEBUG (SyncWorker_4) [custom_components.solarman.sensor] No value recorded for Radiator Temperature
2023-09-15 09:35:11.143 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for Inverter ID
2023-09-15 09:35:11.143 DEBUG (SyncWorker_7) [custom_components.solarman.sensor] No value recorded for Hardware Version
2023-09-15 09:35:11.144 DEBUG (SyncWorker_3) [custom_components.solarman.sensor] No value recorded for DC Master Firmware Version
2023-09-15 09:35:11.144 DEBUG (SyncWorker_4) [custom_components.solarman.sensor] No value recorded for AC Version. Number
2023-09-15 09:35:11.144 DEBUG (SyncWorker_0) [custom_components.solarman.sensor] No value recorded for Rated Power
2023-09-15 09:35:11.145 DEBUG (SyncWorker_5) [custom_components.solarman.sensor] No value recorded for Communication Protocol Version
2023-09-15 09:35:11.145 DEBUG (SyncWorker_3) [custom_components.solarman.sensor] No value recorded for Start-up Self-checking Time 
2023-09-15 09:35:11.146 DEBUG (SyncWorker_2) [custom_components.solarman.sensor] No value recorded for Update Time
2023-09-15 09:35:11.146 DEBUG (SyncWorker_7) [custom_components.solarman.sensor] No value recorded for Soft Start Enable
2023-09-15 09:35:11.147 DEBUG (SyncWorker_5) [custom_components.solarman.sensor] No value recorded for Power Factor Regulation
2023-09-15 09:35:11.148 DEBUG (SyncWorker_1) [custom_components.solarman.sensor] No value recorded for Restore Factory Settings`

drevil54, 2023-09-18:
Same for me, with Firmware 1.57 and without relay it was working fine. After connecting the relay, the inverter has auto updated to version 2.32 and home assistant plugin not working anymore.

Sir-Udo, 2023-09-19:
Hi all, at me the same problem. Version 2.32 and no Information in HA! 
Please Help
Udo

Kevin1904, 2023-09-22:
I guess the relay device is using the modbus, so the port 8899 is already being used?

knightswsift, 2023-09-22:
Seems so. If you switch off AC and restart HA, after few minutes the data appears in HA.

Fereauje, 2023-09-25:
Same problem here. Restarted the inverter, restartet HA, but no data appeared.

DieSeef, 2023-09-26:
Same issue here. Downgrade to firmware version 1.57 made the data available again.

knightswsift, 2023-09-26:
1.4.3 still works, no need to downgrade logger firmware

Fereauje, 2023-09-26:
Solarman 1.5.1 and FW MW3_16U_5406_2.32-D1 doesn't work. Or maybe the issue comes from the relay that blocks the communication???

DieSeef, 2023-09-26:
> 1.4.3 still works, no need to downgrade logger firmware

Thanks for the info, struggeling how to downgrade the solarman to 1.4.3. Is there a instruction available? Redownload old version via GUI does not work.

saz, 2023-09-26:
@DieSeef What's the issue with downgrading via the GUI?

I can confirm, that 1.4.3 is working with the `deye_2mppt.yaml` from 1.5.1, firmware 2.32-D1 and the relay box attached.

DieSeef, 2023-09-26:
> @DieSeef What's the issue with downgrading via the GUI?
> 
> I can confirm, that 1.4.3 is working with the `deye_2mppt.yaml` from 1.5.1, firmware 2.32-D1 and the relay box attached.

Hi saz, within HA and HACS on the solarman integration, there is the option to redownload the integration. In the version dropdown i can pick 1.4.3 and redownload it. Something happens but it jumps back to the solarman integration page, that#s it. Of course tried to reload HA afterwards but the integration still shows version 1.5.1.

saz, 2023-09-26:
@DieSeef After redownloading, there should be a notification in HA. You need to restart (not reload) HA completely.

knightswsift, 2023-09-26:
You can delete Solarman Integration, redownload 1.4.3 and copy deye_2mppt.yaml from 1.5.1 and add in in const.py.
All entities should still work

DieSeef, 2023-09-26:
> You can delete Solarman Integration, redownload 1.4.3 and copy deye_2mppt.yaml from 1.5.1 and add in in const.py. All entities should still work

Ok, deleted the integration and reinstalled it with version 1.4.3. Seems like the integration still shows the latest version 1.5.1
[image].

As i have already downgraded the firmware of both inverters, i will leave the setup as it is for now as it is working. 
Anyway, thank you for your help!

Sir-Udo, 2023-09-27:
> Ok, deleted the integration and reinstalled it with version 1.4.3. Seems like the integration still shows the latest version 1.5.1 [image].
>
> As i have already downgraded the firmware of both [...]

Hi, thank you, youre workaround works. 
Only the DEY2PPT is missing, o used the 4ppt and delete 3+4

Fereauje, 2023-09-27:
> > > You can delete Solarman Integration, redownload 1.4.3 and copy deye_2mppt.yaml from 1.5.1 and add in in const.py. All entities should still work
> > 
> > 
> > Ok, deleted the integration and reinstalled it with version 1.4.3. Seems like the integration still shows the latest version 1.5.1 [image].
> > As i have already downgraded the firmware of both inverters, i will leave the setup as it is for now as it is working. Anyway, thank you for your help!
> 
> Hi, thank you, you're workaround works fine. Only the DEY2PPT is missing, I used the 4ppt and delete 3+4

The downgrade to 1.4.3 also worked for me. Thx a lot!

Dummy0815, 2023-10-01:
I think I found the cause to this error. For Details see https://github.com/StephanJoubert/home_assistant_solarman/discussions/271#discussioncomment-7158802

Kleinheribert, 2023-10-02:
Got the same issue: Two Inverters at home, one "old" withoud relay and FW 1.57 works without issues. The new one got the Relay and the Firmware MW3_16U_5406_2.32-D1. Got it working for at least one day, now weeks without any data. Set up multiple servers but no change. If anyone of you doesn't know about deye inverters: you could set up additional parameters via the website http://xxx.xxx.xx.x/config_hide.html.

MartinGerhards, 2023-11-11:
Hello everyone. Have a deye m80 with relaybox (firmware MW1_16U_5406_2.32-D1). I deleted the Solarman integration and then loaded version 1.4.3. Can you explain to me again exactly how I copy the deye_2mppt_yaml from the 1.5.1 version to the 1.4.3 version?

After loading the Solarman 1.4.3, I did the following: custom_components > const.py > typed the deye_2mppt.yaml (not pasted or copied). After restarting the home Assistant, I was also able to select the 2mppt when configuring. Unfortunately, no device is detected.

Can you help me ???

Many thanks to you.

b-reich, 2023-11-21:
Got my relais last weekend. Since them no data in home assistant. Trying downgrading via hacs when the sun is up.

bhuebschen, 2024-04-20:
i played a bit with the yaml files, after i saw that the deye_hybrid.yaml fetches some (incorrect) data

i made a copy of my deye_2mppt.yaml and changed 

```
requests:
  - start: 0x0001
    end:  0x007D
    mb_functioncode: 0x03
```

to 

```
requests:
  - start: 0x0003
    end: 0x0070
    mb_functioncode: 0x03
  - start: 0x0096
    end: 0x00f8
    mb_functioncode: 0x03
```

i see "Querying registers [150 - 248] failed, aborting" in the log files, but now it seems to work:

[image]