StephanJoubert/home_assistant_solarman #408: Gauges come and go with latest update

mcderd-coder, 2023-09-16:
Hi, I have just updated to 1.5.1 after using your code for many months without issue. Since the upgrade the gauges I have to monitor my solar work sometimes, then they change to messages that say "Entity is non-numeric:" if I just leave it, it seems like when it next refreshes the gauges all work, then a few minutes later I just see the Non-numeric messages.

Also, I have some automations set to inform me of the battery percentage and solar produced throughout the day.

So for example I would get an alert to say solar produced is 5Kwh, then later 10Kwh and so on, now every time I get an alert they are all repeated, so when I get a 15Kwh alert I also get a 10Kwh and a 5Kwh alert, this did not happen before the update.

Screen grabs attached

Many Thanks

[image]
[image]
[image]

samdunne, 2023-09-18:
Same thing happening to me

EspeeFunsail, 2023-09-18:
Also getting this. All values become "NaN" randomly until it updates. Happening after latest update.

ChirpyTurnip, 2023-09-18:
This could be related to my issue: [405](https://github.com/StephanJoubert/home_assistant_solarman/issues/405)

It seems in your case the connection is recovered (and hence data comes back) whereas in my case (perhaps because I have two inverters) the one that goes offline does not come back and everything stays as non-data / NaN. 

If this happens again look to see if the integration is actually connected to the inverter or offline.....In my case I can simply check my display:

[image]

Mileage may vary.....but I went back to 1.4.3 and everything is solid again. So for now I'm ignoring the option to upgrade...

mcderd-coder, 2023-09-18:
Yes, the latest version seems a bit flaky. I have also reverted to 1.4.3 and all is working as normal.

EspeeFunsail, 2023-09-18:
How did you revert back to 1.4.3?

mcderd-coder, 2023-09-18:
I just installed the day befores backup as I had also updated HA, so I was not sure if the problem was with the solarman code or HA

ChirpyTurnip, 2023-09-18:
> How did you revert back to 1.4.3?

Just got back to HACS, remove the integration, then re-add it. When putting it back wait for the drop down list to populate and then you can select the version you want (e.g. v.1.4.1, 1.4.3, 1.5, 1.5.1 etc). The one you select is the one that installs....

Just note that it will then prompt you to upgrade to the newest version - ignore that (for a long time until there is a newer version to try).

ChirpyTurnip, 2023-09-18:
> I just installed the day befores backup as I had also updated HA, so I was not sure if the problem was with the solarman code or HA

It's Solarman - any version of HA with 1.5x doesn't work, any version of HA with 1.4.3 seems to work fine....so that's where my money is....  :-)

mcderd-coder, 2023-09-18:
Thanks for that info, next time I will uninstall and reinstall via HACS 😉

EspeeFunsail, 2023-09-18:
> Just got back to HACS, remove the integration, then re-add it. When putting it back wait for the drop down list to populate and then you can select the version you want (e.g. v.1.4.1, 1.4.3, 1.5, [...]

Ok thanks. Does this keep all existing settings the same? Don't want it to break my current sensors from Solarman.

ChirpyTurnip, 2023-09-18:
It did for me and I have two inverters so double the problems that you would have if it went wrong. The other option, that you may like more,  is to go to HACS > Solarman > Triple Dots in the top right > Re-download. Then select version (1.4.3) and hit download. 

Restart HA and you should be on the old version....

ilkka-rautiainen, 2023-09-21:
Same for me. The reason seems to be that whenever it doesn't get new values from the inverter the values are undefined, before the last received values were used until new values were received. I tend to get some short offline periods during the day and the previous values were quite good approximation.

mcderd-coder, 2023-10-04:
> Just updated to latest version and this issue is still there. Reverting back to 1.4.3 again...

What version did you update to? Was it 1.5.1 as that is the broken version, I wouldn't expect this to work until 1.5.2 is released.

EspeeFunsail, 2023-10-04:
Sorry, you are correct. I saw the update in HACS and thought there was a new update.

Testmangh, 2023-10-05:
I was also having problem, at night all the gauges would go out and the graph showing the current energy would not mark zero... Reverted back to 1.4.3 não everything works, I lost some entities but its fine...

> It did for me and I have two inverters so double the problems that you would have if it went wrong. The other option, that you may like more, is to go to HACS > Solarman > Triple Dots in the top [...]

[image]