StephanJoubert/home_assistant_solarman #416: Zero values

lukazareba, 2023-09-29:
Hi, is there any method so that when the inverter is turned off, the variable values like current, voltage, power, etc. change to 0 and the daily production value clears at midnight and the total production is displayed all the time?

monoboy-hh, 2023-10-05:
With 1.4.3 the old values or at least a value would still be present. Changing to unavailable interferes with many graphs / cards.
After the protocol issues of 1.5.0 have been fixed, I have to downgrade to 1.4.3 again :(
[image]

monoboy-hh, 2023-10-05:
I have to correct myself, even with 1.4.3 but hass 2023.10.0 old values are not kept. So this must be a change how hass handles statistics now. 
with hass 2023.08.x and solarman 1.4.3 there was no "unknown" numerical value.

monoboy-hh, 2023-10-06:
working as expected with 1.4.3 and 2023.10.x
with 1.5.1, when the inverter goes offline, most values are set to unknown / unavailable / zero. resulting in graphs like this (radiator temperature)
[image]
(instead of letting the temperature drop to 0 / NaN / unknown I set it to the outside temperature to have at least some numerical value)