StephanJoubert/home_assistant_solarman #417: Getting enery consumption and other parameter from Deye 6K G03 (Deye_string.yaml)

carlesco, 2023-10-01:
Dear community,

I'm not able to get those parameters from my 3-phaseDeye Sun-6K-G03

- Energy Consumpiton
- Energy Feed-in
- Energy Brought

Those parameters are present in the deye_hybrid.yaml file but not on the deye_string.yaml, checking the following file
[https://domotica.solar/wp-content/uploads/2023/03/sunsynk%20modbus.pdf](url)

I'm not able to find the registers for my inverter, as they change from the hybrid version.

is anybody able to guide me how to get this info from my inverter?

million thanks!
carles

pawelsn, 2023-10-20:
If you will downgrade to version 1.4.3 then it will work out of the box....

https://github.com/StephanJoubert/home_assistant_solarman/issues/388

carlesco, 2023-10-20:
Thanks pawelsn for your reply. I'm already in 1.4.3, and it works, however my problem as I indicated above it's that the yaml definition for deye string does not contain the following parameters

- Energy Consumpiton
- Energy Feed-in
- Energy Brough

That in fact are present in the hybrid definition