StephanJoubert/home_assistant_solarman #423: Config flow could not be loaded: {"message":"Invalid handler specified"}

admkhe, 2023-10-12:
Hi,

When trying to add a solarman device, I get this:

[image]

That's even before filling in anything.

thanks in adv

dj9717, 2023-10-17:
hello,
the same problem here too. 
Homeassistent 2023.10.3 [docker-compose](https://www.home-assistant.io/installation/linux#docker-compose)
Solarman Integration [1.5.1](https://github.com/StephanJoubert/home_assistant_solarman/releases/tag/1.5.1)
please for support
gg

jweea, 2023-11-04:
Try change it back
#https://github.com/StephanJoubert/home_assistant_solarman/commit/a1a5de42f1033b7a0403ad4581ba6fe4b90f6e36#diff-bbed67ea80599e4ac504d7f56d3475b028def737c2fd65a11d3aad364fc1711dL8-L14

KoblerS, 2023-11-06:
unfortunately I do receive the same error, does anyone got a solution for this?

asgaardwro, 2023-11-26:
got the same error :(

chrisvanmeer, 2023-12-10:
That solved my issue.

commit 3fc5bab4 (2024-02-26): Merge pull request #455 from AnnoyingTechnology/patch-1

Update const.py, Fixes #423
commit 7da48d67 (2024-02-26): Update const.py, Fixes #423

Fixes: {"message":"Invalid handler specified"}
closed 2024-02-26 by StephanJoubert