StephanJoubert/home_assistant_solarman #424: Unknown values SN 27xxxxxxxx

ondfrukt, 2023-10-13:
I have a new Deye sun-15K-G05 inverter with a solarman LSW3-wifi logger with an serial number that begins with 27. I have spend days of googling and testing without succeeding.

Tried all available inverter definition files but always with unknown values of every entity.

My HA core loggfile says:
"failed with exception [V5FrameError: V5 frame does not contain a valid Modbus RTU frame]"

I have of course tried to figure out what that means several times but always end up, deep into forums with a subject I dont understand.

Any suggestion how I can proceed?

carlesco, 2023-10-13:
Downgrade integration from 1.5 to 1.43

ondfrukt, 2023-10-14:
> Downgrade integration from 1.5 to 1.43

Wow! It worked perfectly! Thanks a lot @carlesco!

But what is the difference betwen 1.4.3 and later versions?

maitock, 2023-12-13:
Hi, I'm trying to integrate my Hybrid 8K deye inverter, and I'm facing the same issue. My logger SN is: 2788286194, I tested 1.5.1, 1.5.0 and 1.4.3 versions of the integration module. all of these are reporting the same wrong date. For instance, reported Temperature  is -100ºC, some integers are reported as: 6,553.5 kWh and so on....

Any idea?

thanks in advance for reading

hajokim, 2024-02-02:
> Hi, I'm trying to integrate my Hybrid 8K deye inverter, and I'm facing the same issue. My logger SN is: 2788286194, I tested 1.5.1, 1.5.0 and 1.4.3 versions of the integration module. all of these [...]

Hi, I solved this problem, with manual configuration in configuration.yaml. But first delete all entities and configuration. I replaced in this line "lookup_file: deye_hybrid.yaml" with "lookup_file: deye_sg04lp3.yaml". And that super well!