StephanJoubert/home_assistant_solarman #430: Deye Inverter not working

miastekpl, 2023-10-20:
Hello, after the latest update of the solarman add-on, I am not getting results to HomeAssistant. I own a DEYE inverter. Can I count on fixing this problem ?

pawelsn, 2023-10-20:
downgrade to 1.4.3

https://github.com/StephanJoubert/home_assistant_solarman/issues/388

miastekpl, 2023-10-20:
Can you write me how to perform a downgrade ?
I'm just starting my adventure with HomeAssistant

Steinbichler, 2023-10-28:
Restore a backup with 1.4.3.
Regards

frops83, 2023-11-09:
Hi, my inverter has gone to n/a too since last nights (supervisor?) update. I Had the problem with 1.5.1 too and then downgraded to 1.4.3 a few weeks ago. Since then everything went fine until Last night. 
I tried reloading and rebooting without success.