StephanJoubert/home_assistant_solarman #435: Regularly getting disconnected status on my Homeassistant Solarman Intergration (Deye SUN-12K-SG04LP3-EU)

Jan-Ole-Menz, 2023-10-25:
The integration goes from working perfectly to status_connection: disconnected without any interaction from my side.

In the HA logs I find this: 
WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 89] failed with exception [OSError: [Errno 24] No file descriptors available]

Restarting HA fixes the issue temporarly until the next disconnected (trice this week).

I'm running Solarman integration 1.5.1 via HACS

Jan-Ole-Menz, 2023-10-25:
Thanks for the info, hav downgraded to 1.4.3, if it occurs again will post here then.

etam, 2023-10-28:
I also have this problem. It might be related to a change in pysolarmanv5: https://github.com/jmccrohan/pysolarmanv5/blob/main/CHANGELOG.md#v300---2023-05-21

> Prior to this, the socket was implicitly closed when the PySolarmanV5 object was deferenced. Many dependent applications re-instantiate a new object for each poll. These applications will need to either remain on v2.4.0, or upgrade to v3.0.0 and ensure disconnect() is called to close the connection cleanly.

kasper0, 2023-11-08:
@Jan-Ole-Menz, @etam Could you verify if the issue occurs when auto_reconnect is set to false?

Jan-Ole-Menz, 2023-11-09:
Since I downgraded to 1.4.3 it's running perfectly.

etam, 2023-11-09:
I can try. I run the latest version and I'm restarting HA once in a while.

You're asking to edit this line? https://github.com/StephanJoubert/home_assistant_solarman/blob/aed4c7aad3b5c5677671ea2b74e05415a13ff10b/custom_components/solarman/solarman.py#L39

kasper0, 2023-11-09:
Yes, this line, thank you. It seems that disabling auto-reconnect mechanism prevents leaking file descriptors.

etam, 2023-11-14:
So far so good. I checked the HA process with lsof and it seems that there are no leaks. Current uptime is 2 days, though. I'll report again in a few days.

OneScuzzie2k, 2023-11-17:
> I can try. I run the latest version and I'm restarting HA once in a while.
> 
> You're asking to edit this line?
> 
> https://github.com/StephanJoubert/home_assistant_solarman/blob/aed4c7aad3b5c5677671ea2b74e05415a13ff10b/custom_components/solarman/solarman.py#L39

I have done this, and today after 3 days mine has gone down again, rolling back to 1.4.3

kasper0, 2023-11-17:
Mine has been working correctly since I turned off the auto-reconnect feature a week ago.

@OneScuzzie2k Could you share the HA log?

Reinqusito, 2023-12-06:
same issue with 1.5.1 (never a problem with previous version).
Downgraded to 1.4.3 and everything back to work perfectly.

etam, 2023-12-06:
Meanwhile my instance has almost 24 days of uptime and no problems so far.

For the record: As written in [comment above](https://github.com/StephanJoubert/home_assistant_solarman/issues/435#issuecomment-1803654821), I have changed `auto_reconnect` to `False`.

Jay66Bee, 2023-12-09:
> In HA under HACS, click on the vertical ... and select redownload, in the pop-up you will find select version you open the dropdown and select 1.4.3 and finally click download

I reloaded to 1.4.3 and reloaded HA but integration still showing 1.5.1 is that normal or did it fail to roll back?

Jay66Bee, 2023-12-13:
Changing auto_reconnect=True to False for my KStar inverter did not stop the disconnection.

I have successfully rolled back to 1.4.3 

Any benefits of 1.5.X and thereafter will have to wait until it is proven to work.

JPEloff, 2024-03-03:
Hi. My Solarman integration keeps disconnecting. I change auto reconnect to false but no luck. I am trying to downgrade from 1.5.1 to 1.4.3 but I get this message : "The version selector will be removed in a future release. If you need to install a specific version, you can do so by using the service call for update entities." How do I downgrade? Please help.

kilian1024, 2024-05-11:
> Hi. My Solarman integration keeps disconnecting. I change auto reconnect to false but no luck. I am trying to downgrade from 1.5.1 to 1.4.3 but I get this message : "The version selector will be [...]

Hi, had the same issue... Hope you already got this. In case someone else stumbles over this and needs help (took me an hour to find out...):
Go to Developer Tools -> Services and use the service update.install

Choose update.solarman_integration_update as the Entity you want to update (make sure you use the entity, not the device). If you can not find the update entity, make sure that experimental features are enabled in the HACS Configuration.

Check the Version box and insert v1.4.3 and fire away. You will get an error if you got something wrong but no "update successful" screen or something like that if you got it right.

Wait a Minute and restart home assistant. If you have auto updates on the integration enabled, you can see that it worked because home assistant found the update to the newer version. Make sure to skip this so you do not update again.

PV-Joe, 2024-05-11:
Delete all remaining files with File editor after uninstalling and before downgrading to 1.4.3.

kilian1024, 2024-05-11:
> Delete all remaining files with File editor after uninstalling and before downgrading to 1.4.3.

Could you get more detailed on how to do that?

PV-Joe, 2024-05-11:
> > Delete all remaining files with File editor after uninstalling and before downgrading to 1.4.3.
> 
> Could you get more detailed on how to do that?

Go to folder:
custom components > Solarman

Delete all remaining files and restart before installing 1.4.3.

kilian1024, 2024-05-11:
yeah, but going the way with updating over the developer-Tools, you have to have the newer version already installed... So I can't uninstall it before? Or is there a possibility to directly install the older Version? Over HACS, it seems like it is not possible

PV-Joe, 2024-05-11:
If you have the 1.4.3 files already then:
1. Delete the files described above
2. Restart 
3. Manual installation of 1.4.3.
4. Restart
5. Configure

Pigi-102, 2024-05-17:
I have this problem too. Same symptoms. Rollback to 1.4.3 seems a valuable workaround.

PV-Joe, 2024-09-02:
Go and install this version:

https://github.com/davidrapan/ha-solarman

More Sensors
5 sec Intervall

It works perfectly with Deye!

bert1111, 2024-09-02:
Yes I use it too. Works great with my sofar.

mario37359, 2024-09-02:
Thanks for your fast help. Installed the new version but doesnt work. Any hints on which config file to choose? 
[image]

PV-Joe, 2024-09-02:
Ask here to get full support:

https://github.com/davidrapan/ha-solarman/discussions