StephanJoubert/home_assistant_solarman #436: greensun_hybrid.yaml  Please add this new inverter definition

xenfasa, 2023-10-25:
New Solar Inverter definition I made and use for my own inverter

greenshun_hybrid.yaml

`requests:
 - #Alarm, PV, Bat, BMS
   start: 0x3100
   end: 0x314E
   mb_functioncode: 0x03

 - #Analog
   start: 0x3150
   end: 0x3181
   mb_functioncode: 0x03

 - #Inverter Info
   start: 0x3190
   end: 0x319C
   mb_functioncode: 0x03

 - #OTHER ANALOG #PV Power
   start: 0x16A2
   end: 0x16A2
   mb_functioncode: 0x03

 - #Other Power
   start: 0x2282
   end: 0x228B
   mb_functioncode: 0x03

parameters:
 - group: solar
   items: 
    - #DONE
      name: "PV1 Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 1
      registers: [0x3132]
      icon: 'mdi:solar-power'

    - #DONE
      name: "PV2 Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 1
      registers: [0x3135]
      icon: 'mdi:solar-power'

    - #DONE
      name: "PV1 Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x3130]
      icon: 'mdi:solar-power'

    - #DONE
      name: "PV2 Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
[... 846 lines cut ...]
      state_class: "measurement"      
      uom: "W"
      scale: 1
      rule: 2
      registers: [0x228B]
      icon: 'mdi:home-lightning-bolt'

 - group: Alert
   items: 
    - name: "Alert Orig"
      class: ""
      state_class: ""      
      uom: ""
      scale: 1
      rule: 6
      registers: [0x0065,0x0066,0x0067,0x0068,0x0069,0x006A]

    - name: "Alert Test"
      class: ""
      state_class: ""      
      uom: ""
      scale: 1
      rule: 1
      registers: [0x3100]
      `

josemonino, 2023-10-28:
Thanks a lot. 

How did you do it? Did you already know the registers before making it? 
It would be wonderful if there were any step by step on how to do it...

I would like to do it for my Sofar inverter.

xenfasa, 2023-10-29:
Hi,

I bugged the company until they sent me a spreadsheet with some chinese and
register information.  I still had to decode a lot and I only mapped the
data that I was needing.
There is still some info i'm not sure of like alarm codes and I did not add
info for a second battery since I don't have one yet.
Attached is the version of the spreadsheet I made more readable but still
not easy to figure out... The original was worse.... but this gives you an
idea of what I started with.

I also found some python scripts to probe the registers... Some worked and
some did not.  Creating my own inverter definition with ideas from the
registers in the spreadsheet and seeing what numbers I got back eventually
got me something useful.    I would ask the company for such a spreadsheet
and data.

Ben

cyh0928, 2024-03-29:
This is an agreement with megarevo. Greensun’s inverter is also manufactured by megarevo.

xenfasa, 2024-03-30:
What agreement where. I don't see it or am I missing something?

cyh0928, 2024-03-30:
> 哪里有什么协议。我没有看到它还是我错过了什么？
> [……](#)

Your greensun inverter is manufactured by megarevo, so this agreement is the same as that of the invt inverter.
[image]