StephanJoubert/home_assistant_solarman #438: Not reading anymore from SOLARMAN LW3

Mak69, 2023-10-26:
Inverter ZCS 1PH HYD 6000 ZSS HP + SOLARMAN USB LW3 stick.
Yesterday my LW£ performed a self update to firmware version LSW3_15_270A_1.68
Since yesterday Home Assistant can't read anymore the inverter parameters.
I was using hyd-zss-hp-3k-6k.yaml file.
If I go into the solarman integration configuration menu I can't find anymore this file even if it's present in /custom_components/solarman/inverter_definitions.
What I'm doing wrong?
Thx in advance

Mak69, 2023-10-26:
Debug enabled
**First message**

```
Questo errore ha avuto origine da un'integrazione personalizzata.

Logger: custom_components.solarman.solarman
Source: custom_components/solarman/solarman.py:88
Integration: Solarman (documentation, issues)
First occurred: 11:27:07 (10 occurrences)
Last logged: 11:29:07

Querying [1024 - 1067] failed with exception [OSError: [Errno 24] No file descriptors available]
```

**Second message**

```
Questo errore ha avuto origine da un'integrazione personalizzata.

Logger: custom_components.solarman.solarman
Source: custom_components/solarman/solarman.py:91
Integration: Solarman (documentation, issues)
First occurred: 11:27:07 (10 occurrences)
Last logged: 11:29:07

Querying [1024 - 1067] failed, [1] retry attempts left
Querying [1024 - 1067] failed, [0] retry attempts left
```

**Third message**

```
Questo errore ha avuto origine da un'integrazione personalizzata.

Logger: custom_components.solarman.solarman
Source: custom_components/solarman/solarman.py:96
Integration: Solarman (documentation, issues)
First occurred: 11:27:07 (5 occurrences)
Last logged: 11:29:07

Querying registers [1024 - 1067] failed, aborting.
```

wjt333, 2023-11-04:
check this integration:

https://gist.github.com/MatteCarra/3524c1c29f4ef58028ae3d182694490f

Nusserdt, 2023-11-22:
I think I have the same issue:

[image]

I have no plan how to fix.

waaslandwolf, 2023-12-26:
Same issue here. Firmware is now LSW3_15_270A_1.68 and the Solarman integration in Home Assistant is now broken. I guess we'll have to figure out the registers all over again... :-(

waaslandwolf, 2023-12-26:
>I was using hyd-zss-hp-3k-6k.yaml file. If I go into the solarman integration configuration menu I can't find anymore this file even if it's present in /custom_components/solarman/inverter_definitions. What I'm doing wrong? Thx in advance

You could try to rename (or even better, make a copy of) your "hyd-zss-hp-3k-6k.yaml" file to "custom_parameters.yaml", and then configure that in the SolarMan integration. Restart Home Assistant, and with a bit if luck this could work.

Picard2002, 2023-12-30:
Does anyone have the custom_parameters.yaml file for a sofar 5ktlm g3?

wjt333, 2023-12-31:
> Does anyone have the custom_parameters.yaml file for a sofar 5ktlm g3?

this works for 11ktlx-g3
https://gist.github.com/MatteCarra/3524c1c29f4ef58028ae3d182694490f

waaslandwolf, 2024-01-04:
[custom_parameters.yaml-working.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/13833789/custom_parameters.yaml-working.txt)
This one works with Sofar 8.8KTL-X. You can try to see if it works with your inverter.

DynamiteAndric, 2024-03-09:
Did someone have a working configuration YAML for LSW3_15_270A_1.68 firmware?

Vityok01030103, 2024-03-14:
> Отладка включена **Первое сообщение**
> 
> ```
> Questo errore ha avuto origine da un'integrazione personalizzata.
> 
> Logger: custom_components.solarman.solarman
> Source: custom_components/solarman/solarman.py:88
> Integration: Solarman (documentation, issues)
> First occurred: 11:27:07 (10 occurrences)
> Last logged: 11:29:07
> 
> Querying [1024 - 1067] failed with exception [OSError: [Errno 24] No file descriptors available]
> ```
> 
> **Второе сообщение**
> 
> ```
> Questo errore ha avuto origine da un'integrazione personalizzata.
> 
> Logger: custom_components.solarman.solarman
> Source: custom_components/solarman/solarman.py:91
> Integration: Solarman (documentation, issues)
> First occurred: 11:27:07 (10 occurrences)
> Last logged: 11:29:07
> 
> Querying [1024 - 1067] failed, [1] retry attempts left
> Querying [1024 - 1067] failed, [0] retry attempts left
> ```
> 
> **Третье сообщение**
> 
> ```
> Questo errore ha avuto origine da un'integrazione personalizzata.
> 
> Logger: custom_components.solarman.solarman
> Source: custom_components/solarman/solarman.py:96
> Integration: Solarman (documentation, issues)
> First occurred: 11:27:07 (5 occurrences)
> Last logged: 11:29:07
> 
> Querying registers [1024 - 1067] failed, aborting.
> ```

Vityok01030103, 2024-03-14:
Привет. Имеется Sofar 8.8KTLX-G3. Подскажите, пожалуйста, возможно ли сделать работу инвертора от 190 воль? А то при 195 вольтах он уходит в защиту. Может кто знает решение? Спасибо.

Vityok01030103, 2024-03-14:
Прошивка на нем такая. LSW3_15_270A_1.68

mikla90, 2025-03-18:
Hey all, I just setup Solarman on home assistant and have the same issue... But with no backup of the yaml file.

How could I solve this?