StephanJoubert/home_assistant_solarman #444: DEYE SUN-29.9/30/35/40/50K-SG01HP3-EU

JanKraslice, 2023-11-08:
Hello, the DEYE HV system does not correctly display values such as PV or battery voltage 
It is possible that an integration from you will work, it uses the classic solarman smart or business portal and here the values are perfectly fine.
The inverter registers should be identical as for 8,10,12K 

Thank you

JanKraslice, 2023-12-14:
I am sending the latest version that is available directly from DEYE support. This is the latest version and I have indeed confirmed that the values for the low voltage inverter and also for the high voltage version are in the sent document.
In the document they differentiate between LV low and high HV

The one sent is only the updated version of this year.
Is this correct?
Thank you very much[
[MODBUS RTU_3PLV_HV_Protocol_V104.pdf](https://github.com/StephanJoubert/home_assistant_solarman/files/13670768/MODBUS.RTU_3PLV_HV_Protocol_V104.pdf)
](url)

hacek, 2024-04-15:
Hey, Anyone got working this model  SUN-10K-SG01HP3-EU-am2 ???

hacek, 2024-05-03:
I've manage to change few lines and have working yaml file with this inverter
[SUN-10K-SG01HP3-EU-am2.zip](https://github.com/StephanJoubert/home_assistant_solarman/files/15198330/SUN-10K-SG01HP3-EU-am2.zip)

andre2800602, 2026-04-08:
Hey, Anyone got working this model SUN-60K-SG01HP3-EU-EM6 ???