StephanJoubert/home_assistant_solarman #453: When using Deye SUN Micro Inverters inc Relais pls use 1.4.3 instread of newest 1.5.1!!!

oriziena, 2023-11-28:
Hi all,

I see some issues coming up and it looks like all of them are not answered. So lets try to have one issue to explain whats needs to be done!

In case you have the relais integrated with your deye the microinverter will be upgrade to 2.32. With that usage of 1.5.1 is broken! 
Easiest way to fix that for now is to downgrade to 1.4.3 and "copy" the deye inverter files manually into the right location of your solarman installation. (how this is done is out of this description)

In my case here the steps to do for an raspberry pi4 installation running on ssd with latest homeassistent version after downgrade to 1.4.3:

so you need to connect to your homeassistant base linux system via ssh and then copy the files from current "master" branch to your local installation. following as example for the deye_2mppt.yaml file
```bash
sudo wget https://raw.githubusercontent.com/StephanJoubert/home_assistant_solarman/main/custom_components/solarman/inverter_definitions/deye_2mppt.yaml -O /usr/share/hassio/homeassistant/custom_components/solarman/inverter_definitions/deye_2mppt.yaml
```

next step is you need to add that file const.py file like shown in following output. pls manipulate the outputs to your needs!
```bash
sudo cat /usr/share/hassio/homeassistant/custom_components/solarman/const.py
from datetime import timedelta

DOMAIN = 'solarman'

DEFAULT_PORT_INVERTER = 8899
DEFAULT_INVERTER_MB_SLAVEID = 1
DEFAULT_LOOKUP_FILE = 'deye_2mppt.yaml'
LOOKUP_FILES = [
    'deye_2mppt.yaml',
    'deye_4mppt.yaml'
]

MIN_TIME_BETWEEN_UPDATES = timedelta(seconds=30)

CONF_INVERTER_HOST = 'inverter_host'
CONF_INVERTER_PORT = 'inverter_port'
CONF_INVERTER_SERIAL = 'inverter_serial'
CONF_INVERTER_MB_SLAVEID = 'inverter_mb_slaveid'
CONF_LOOKUP_FILE = 'lookup_file'

SENSOR_PREFIX = 'Solarman'
```

in case of questions pls add comments here and I will upgrade the issue text.

br
Oriziena

oriziena, 2023-11-28:
@StephanJoubert how about creation of a 1.4.4 with the newest files to make live easier for some people out there until the issue with deye+relais is fixed in 1.5.x?

schlaufon, 2023-12-16:
pls see my new issue!

MajorTomD, 2024-03-04:
Hi!

I tried to change the yaml flies using the File editor. Unfortunately I haven't had succes using Terminal + SSH, maybe due to lack of experience.

Unfortunately my Deye Sun M80G4 EU Q4 still are not implemanted in HA. I tried deye_hybeid.yaml and deye_string.yaml.

Does anybody had success with the same converter?

schlaufon, 2024-03-04:
> Hi!
>
> I tried to change the yaml flies using the File editor. Unfortunately I haven't had succes using Terminal + SSH, maybe due to lack of experience.
>
> Unfortunately my Deye Sun M80G4 EU Q4 still [...]

https://github.com/StephanJoubert/home_assistant_solarman/issues/467#issuecomment-1869455502