StephanJoubert/home_assistant_solarman #459: Status stays disconnected in HA

cseb17, 2023-11-30:
Hello,

I just installed this extension for HA to manage my 2 solar panels but it seems disconnected. I followed the instruction:
[image]

I entered the correct ip and serial number but the status stays "disconnected":
[image]

my config:
[image]

Thanks

Jay66Bee, 2023-12-04:
I am getting the same. Since upgrading to the latest integration (three weeks ago), there have been two occasions where Solarman has disconnected and not reconnected until I became aware of it and restarted HA.

Most annoying.

I have no idea why the new integration does this. I had no such problems with the previous one.

jmccrohan, 2023-12-07:
@cseb17 @Jay66Bee Can you enable debug logs and share when the issue reoccurs please?

Jay66Bee, 2023-12-07:
> @cseb17 @Jay66Bee Can you enable debug logs and share when the issue reoccurs please?

More than happy to, if you can give me details as to what I'm looking for and how to find it.

Thanks.

Jay66Bee, 2023-12-09:
> @cseb17 @Jay66Bee Can you enable debug logs and share when the issue reoccurs please?

This was coming up in the log (multiple times) before Solarman failed again last night...

2023-12-08 20:51:11.738 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [3000 - 3124] failed with exception [OSError: [Errno 24] No file descriptors available]

I have attempted to roll back to 1.4.3 but it is still showing 1.5.1 in HAS

eyk107, 2023-12-09:
I also have the same issue, the IP and the serial number of the wifi logger are checked as well.
My logs are attached
[home-assistant_2023-12-09T11-36-24.302Z.log](https://github.com/StephanJoubert/home_assistant_solarman/files/13623940/home-assistant_2023-12-09T11-36-24.302Z.log)

cseb17, 2023-12-12:
> @cseb17 @Jay66Bee Can you enable debug logs and share when the issue reoccurs please?

Here it is
[home-assistant_2023-12-12T12-33-55.359Z.log](https://github.com/StephanJoubert/home_assistant_solarman/files/13648124/home-assistant_2023-12-12T12-33-55.359Z.log)

schlaufon, 2023-12-17:
join the party with same above problems.
 With 200 issues ahead an queing, I doubt that anything will happen -say- within the next 3 months, if ever.
 Looks like the integration gets no serious support anymore.
 I would help, if I could....

vinceg0267, 2023-12-29:
Same problem

eyk107, 2023-12-29:
Rolling back to version 1.4.3 fixed the issue for me

cseb17, 2024-01-04:
> > Same problem
> 
> @vinceg0267 If you have a relay installed. see "my" solved issue #467 . May be of help.

Dear all,

following the steps described by @vinceg0267, I removed the V1.5.1, restart HA and install the V1.4.3 and...it works :)
Thanks for the sharing.
We will see with the the next release ;)

closed 2024-01-04 by cseb17

kdober, 2024-02-05:
Hi, 
I have a Deye 5k-SG03LP1-EU and I had the same issue with verion 1.5.1. After a few hours all sensors stop getting data. I'm now using version 1.4.3 as suggested and its working perfect for the last 5 days. 

Cheers