StephanJoubert/home_assistant_solarman #460: Does not always recover from network disconnect

NemesisXB, 2023-12-01:
I have 2 systems running, one runs perfectly without issue. The second keeps going into a state where no data is received from the inverter. I think it happens when the network connection breaks momentarily (the wifi in this are is quite bad). All my inverter related UI elements then just shows NaN. Reloading the integration fails, 
[image]
The only fix is to restart home assistant.

It did not happen in version 1.4.3, but is now happening in version 1.5.1. 

While in that state, I can still ping the dongle, so the network connection was restored, but the integration does not seem to restore.

Jay66Bee, 2023-12-04:
Same here. No problems with 1.4.3 but since installing 1.5.1 three weeks ago I had one occasion each week where Solarman disconnects and stays in state until HA is restarted.

rimelis, 2023-12-04:
We have already discussed some time ago that MODBUS implementation starting from 1.5.x  has been passed to external module, which seems not very stable and mature

https://github.com/StephanJoubert/home_assistant_solarman/issues/394#issuecomment-1715877407

WimDeceuninck, 2024-04-27:
> I have 2 systems running, one runs perfectly without issue. The second keeps going into a state where no data is received from the inverter. I think it happens when the network connection breaks [...]

Hi all, I also tried to add the Solarman integration version 1.4.3. to connect with my Solis dataloger (solis_hybrid.yaml). but the the status remains 'disconnected'. Although I can ping the datalogger. Did anyone find a solution for this?