StephanJoubert/home_assistant_solarman #463: I can't connect my sofar inverter

Picard2002, 2023-12-06:
I have recently installed a self-consumption otovoltaic plant in my home, it is made up of 10 550w solar panels, a Sofar 5ktlm G3 inverter with a USB wifi stick and a wifi meter, I have tried the integration from hacs and manually, in none of them. In both cases I can get the connection, I would love to be able to integrate my photovoltaic system into Home Assistant, but so far I have only failed, I have read all the posts about the problem and I see that some of you have done it through mqtt, this escapes my knowledge, since it is very basic, can you help me do the integration through hacs?
thank you

Picard2002, 2023-12-07:
I am puzzled, in my system there are two loggers plus a meter, when trying to do the integration, if it is at night one of the loggers is disconnected, so it is impossible for me, I have tried it during the day but the result is the same as with the other recorder that is always connected, status "unknown", I no longer know if I am doing it wrong, right or it will simply be impossible for me to perform the integration, I put images in case someone sees something that I missed

[image]

[image]

[image]

[image]

kuznets, 2024-01-23:
I have same issue with Sofar 15ktlm G3. Integration is added by not connect with logger.
[image]

Picard2002, 2024-01-23:
I got something by copying the Azzurro inverter code and pasting it into a yaml file that I called custom_parameters.yaml, it could work for you, it works on the sofa 12 ktlm g3, maybe on the 15 ktlm too

h4ckm4t, 2024-04-16:
same issue here: HYD 15KTL-3PH
all sensor status = Unknown as above

kuznets, 2024-04-17:
I use **zcs_azzurro-ktl-v3.yaml** and now it's work.

h4ckm4t, 2024-04-17:
Unfortunately, none of the available inverter YAML configurations work with my hybrid three-phase inverter, a Sofar Solar HYD 15KTL-3PH. It should run with the sofar_g3hyd.yaml, but as mentioned, none of the configurations for Sofar or Azzurro work for me. Any help or guidance would be appreciated! :)