StephanJoubert/home_assistant_solarman #467: HACS integration wont work with my microinverter  SUN-M80G3-EU plus SUN-MI-RELAY-01

schlaufon, 2023-12-16:
[image]

See above my data when connected via browser to the inverter/logger.
I can see all data on my iPhone with the solarman app and also with the windows browser on my laptop.
The thing is working.

HOWEVER I do not get ANY information in HA. All entities / sensors are labeled "unbekannt" (=unknown) or in case of "status"  ist "disconnected".

I have tried all yamls (deye_hybrid, deye_2mppt, dey_string). None is working. 1 week now , no avail of any data.

**Is it possible that this mandotary relay that is connected between the inverter and the mains, causes trouble for the integration?**

Please advise!

schlaufon, 2023-12-16:
[image]
This is result when installing 1.4.3 . Why renaming the entities?

Murlwauf, 2023-12-18:
I have the same problems with this.

Nappyy, 2023-12-18:
Same Problem here.

jfreak80, 2023-12-21:
Same here. But glad, it seems to be not my fault :-)

Pompom009, 2023-12-22:
I have the same problems with this.

stephan101, 2023-12-25:
Same probleme. If a check the console i see the error:
...
2023-12-25 12:32:28.802 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid start or end values]
2023-12-25 12:32:29.224 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [1 - 125] failed, [1] retry attempts left
2023-12-25 12:32:29.349 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid start or end values]
.....

foundnobetter, 2023-12-25:
Same here.

Deye SUN600G3-EU-230 working with Relay firmware MW3_16U_5406_2.32-D1
Serial starts with 41. So maybe i have to switch to "component for V4 of the protocol" as stated in documentation.
I have no idea how to do the switch. Any advise?
Thanks

`KeyError: 'solarman'
2023-12-25 12:49:24.341 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid start or end values]
2023-12-25 12:49:24.654 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125] failed, [1] retry attempts left
2023-12-25 12:49:24.768 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid start or end values]
2023-12-25 12:49:25.189 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying [1 - 125] failed, [0] retry attempts left
2023-12-25 12:49:25.189 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying registers [1 - 125] failed, aborting.
2023-12-25 12:49:45.568 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry 192.168.201.30 for solarman
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 557, in async_unload
result = await component.async_unload_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solarman/__init__.py", line 28, in async_unload_entry
hass.data[DOMAIN].pop(entry.entry_id)
~~~~~~~~~^^^^^^^^
KeyError: 'solarman'`

foundnobetter, 2023-12-25:
Update on this issue: It works!
As stated here https://github.com/StephanJoubert/home_assistant_solarman/issues/407

you have to
- remove solarman integration 1.51 from HACS
- install solarman integration 1.43 from HACS
- If you need the deye_2mppt.yaml, as me
-- copy deye_2mppt.yaml to config/cutom_components/solarman/inverter_definitions
-- add 'deye_2mppt.yaml', to section Lookup_files in config/cutom_components/solarman/const.py

And dont forget to restart HA
And do not update to Version 1.51!

schlaufon, 2023-12-25:
> Update on this issue: It works! As stated here #407
>
> you have to
>
> * remove solarman integration 1.51 from HACS
> * install solarman integration 1.43 from HACS
> * If you need the deye_2mppt.yaml, as [...]

Could you be so kind an d show  a pic of your _const.py_ ?!

Because I do not understand your work around (if it is). The _Lookup_Files_ are well defined in the _const.py_ .
Why copying it again into _const.py_ ?

Here my screenshot:

[image]

foundnobetter, 2023-12-25:
Deye SUN600G3-EU-230 working with Relay firmware MW3_16U_5406_2.32-D1.

Your const.py ist correct.
There is no 2mppt after you downgrade from 1.51 to 1.43.
Thats the reason i added it manually.

`from datetime import timedelta

DOMAIN = 'solarman'

DEFAULT_PORT_INVERTER = 8899
DEFAULT_INVERTER_MB_SLAVEID = 1
DEFAULT_LOOKUP_FILE = 'deye_hybrid.yaml'
LOOKUP_FILES = [
    'deye_2mppt.yaml',
    'deye_4mppt.yaml',
    'deye_hybrid.yaml',
    'deye_sg04lp3.yaml',
    'deye_string.yaml',
    'sofar_lsw3.yaml',
    'sofar_wifikit.yaml',
    'solis_hybrid.yaml',
    'solis_1p8k-5g.yaml',
    'sofar_g3hyd.yaml',
    'sofar_hyd3k-6k-es.yaml',
    'zcs_azzurro-ktl-v3.yaml',
    'custom_parameters.yaml'
]

MIN_TIME_BETWEEN_UPDATES = timedelta(seconds=30)

CONF_INVERTER_HOST = 'inverter_host'
CONF_INVERTER_PORT = 'inverter_port'
CONF_INVERTER_SERIAL = 'inverter_serial'
CONF_INVERTER_MB_SLAVEID = 'inverter_mb_slaveid'
CONF_LOOKUP_FILE = 'lookup_file'

SENSOR_PREFIX = 'Solarman'
`

schlaufon, 2023-12-26:
> Deye SUN600G3-EU-230 working with Relay firmware MW3_16U_5406_2.32-D1.
>
> Your const.py ist correct. There is no 2mppt after you downgrade from 1.51 to 1.43. Thats the reason i added it manually. [...]

Wow!
I confirm your solution works with me. ThanX alot.

[image]

I'd  like to add that the installation is tricky, if you are not familiar with a downgrade process. Such as I am. For me that is just to much fiddling around and NOT a proper solution with an integration (even if not official and a HACS repository.
I' d like to thank @StephanJoubert . I hope has spends a splendid Xmas season and will return in good shape for a new year and some additional fixing. Looks like he's out of the project since a month or two. Hope he will come back. Fingers crossed.

For starters let me repeat the steps to downgrade.
1. Go to the installed repositories in HACS(probably Solarman 1.5.1) and DELETE that completely. 
2. Check with your file editor, that all files in "custom components" under solarman are gone.
3. Search in HACS now for the solarman repository. DOWNLOAD Version 1.5.1 (!!!!). 
4. Install the Solarman 
5. Reboot HA
6. Go to the solarman files -as @foundnobetter describes above- and DOWNLOAD the 'deye_2mppt.yaml' (it will be stored on your PC where any other downloaded files go, normally under Windows into the file drawer "downloads"
7. Repeat now steps 1. and 2.
8. Search in HACS now for the solarman repository. DOWNLOAD Version 1.4.3 (!!!!). 
9. DO NOT Install the Solarman yet from HACS!
10. Reboot HA
11. Go NOW to the solarman files -as @foundnobetter describes above- and UPLOAD the 'deye_2mppt.yaml' from your PC drive INTO the folder with all the other solarman components
12. PLUS now Change the _const.py_ by adding one line in 
LOOKUP_FILES = [
'deye_2mppt.yaml',
.......]
13. Make sure there ist a commata "," behind your added line
14. ONLY NOW install Solarman from HACS
15. Choose "deye_2ppt.yaml" in the pop-up, together with you device ID (=logger, NOT inverter), and the IP.
16. Almost done.
17. Reboot HA 

Have fun!

foundnobetter, 2023-12-26:
One more offtopic hint:
Within HA, for me Studio Code Server is way better than File editor.

stephan101, 2023-12-28:
Hey, 

the solution above works for my deye SUN-M80G3-EU plus SUN-MI-RELAY-01 Relay firmware MW3_16U_5406_2.32-D1.

For the ID i use the logger ID (39xxxxxxxx) not the microinverter ID (23xxxxxxxx).
I see the PV1, PV2 current/voltage, Total AC Output Power (Active) and so on.

Only same parameters are not filled: Communication Prototocal Version, Hardware Version, Update Time,....

I don't understand realy whats the difference between 1.51 an 1.43.

One hint: If I downgraded in hacs the solarman version I had to wait > 1 minute befor i pressed the Button Download. 
If I was too fast, i always downloaded the version 1.51. But may be the reason for this is my old hardware (Raspberry Pi 2 Model B with debian bookworm).

Thanks a lot 
Stephan

1bieber, 2023-12-30:
I've the same issue with my SUN-M80G3-EU with firmware MW3_16U_5406_2.32-D1.
Since a few days all my solarman sensors switched to "unknown" state. Honestly I was not able to trace it back to an update or something else.
I appriciate the solution mentioned before here but I'm very lazy. Therefore the question: Is there any chance to just wait for an update to fix the issue?

schlaufon, 2023-12-30:
> I've the same issue with my SUN-M80G3-EU with firmware MW3_16U_5406_2.32-D1. Since a few days all my solarman sensors switched to "unknown" state. Honestly I was not able to trace it back to an [...]

I doubt that @StephanJoubert is listening these days. The amount of "unfinished" issues has formed quite a queue.
I do no see any other developer involved in this rerpository. Sad enough. So, this might take a long while before fixed...

I am lazy too, plus I do not have any mood to play with .yaml files. However I took the steps (1.to17.)  described above and it worked.
Just give it a try! Good luck!

stheid, 2024-01-10:
Is the issue with this inverters caused by the changes in the inverter definitions (yaml) or in the 1.5.x changes inside the code?

1bieber, 2024-01-10:
> I doubt that @StephanJoubert is listening these days. The amount of "unfinished" issues has formed quite a queue. I do no see any other developer involved in this rerpository. Sad enough. So, this [...]

Thanks!

Small feedback from my side:
I removed the solarman integration completely and bought a TP-Link Tapo P110 smart plug  (10-15€) which works really great. There is a really well working HACS integration which deploys daily and monthly generated energy sensors out of the box. I really enjoy this smart plug combinated with the HACS integration in HA, especially for energy components.

stheid, 2024-01-11:
After trying it out myself, the whole issue boils down to:

- install version 1.4.3
- copy the [inverter definition of this master version](https://github.com/StephanJoubert/home_assistant_solarman/blob/2579128d7854b362ebe0ea98cd48a79760369e63/custom_components/solarman/inverter_definitions/deye_2mppt.yaml)
- paste it into a [custom-component](https://github.com/StephanJoubert/home_assistant_solarman/blob/main/customization.md)

derklausi, 2024-03-18:
Thanks for this Link, i think it will work for me too with 1.43 ... hoping for sun tomorrow to see :-)

Kolbi, 2024-03-21:
So in conclusion the inverter definition is alright? Need to see what changed from 1.4.3 to 1.5.1.

> After trying it out myself, the whole issue boils down to:
>
> * install version 1.4.3
> * copy the [inverter definition of this master [...]

Kolbi, 2024-03-21:
I guess it's related to switching to https://github.com/jmccrohan/pysolarmanv5

schlaufon, 2024-03-21:
@Kolbi  ????

bhuebschen, 2024-04-20:
https://github.com/StephanJoubert/home_assistant_solarman/issues/407#issuecomment-2067604201

jmccrohan, 2024-04-28:
I've released [pysolarmanv5 v3.0.2](https://github.com/jmccrohan/pysolarmanv5/releases/tag/v3.0.2) which (I hope) will resolve the issues with Deye Microinverters.

Kolbi, 2024-04-29:
Can someone please try to uninstall Solarman via HACS, restart HASSIO, install Solarman again and restart again? I guess then latest pysolarmanv5 should be used.

derklausi, 2024-04-29:
Hi, can someone, who is using Deye Microinverters test the new Version V5 to see if it now works again?
I downgraded to v4, so its more work for me to test and re downgrade if its not working. If someone has not downgrade it would be easy to test and give feedback.

TheYoshy, 2024-04-29:
For me it works now (Deye SUN-80G3-EU + relaybox). Updatet today directly from 1.4.3 to 1.5.1. 
THX

jmccrohan, 2024-04-29:
> Hi, ok, i also did it, updated from 1.4.3 to 1.5.1 directly. Deye Sun M80G3 (MW3_16U_5406_2.32-D1) and two Sun 600-G3 (MW3_16U_5406_2.32-D1) are working. All with Relay Boxes. I'm just a litte bit confused, because 1.5.1 was the Version before. Is it replaced under the same Version Number? Thank you very much!!!

The Solarman integration doesn't include a versioned depends on pysolarmanv5, so by reinstalling you are getting the same version of Solarman v1.5.1, but the latest version of pysolarmanv5 (v3.0.2).

flxhaas, 2024-04-29:
I'm seeing: 

Logger: custom_components.solarman.solarman
Source: custom_components/solarman/solarman.py:88
Integration: Solarman (documentation, issues)

**Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid start or end values]**

After complete un-/reinstall

:-(

jmccrohan, 2024-04-30:
> I'm seeing:
> 
> Logger: custom_components.solarman.solarman Source: custom_components/solarman/solarman.py:88 Integration: Solarman (documentation, issues)
> 
> **Querying [1 - 125] failed with exception [V5FrameError: V5 frame contains invalid start or end values]**
> 
> After complete un-/reinstall
> 
> :-(

Please enable debug logging and attach same.

Add the following to `configuration.yml`:
```
logger:
  logs:
     custom_components.solarman: debug
```

jasonkaisersmith, 2024-08-04:
I have the same issue here, but I have a DEYE SUN-M80G4-Eu-Q0

schlaufon, 2024-08-04:
with HACS integration wont work with my microinverter SUN-M80**G3**-EU plus SUN-MI-RELAY-01 it still works.
Maybe open another thread or consult DEYE about the difference. I am sorry I cannot assist in this matter.
Good luck!
@jasonkaisersmith

Paddel0815, 2024-08-22:
I also have the 'Deye SUN-M80G3-EU-Q0' and unfortunately, I can't read anything. Which SN should be entered now? And which version should be running?
[image]