StephanJoubert/home_assistant_solarman #469: CRCError: CRC validation failed

onetoomany, 2023-12-18:
My logs are full of the following error message:
```
2023-12-17 17:35:13.681 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [512 - 597] failed with exception [CRCError: CRC validation failed.]
2023-12-17 17:35:14.182 WARNING (SyncWorker_1) [custom_components.solarman.solarman] Querying [512 - 597] failed, [1] retry attempts left
2023-12-17 17:46:14.017 WARNING (SyncWorker_0) [custom_components.solarman.solarman] Querying [4272 - 4284] failed with exception [CRCError: CRC validation failed.]
2023-12-17 17:46:14.517 WARNING (SyncWorker_0) [custom_components.solarman.solarman] Querying [4272 - 4284] failed, [1] retry attempts left
```

My inverter type is Sofar HYD 5000 ES (SM1ES150). Firmware G360.

The data seems to be read ok, with the exception of 0x022B - "Inverter alert message"; this appears to be classed incorrectly as an alert for my system.

jayzees, 2024-02-07:
same here with the same setup

darkman82, 2024-03-31:
Same with HYD 6000 (ZM1ES060) with Firmware G363.  HA on raspberry Pi 3+.

Usually 30% of readings fail with that error, the read is ok after 1-2 attempts.

CRC errors come from uModbus library. 

Already tried two different stick loggers 17**** and 27**** . Same results.

sjmtlewy, 2024-05-16:
I'm getting this as well. I've worked round it by upping the retry limit to 4

darkman82, 2024-05-16:
@sjmtlewy I'd suggest you put a sleep between retries. It helped. 1 second is enough in my case.

Have a look at #523 too. I left a piece of code you could try.