StephanJoubert/home_assistant_solarman #472: Modify lsw3_yaml

Picard2002, 2023-12-25:
Hello, due to the impossibility of being able to obtain data from my sofar 5KTLM G3 inverter, I have been applying modifications from other users to the sofar_lsw3.yaml file, I have obtained different results, but that gives me hope that an expert can help us.
By modifying this file with this code [https://gist.github.com/gushcs/223ca2c61af2d345238d286746617650](url)I get some data, but I have a ddzy422-d2 meter that should provide me with additional information, such as the amount of energy poured into the grid, but this modification,[ https://gist.github.com/gushcs/223ca2c61af2d345238d286746617650](url)It is not made for the modbus connection with the meter, I have uncommented some lines of the code to try if I could obtain the data from the meter, I would appreciate the collaboration of some expert to be able to have the integration fully functional, thank you

ajgball, 2023-12-31:
Hi I have the 6KTLM G2 inverter and am able to get all data except the grid output using the sofar_lsw3.yaml file. I had to add a CT clamp to my inverter to measure the output power going to grid. I will niw attempt to add this to the yaml file

Picard2002, 2023-12-31:
I have managed to obtain this data in real time, using zcs_azzurro-ktl.v3.yaml as custom_parameters.yaml, but I do not have instantaneous production data, only total production of the day and total accumulated production, I do not obtain consumption or discharge data to the network , I believed that having a WiFi wattmeter (solarman ddzy 422-d2) the integration would also provide them to me, it would be of great help if someone who understood the system would publish how to obtain that consumption and discharge data.
thank you

[image]

mrlordziemniak, 2024-01-14:
i have meter and tried to modify  custom_parameters for this meter https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_igen_dtsd422.md

`requests:
  - start: 0x0000
    end:  0x0027
    mb_functioncode: 0x03

parameters:
 - group: solar
   items: 
    - name: "Total Positive Energy CT1"
      class: "energy"
      state_class: "total_increasing"      
      uom: "kWh"
      scale: 1
      rule: 3
      registers: [0x003F]
      icon: 'mdi:solar-power'      

 - group: Output
   items: 
    - name: "CT1 Voltage"
      class: "voltage"
      state_class: "measurement"      
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x0001]
      icon: 'mdi:home-lightning-bolt'

    - name: "CT1 Current"
      class: "current"
      state_class: "measurement"      
      uom: "A"
      scale: 0.01
      rule: 1
      registers: [0x0007,0x0008]
      icon: 'mdi:home-lightning-bolt'

    - name: "CT2 Voltage"
      class: "voltage"
      state_class: "measurement"      
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x0002]
      icon: 'mdi:home-lightning-bolt'

    - name: "CT2 Current"
      class: "current"
      state_class: "measurement"      
      uom: "A"
      scale: 0.01
      rule: 1
      registers: [0x0009,0x000A]
      icon: 'mdi:home-lightning-bolt'

    - name: "CT3 Voltage"
      class: "voltage"
      state_class: "measurement"      
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x0003]
      icon: 'mdi:home-lightning-bolt'

    - name: "CT3 Current"
      class: "current"
      state_class: "measurement"      
      uom: "A"
      scale: 0.01
      rule: 1
      registers: [0x000B,0x000C]
      icon: 'mdi:home-lightning-bolt'

 

`

and for now it is working,  but i have no data from total positive energy, im still testing and trying :)

Noyax-37, 2024-07-13:
Hi, do you have other data now?

Thanks for help

mrlordziemniak, 2024-07-14:
Yes I have, it is working fine, code is on my git hub pages

Noyax-37, 2024-07-14:
Thanks a lot 👍