StephanJoubert/home_assistant_solarman #479: Solarman Multi Deye 5kwa gives installation error. It does not connect.

basersoft, 2024-01-02:
09:44:55 ERROR   Error retry attempt(4): unpack requires a buffer of 2 bytes
09:44:57 ERROR   OSError reading 1 registers from 219: Failed to read register 219 Battery Low Capacity
09:44:57 ERROR   Error retry attempt(0): unpack requires a buffer of 2 bytes
09:44:59 ERROR   Error retry attempt(1): unpack requires a buffer of 2 bytes
09:45:01 ERROR   Error retry attempt(2): unpack requires a buffer of 2 bytes
09:45:03 ERROR   Error retry attempt(3): unpack requires a buffer of 2 bytes
09:45:06 ERROR   Error retry attempt(4): unpack requires a buffer of 2 bytes
09:45:08 ERROR   OSError reading 1 registers from 254: Failed to read register 254 Prog5 Time
09:45:08 CRITICAL Could not read sensors: ['Prog2 Time', 'Prog1 Time', 'Rated power', 'Prog6 Time', 'Serial', 'Prog3 Time', 'Prog4 Time', 'Battery Low Capacity', 'Prog5 Time']
09:45:08 INFO    ############################################################
09:45:08 INFO    No response on the Modbus interface tcp://192.168.1.243:8899, see https://kellerza.github.io/sunsynk/guide/fault-finding
09:45:08 INFO    ############################################################
09:45:08 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.
09:45:38 ERROR   Task was destroyed but it is pending!
task: <Task pending name='ConnKeeper' coro=<PySolarmanV5Async._conn_keeper() running at /usr/local/lib/python3.11/site-packages/pysolarmanv5/pysolarmanv5_async.py:139> wait_for=<Future pending cb=[Task.task_wakeup()]>>
[09:45:38] INFO: Service restart after closing
09:45:39 INFO    Loading configuration: /data/options.json
09:45:39 INFO    sunsynk library version: 0.6.4
09:45:39 INFO    Schedules:
+-----------+-----+------+--------+-----------+----------+------------+
|    Key    | src | Read | Report | Change by | Change % | Change any |
+-----------+-----+------+--------+-----------+----------+------------+
| date_time |     |  60  |   60   |           |          |    True    |
|     rw    |     |  5   |  300   |           |          |    True    |
|     w     |  *  |  15  |   60   |     80    |          |            |
|    kwh    |     | 300  |  300   |           |          |            |
|  any_unit |     |  15  |  300   |           |          |            |
|  no_unit  |     |  15  |  300   |           |          |    True    |
+-----------+-----+------+--------+-----------+----------+------------+
09:45:39 INFO    Using Single phase sensor definitions.
09:45:39 INFO    Added hidden sensors as other sensors depend on it: Rated power, Serial
09:45:39 INFO    Connecting to tcp://192.168.1.243:8899
09:45:39 INFO    Reading startup sensors Prog4 Time, Prog5 Time, Battery Low Capacity, Prog2 Time, Prog3 Time, Rated power, Prog1 Time, Prog6 Time, Serial
09:45:39 ERROR   Error retry attempt(0): unpack requires a buffer of 30 bytes
09:45:42 ERROR   Error retry attempt(1): unpack requires a buffer of 30 bytes
09:45:44 ERROR   Error retry attempt(2): unpack requires a buffer of 30 bytes
09:45:46 ERROR   Error retry attempt(3): unpack requires a buffer of 30 bytes
09:45:48 ERROR   Error retry attempt(4): unpack requires a buffer of 30 bytes
09:45:50 ERROR   Error retry attempt(0): unpack requires a buffer of 2 bytes
09:45:54 ERROR   Error retry attempt(1): unpack requires a buffer of 2 bytes
09:45:56 ERROR   Error retry attempt(2): unpack requires a buffer of 2 bytes
09:45:58 ERROR   Error retry attempt(3): unpack requires a buffer of 2 bytes
09:46:00 ERROR   Error retry attempt(4): unpack requires a buffer of 2 bytes
09:46:01 ERROR   Error retry attempt(0): unpack requires a buffer of 12 bytes
09:46:03 ERROR   Error retry attempt(1): unpack requires a buffer of 12 bytes
09:46:06 ERROR   Error retry attempt(2): unpack requires a buffer of 12 bytes
09:46:08 ERROR   Error retry attempt(3): unpack requires a buffer of 12 bytes
09:46:10 ERROR   Error retry attempt(4): unpack requires a buffer of 12 bytes
09:46:12 ERROR   OSError reading 15 registers from 3: Failed to read register 3; timeout reading 1 registers from 219; OSError reading 6 registers from 250: Failed to read register 250
09:46:12 ERROR   Error retry attempt(0): unpack requires a buffer of 30 bytes
09:46:14 ERROR   Error retry attempt(1): unpack requires a buffer of 30 bytes
09:46:17 ERROR   Error retry attempt(2): unpack requires a buffer of 30 bytes
09:46:19 ERROR   Error retry attempt(3): unpack requires a buffer of 30 bytes
09:46:21 ERROR   Error retry attempt(4): unpack requires a buffer of 30 bytes
09:46:23 ERROR   Error retry attempt(0): unpack requires a buffer of 2 bytes
09:46:26 ERROR   Error retry attempt(1): unpack requires a buffer of 2 bytes
09:46:28 ERROR   Error retry attempt(2): unpack requires a buffer of 2 bytes
09:46:30 ERROR   Error retry attempt(3): unpack requires a buffer of 2 bytes
09:46:34 ERROR   Error retry attempt(4): unpack requires a buffer of 2 bytes
09:46:34 ERROR   Error retry attempt(0): unpack requires a buffer of 12 bytes
09:46:36 ERROR   Error retry attempt(1): unpack requires a buffer of 12 bytes
09:46:43 ERROR   Error retry attempt(2): unpack requires a buffer of 12 bytes
09:46:45 ERROR   OSError reading 15 registers from 3: Failed to read register 3; timeout reading 1 registers from 219; timeout reading 6 registers from 250
09:47:05 ERROR   Error retry attempt(0): unpack requires a buffer of 2 bytes
09:47:07 ERROR   Error retry attempt(0): unpack requires a buffer of 12 bytes
09:47:09 ERROR   Error retry attempt(1): unpack requires a buffer of 12 bytes
09:47:11 ERROR   Error retry attempt(2): unpack requires a buffer of 12 bytes
09:47:14 ERROR   Error retry attempt(3): unpack requires a buffer of 12 bytes
09:47:16 ERROR   Error retry attempt(4): unpack requires a buffer of 12 bytes
09:47:18 ERROR   timeout reading 15 registers from 3; timeout reading 1 registers from 219; OSError reading 6 registers from 250: Failed to read register 250
09:47:18 WARNING Retrying individual sensors: ['prog4_time', 'prog5_time', 'battery_low_capacity', 'prog2_time', 'prog3_time', 'rated_power', 'prog1_time', 'prog6_time', 'serial']
09:47:18 ERROR   Error retry attempt(0): unpack requires a buffer of 2 bytes
09:47:20 ERROR   Error retry attempt(1): unpack requires a buffer of 2 bytes
09:47:22 ERROR   Error retry attempt(2): unpack requires a buffer of 2 bytes
09:47:24 ERROR   Error retry attempt(3): unpack requires a buffer of 2 bytes
09:47:26 ERROR   Error retry attempt(4): unpack requires a buffer of 2 bytes
09:47:28 ERROR   OSError reading 1 registers from 253: Failed to read register 253 Prog4 Time
09:47:29 ERROR   Error retry attempt(0): unpack requires a buffer of 2 bytes
09:47:31 ERROR   Error retry attempt(1): unpack requires a buffer of 2 bytes
09:47:33 ERROR   Error retry attempt(2): unpack requires a buffer of 2 bytes
09:47:35 ERROR   Error retry attempt(3): unpack requires a buffer of 2 bytes
09:47:37 ERROR   Error retry attempt(4): unpack requires a buffer of 2 bytes
09:47:39 ERROR   OSError reading 1 registers from 254: Failed to read register 254 Prog5 Time
09:47:39 ERROR   Error retry attempt(0): unpack requires a buffer of 2 bytes
09:47:42 ERROR   Error retry attempt(1): unpack requires a buffer of 2 bytes
09:47:44 ERROR   Error retry attempt(2): unpack requires a buffer of 2 bytes
09:47:46 ERROR   Error retry attempt(3): unpack requires a buffer of 2 bytes
09:47:48 ERROR   Error retry attempt(4): unpack requires a buffer of 2 bytes
09:47:50 ERROR   OSError reading 1 registers from 219: Failed to read register 219 Battery Low Capacity
09:47:50 ERROR   Error retry attempt(0): unpack requires a buffer of 2 bytes
09:47:52 ERROR   Error retry attempt(1): unpack requires a buffer of 2 bytes
09:47:54 ERROR   Error retry attempt(2): unpack requires a buffer of 2 bytes
09:47:57 ERROR   Error retry attempt(3): unpack requires a buffer of 2 bytes
09:47:59 ERROR   Error retry attempt(4): unpack requires a buffer of 2 bytes
09:48:01 ERROR   OSError reading 1 registers from 251: Failed to read register 251 Prog2 Time
09:48:01 ERROR   Error retry attempt(0): unpack requires a buffer of 2 bytes
09:48:03 ERROR   Error retry attempt(1): unpack requires a buffer of 2 bytes
09:48:05 ERROR   Error retry attempt(2): unpack requires a buffer of 2 bytes
09:48:11 ERROR   Error retry attempt(3): unpack requires a buffer of 2 bytes
09:48:12 ERROR   timeout reading 1 registers from 252 Prog3 Time

clark249, 2024-02-05:
Hi there have exactly the same problem, really struggling to find anything on it, there is some stuff going back to 2022 regarding changing int to int32 but I can't work that out, where to change it, and it seems like such a while ago that it may not be relevant now

basersoft, 2024-02-05:
> Hi there have exactly the same problem, really struggling to find anything on it, there is some stuff going back to 2022 regarding changing int to int32 but I can't work that out, where to change it, and it seems like such a while ago that it may not be relevant now

I ran it with version 1.4. I didn't do much.
I used deye_string.yaml. It works without any problems.

clark249, 2024-02-06:
Thank you for that information Stephan, it is very helpful. I have tried it but still not working for me. Can I be a pain and ask a couple of things that I am struggling with ?

  1.  What version of Home assistant and Home assistant Core are you using ?
  2.  With the installation via a custom install, I can get the integration installed, but there does not seem to any way to access the data read as there is in the app install (the one that didnt work). Am I missing something, I have the integration, but no APP in Add-Ons, do you have anything like that, if so what did you do after installing the integration ?

Sorry to be a pain, but any help would be very much appreciated

Kind Regards,

Peter