StephanJoubert/home_assistant_solarman #485: Status connection

vinceg0267, 2024-01-10:
I often have such a situation and the sensors become unknown. What could I check?

[image]

schlaufon, 2024-01-10:
Looks to me like a bad Wifi connection between the inverter and your HA server (Raspberry, NUC etc.).
You may check (if you've got the time) and check signal quality of the inverter / device (during daytime) by directly connecting your smart phone (in any case: switch of mobile data usage!!) with device's APxxxxxxxxx (SSID) under it's  IP 10.10.100.254 with login: admin 
 & admin. There see your wifi connections quality.
 I had the same problem even with a signal strength approx 30%. You want to try to establish a better connection then, maybe above 35% should work.
 Good luck!

vinceg0267, 2024-01-11:
Sorry I do not understand. The HA server is connected to the modem via Ethernet cable and the modem is located 1 meter from the device.