StephanJoubert/home_assistant_solarman #486: Use service "Write holding registers" for multiple Deye inverter

TomtomWa, 2024-01-10:
Hi,

tldr: Need assistance to control 2 Deye Sun600 via the service "Write Multiple Holding Registers", at the moment only 1 Deye Sun600 is controllable via the Solarman integration.

I have the following **setup**:
- 2x Deye Sun600
- 1x Pylontech US3000C
- 2x Victron SmartSolar (+PV)
- 2x Shelly 3EM (1 for monitoring Deye AC Power, 1 for monitoring grid power)

I managed to integrate all components into my home assistant to monitor the data (to setup automatizations, monitor efficiency, ...).

My **goal** is:
Control the Deye's in order to not donate PV electricity to the grid. The Deye's shall always feed as much as the house needs at the moment. In german so called "Nulleinspeisung".

My **problem** at the moment:
The service to control the Deye's can only control the one Sun600, that I added as last. That means, I cannot control the first added Inverter. It seams to me, that the service was registered with the same generic name "solarman.write_...".
I tried to find the place in the software, where it registers the service to see, if I could change and give another name, but I was not successful in the end. The service is always named "solarman.write....", independently which name is given to the specific Deye integration.

I thought about **2 options** (but there might be even more?):
1. Register service with specific inverter name (in my case Deye01 and Deye02 instead of generic "solarman"; preferable option)
2. Duplicate Solarman integration with different name in my Home Assistant

Anyone has an idea, how to get two services with different names for both inverters?

titzone, 2024-02-02:
Hi, ich have exact the same issue, I have a deye inverter and nox a new Afore one. After i added the Afore i wondered why the deye does not receive the write_multiple_registers value. 
Did you find a workarround for the Problem?

mgherzan, 2024-04-08:
@TomtomWa , were you able to achieve the **dynamic** "zero export" ("Nulleinspeisung") with just one Deye device? If yes, how?
BTW, what is your base consumption/power ("Grundlast")?

Elijen, 2024-04-08:
Same issue here. It would be enough to control the master inverter (slave would just copy the values) but unfortunately my HA integration seems to change the values on slave instead.

mgherzan, 2024-04-10:
> Indeed last week I finally had the time and managed it to get dynamic zero export. I set the Deye01 to 22%, giving constantly ~140W until the battery is low. For the second Deye02 (the one I added last) I had an automation for the dynamic part. Unfortunately in the next night my Home Assistant crashed, destroying the sd card and the weekly backup was not yet gone through....So I managed to get the last backup running on the NAS (hopefully more stable than the RPi4) and need to do it again.
> 
> Basically, as a first shot, I created some helpers in HA, that calculated a Setpoint for Deye. Calculation e.g.: (CurrentPowerDeye + CurrentPowerHome - ConstOffsetToZero) / MaxPowerDeye = SetPointIn% E.g. (160W + 250W - 100W) / 600W = 0,52 (Multiply this value with 100 to get the actual number 52%, that you can pass to the inverter). The helper also applied max and min to the calculation. Max would be 600W (or 100%) and min 1% (with 0% it ended up to set the inverter to 100% instead of 0%). If you do not measure the power of your inverter directly, you could also just use the current SetPoint for your calculation.

Wow, this is really great! Please, could you post a schematic/drawing of your wiring? You might be setting the gold standard for the German/EU market in terms of ***fanless*** dynamic zero-export, and that with one of the few micro-inverters [capable of AC-side throttling](https://www.youtube.com/watch?v=9W1WTIYNEyo). 

BTW, are you using inrush current limiters on the wires from the battery to the inverter?

Micc0548, 2024-05-13:
> Same issue here. It would be enough to control the master inverter (slave would just copy the values) but unfortunately my HA integration seems to change the values on slave instead.
> 
> edit: I managed to modify `services.py` to append inverter serial number at the end of service name. This breaks the UI (static definition) but dynamic calls from yaml work fine.
> 
> ```python
>     hass.services.async_register(
>         DOMAIN, f"{SERVICE_WRITE_REGISTER}_{inverter._serial}", write_holding_register, schema=SERVICE_WRITE_REGISTER_SCHEMA
>     )
> 
>     hass.services.async_register(
>         DOMAIN, f"{SERVICE_WRITE_MULTIPLE_REGISTERS}_{inverter._serial}", write_multiple_holding_registers, schema=SERVICE_WRITE_MULTIPLE_REGISTERS_SCHEMA
>     )
> ```

I tries this. I see all available service calls, but still only the last added inverter responds.

butalive666, 2024-06-17:
> I will also try to get the GUI generated helpers into a YAML or even a template, so that I could provide the code for everyone. This is unfortunately still tricky for me, but I will try my best.

A Template would be nice :-)

i need a solution for 2 cases. When we shower, and the heating system is on.  In both cases i dont need to export anything.

iroc-fabi, 2024-06-24:
Wie kann man in HomeAssistent die Nulleinspeisung für einen Deye einrichten?
Gibt es dafür eine Anleitung?

Ricox1975, 2024-06-30:
@iroc-fabi  schaust du hier:

[](https://community.home-assistant.io/t/deye-solar-inverter-zero-feed-in-nulleinspeisung/740718)

ugath, 2024-10-02:
> > Same issue here. It would be enough to control the master inverter (slave would just copy the values) but unfortunately my HA integration seems to change the values on slave instead.
> > edit: I managed to modify `services.py` to append inverter serial number at the end of service name. This breaks the UI (static definition) but dynamic calls from yaml work fine.
> > ```python
> >     hass.services.async_register(
> >         DOMAIN, f"{SERVICE_WRITE_REGISTER}_{inverter._serial}", write_holding_register, schema=SERVICE_WRITE_REGISTER_SCHEMA
> >     )
> > 
> >     hass.services.async_register(
> >         DOMAIN, f"{SERVICE_WRITE_MULTIPLE_REGISTERS}_{inverter._serial}", write_multiple_holding_registers, schema=SERVICE_WRITE_MULTIPLE_REGISTERS_SCHEMA
> >     )
> > ```
> 
> EDIT: Fixed it. I filled in all serials where i should leave your code as it is. Amazing job!
> 
> I tried this. I see all available service calls, but still only the last added inverter responds.

Hi!
I have adapted service.py accordingly.
How do I have to/can I specify the inverter serial number?

[image]

Elijen, 2024-10-02:
@ugath Just check the new services that are generated and copy paste the names.

BTW I have since stopped using this integration and am using https://github.com/davidrapan/ha-solarman which works out of the box and has more features (like ability to change inverter configuration).

ugath, 2024-10-03:
> This is what i get. Dont need to specify the serials myself. Same goes for the write multiple holding registers, which are the ones you need.

Thanks! 
I forgot to restart Home Assistant. :)