StephanJoubert/home_assistant_solarman #487: Problem connecting to Deya SUN-5K-G

alexmerqq, 2024-01-13:
.
I have two inverters - Deye SUN-5K-SG03LP1 and Deya SUN-5K-G .
I connect to the hybrid inverter without any problems and read all the necessary parameters.
I configure the second inverter in exactly the same way, but its status remains Disabled.
What could be the problem and what else can I check?

valeriosouza, 2024-01-16:
From December 17th until now I also lost connection. using deye_string

qfinucane, 2024-01-20:
> From December 17th until now I also lost connection. using deye_string

I have the same issue after updating Home assistant core in December. I would suggest checking you core logs under system. See if you get this error:

2024-01-20 10:46:10.684 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 112] failed with exception [TypeError: unsupported operand type(s) for -: 'str' and 'int']
2024-01-20 10:46:11.184 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 112] failed, [0] retry attempts left
2024-01-20 10:46:11.184 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying registers [3 - 112] failed, aborting.

I restore to the last version i had installed 2023.4.4. The integration started working again. How to investigate why the update causes this.

basersoft, 2024-01-27:
I've been having the same problem for days. I went back to the old version 1.4 and ran it. Now it works without any problems.

valeriosouza, 2024-01-29:
solarman 1.4?

Ramaddan, 2024-02-04:
Yes, I have similar problems with the sun-8k-sg01lp1

Have to keep to version 1.4 which does not have a problem

valeriosouza, 2024-02-04:
This 1.4.1?
[image: image.png]

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
04/02/24,
15:32:58

Ramaddan, 2024-02-04:
Sorry, using 1.4.3 to be exact

basersoft, 2024-02-05:
I ran it with version 1.4. I didn't do much.
I used deye_string.yaml. It works without any problems.

Ramaddan, 2024-02-05:
Yes, it works with the versions below 1.5, but the minute I try version 1.5 and above, it stops working.

qfinucane, 2024-02-05:
Thanks @Ramaddan , How do you see which version is integrated in HA?

Ramaddan, 2024-02-05:
Hi,

I used HACS to install in Home Assistant.

I just went into the HACS page, and looked at installed integrations.

Chose Solarman, and here it should tell you what version you are on the top left hand side.

If I remember right, I had to remove the integration first of the version 1.5, and then clicked on the three dots on the right side, and downloaded the specific version I needed.

qfinucane, 2024-02-05:
@Ramaddan , Thank you very much.

I'll give it a try.