StephanJoubert/home_assistant_solarman #492: Problem Board HA board Solar production.

denimassi, 2024-01-19:
Good morning, something strange happens to me on the Energy Dashboard, the Solar production sensor, every morning it records a negative value that varies from 5 to 8 Kw. If I check the sensor from the HA History the negative value is not present. Do you have any idea where to find the error. Thank you

fraenkn, 2024-01-28:
I Have the same Problem
[image]

MikealSouth, 2024-01-31:
Same issue here.
I've just switched the dashboard from using "daily_production" to using "Total Production". This number should only increase and never be negative.
I'll report back in two days if I still experience this issue.

Inverter: SUN-M60G3 (not experiencing this problem with my 3-phase inverter)
Version used: 1.5.1
Core: 2024.1.6
Supervisor: 2023.12.1
Operating System: 11.4
Frontend: 20240104.0
![Screenshot from 2024-01-31 12-50-45]

MikealSouth, 2024-02-02:
Update: This has solved it for me, no more negative solar production.