StephanJoubert/home_assistant_solarman #493: Solarman

denimassi, 2024-01-19:
[image]

Can anyone explain to me why PV1 is higher than Energy Total?

qfinucane, 2024-01-20:
As far as I am aware the Inverter itself consumes between 50 and 100watt. I don't think it's taken into consideration with the calculation.

denimassi, 2024-01-20:
you found it… thanks. I did not think about it. as usual the answer right under my nose. Thank you.
Inviato da iPhone

> Il giorno 20 gen 2024, alle ore 09:29, qfinucane ***@***.***> ha scritto:
> 
> As far as I am aware the Inverter itself consumes between 50 and 100watt. I don't think it's taken into consideration with the calculation