StephanJoubert/home_assistant_solarman #494: Integration fails

qfinucane, 2024-01-20:
My Solarman integration is failing post a Home Assistant update at the end of December. has anyone else experienced this?

2024-01-20 10:46:10.684 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 112] failed with exception [TypeError: unsupported operand type(s) for -: 'str' and 'int']
2024-01-20 10:46:11.184 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying [3 - 112] failed, [0] retry attempts left
2024-01-20 10:46:11.184 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying registers [3 - 112] failed, aborting.

Slawcio2000, 2024-01-20:
Hi,
After update Home Assistant i cant get integration with SOFAR....
HA versions:
Core
2024.1.4
Supervisor
2023.12.1
Operating System
11.4

qfinucane, 2024-01-20:
I restored to the below Core version and the integration started working again.

Version
core-2023.4.1

bbjunkie, 2024-01-21:
Likewise, broken here too after update, I believe this has been logged already as issue #474