StephanJoubert/home_assistant_solarman #499: error: Querying [512 - 597] failed with exception [OSError: [Errno 30] Read-only file system]

murraythegoz, 2024-01-23:
Everything worked till last week. then it suddenly stopped working.

running against a ZCS azzurro ZM1ES060 with firmware V290, with the sofar_hyd3k-6k-es.yaml template.

now when trying I get the following errors: 
Querying [512 - 597] failed with exception [OSError: [Errno 30] Read-only file system]
Querying [512 - 597] failed with exception [NoSocketAvailableError: No socket available]

any suggestion?

alain57, 2024-01-28:
same issue, i was thinking that all this was my unstable wifi, and then saw he logs and this message :/ 
If someone has any clue on how to fix that i would highly appreciate it

murraythegoz, 2024-01-29:
opened a ticket to the vendor.

they are evaluating a firmware update to check if it could fix the issue.

just for the records, the not working configuration releases:
Inverter: V290
logger: ME_08_2701_2.06
connected via ethernet

ferrarisimo, 2024-01-29:
Hi, I have a ZCS HYD 5000 HP and I'm facing the same issue too 
In my log files I see the same Read-only logs and in addiction I also see:
Querying registers [1028 - 1056] failed, aborting.
The wifi connection is ok, it seems a "server" change in a registry read list ... 
The official Azzuro App works fine.
Hope this could be useful

murraythegoz, 2024-01-29:
updated inverter to V360, and logger to ME_0D_2701_5.09 (activities performed by Zucchetti following up a request from me).

hard reboot the whole inverter, but still not working.

murraythegoz, 2024-01-29:
I was redirected to the standard documentation:

https://pysolarmanv5.readthedocs.io/en/stable/solarmanv5_protocol.html#frame-diagrams

Sutek, 2024-01-30:
I had the same problem with this integration, but I managed to find a solution. I have Home Assistant Supervisor installed. It turned out that Docker version 25.0.1, which was supposed to be HA compatible, caused problems with Solarman integration. Downgrading Docker to version 24.0.7 solved the problem.
```
sudo apt install docker-ce=5:24.0.7-1~debian.12~bookworm docker-ce-cli=5:24.0.7-1~debian.12~bookworm docker-buildx-plugin=0.11.2-1~debian.12~bookworm docker-compose-plugin=2.21.0-1~debian.12~bookworm docker-ce-rootless-extras=5:24.0.7-1~debian.12~bookworm
sudo apt-mark hold docker-ce docker-ce-cli docker-buildx-plugin docker-compose-plugin docker-ce-rootless-extras
sudo reboot
```

ferrarisimo, 2024-01-30:
> I had the same problem with this integration, but I managed to find a solution. I have Home Assistant Supervisor installed. It turned out that Docker version 25.0.1, which was supposed to be HA [...]

Thanks Sutek it works for me.
I forgot that yesterday I upgraded my Debian and the only updated packages were docker 25

murraythegoz, 2024-01-30:
still not working, but good point, thank you.

Querying registers [512 - 597] failed, aborting.
09:43:01 – (WARNING) Solarman (custom integration) - message first occurred at 09:36:33 and shows up 14 times
Querying [512 - 597] failed, [1] retry attempts left
09:43:01 – (WARNING) Solarman (custom integration) - message first occurred at 09:36:31 and shows up 28 times
Querying [512 - 597] failed with exception [V5FrameError: V5 frame contains invalid start or end values]
09:43:01 – (WARNING) Solarman (custom integration) - message first occurred at 09:36:30 and shows up 28 times

alain57, 2024-01-30:
I confirm that @Sutek found a working way 
i have again my deye working correctly in HA thanks a lot ...
I just hope that it will be fixed in HA or docker so that the package is up to date :)

Joshua2372, 2024-02-10:
> I had the same problem with this integration, but I managed to find a solution. I have Home Assistant Supervisor installed. It turned out that Docker version 25.0.1, which was supposed to be HA [...]

Good practice! problem solved!
now let's hope they solve Docker with a new upgrade.
ThankU a lot.

81ondo, 2024-02-23:
What about not supervised HA installation? Any advices to fix this?

doreks, 2024-03-14:
finding the same problem here but i cannot fixed with the WA, any idea :

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package docker-ce is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  docker-ce-cli

Package docker-buildx-plugin is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package docker-ce-cli is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  docker-buildx-plugin

Package docker-compose-plugin is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

```
Package docker-ce-rootless-extras is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Version '5:24.0.7-1~debian.12~bookworm' for 'docker-ce' was not found
E: Version '5:24.0.7-1~debian.12~bookworm' for 'docker-ce-cli' was not found
E: Version '0.11.2-1~debian.12~bookworm' for 'docker-buildx-plugin' was not found
E: Version '2.21.0-1~debian.12~bookworm' for 'docker-compose-plugin' was not found
E: Version '5:24.0.7-1~debian.12~bookworm' for 'docker-ce-rootless-extras' was not found

```

kefrens, 2024-04-11:
I just upgraded to HAOS 12.2 and have the same issue.
Looking at the release notte they have upgrade to docker 25 which obsivouly is not working anymore.

Does anyone have a fix for it? I can't run the sudo command and I am using HAOS through a virtualbox image....
I dont even think that the backup will revert to HAOS 12.1...

kefrens, 2024-04-11:
> What about not supervised HA installation? Any advices to fix this?

See above , in termnal: `ha os update 12.1`

fhrbek, 2024-04-12:
Same here, solarman stopped working after upgrading to 12.2. The downgrade to 12.1 helped but the upgrade hint is back. I hope this will be solved so that I can upgrade again.

kefrens, 2024-04-12:
> Same here, solarman stopped working after upgrading to 12.2. The downgrade to 12.1 helped but the upgrade hint is back. I hope this will be solved so that I can upgrade again.

Turns out you just need to do following command on 12.2 after upgrade`ha core rebuild`

It works for me now on 12.2

doreks, 2024-04-12:
> > Same here, solarman stopped working after upgrading to 12.2. The downgrade to 12.1 helped but the upgrade hint is back. I hope this will be solved so that I can upgrade again.
> 
> Turns out you just need to do following command on 12.2 after upgrade`ha core rebuild`
> 
> It works for me now on 12.2

 applied this WA in container hassio_cli and now works!

murraythegoz, 2024-04-15:
Just upgraded to core 2024.4.3 and upgraded solarman to 1.5.1 (my probe has serial 21xxxxx).

Still no luck :-(

AndyNew2, 2024-04-17:
If anyone interested you can go back to HA OS 12.1 with this in the shell:
ha os update --version 12.1

theyo-tester, 2024-07-31:
> > Same here, solarman stopped working after upgrading to 12.2. The downgrade to 12.1 helped but the upgrade hint is back. I hope this will be solved so that I can upgrade again.
> 
> Turns out you just need to do following command on 12.2 after upgrade`ha core rebuild`
> 
> It works for me now on 12.2

Yes, this fixed it on my side too, after upgrading to 12.4

akhoury, 2025-02-27:
> applied this WA in container hassio_cli and now works!

This worked for me! thanks!

```
docker exec -it hassio_cli bash ⁠
ha core rebuild
```